arendst/Tasmota #7095: (Sonoff POW) NTP not working after Upgrade to 7.1

Lu-Fi, 2019-12-03:
### PROBLEM DESCRIPTION
After Upgrade from 6.6.0 to 7.1.0 to 7.1.1 NTP is broken.
I have seen, that a redisign is planned for 2.0 but did not find any Information regarding it's currently broken.
I tested this against my Router (Fritz!Box) and also some pool ntp server.
Device Type: Sonoff POW 1

```
00:11:00 CMD: ntpserver 192.168.178.1
00:11:00 MQT: stat/SonoffPow_7/RESULT = {"NtpServer1":"192.168.178.1"}
00:11:04 CMD: Time 0
00:11:05 MQT: stat/SonoffPow_7/RESULT = {"Time":"1970-01-01T00:11:05","Epoch":665}
00:11:45 CMD: ntpserver 0.de.pool.ntp.org
00:11:45 MQT: stat/SonoffPow_7/RESULT = {"NtpServer1":"0.de.pool.ntp.org"}
00:11:50 CMD: Time 0
00:11:51 MQT: stat/SonoffPow_7/RESULT = {"Time":"1970-01-01T00:11:51","Epoch":711}
00:12:08 CMD: Time
00:12:08 MQT: stat/SonoffPow_7/RESULT = {"Time":"1970-01-01T00:12:08","Epoch":728}
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: _____
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: TasmoAdmin (Tasmota Webinterface)
- [x] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
00:18:14 MQT: stat/SonoffPow_7/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
00:18:15 MQT: stat/SonoffPow_7/RESULT = {"Module":{"6":"Sonoff Pow"}}
00:18:15 MQT: stat/SonoffPow_7/RESULT = {"GPIO":"Not supported"}

  ```
- [-] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
00:18:55 MQT: stat/SonoffPow_7/STATUS = {"Status":{"Module":6,"FriendlyName":["Sonoff"],"Topic":"SonoffPow_7","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:18:55 MQT: stat/SonoffPow_7/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.178.45:9999/data/firmwares/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:18:51","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":279,"SaveCount":1576,"SaveAddress":"F6000"}}
00:18:55 MQT: stat/SonoffPow_7/STATUS2 = {"StatusFWR":{"Version":"7.1.1(tasmota)","BuildDateTime":"2019-12-01T13:00:09","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
00:18:55 MQT: stat/SonoffPow_7/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FB","FB"],"TelePeriod":300,"Resolution":"558180D0","SetOption":["04004009","2805C8000100060000005AFF000000000000","00000001","00000000"]}}
00:18:55 MQT: stat/SonoffPow_7/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":4,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
00:18:55 MQT: stat/SonoffPow_7/STATUS5 = {"StatusNET":{"Hostname":"POW-KE-SCHRANK","IPAddress":"192.168.178.29","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"2C:3A:E8:3B:7C:3F","Webserver":2,"WifiConfig":4}}
00:18:55 MQT: stat/SonoffPow_7/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.45","MqttPort":7625,"MqttClientMask":"SonoffPow_7","MqttClient":"SonoffPow_7","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:18:55 MQT: stat/SonoffPow_7/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:18:55 1970","Local":"Thu Jan 01 00:18:55 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:18:55 MQT: stat/SonoffPow_7/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:18:55 MQT: stat/SonoffPow_7/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:18:55","Epoch":1135,"ENERGY":{"TotalStartTime":"2019-01-03T21:39:37","Total":65.217,"Yesterday":0.386,"Today":0.881,"Power":16,"ApparentPower":31,"ReactivePower":26,"Factor":0.52,"Voltage":231,"Current":0.134}}}
00:18:55 MQT: stat/SonoffPow_7/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:18:55","Epoch":1135,"Uptime":"0T00:18:51","UptimeSec":1131,"Heap":4,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"on","Wifi":{"AP":2,"SSId":"FB","BSSId":"36:B5:C2:75:78:75","Channel":9,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:06"}}}

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
00:00:37 CMD: weblog 4
00:00:37 MQT: stat/SonoffPow_7/RESULT = {"WebLog":4}
00:00:37 CFG: Saved to flash at F4, Count 1578, Bytes 4096
00:00:45 WIF: Checking connection...
00:00:45 WIF: Connected
00:01:05 WIF: Checking connection...
00:01:05 WIF: Connected
00:01:06 CMD: ntpserver 192.168.178.1
00:01:06 SRC: WebConsole from 192.168.178.201
00:01:06 CMD: Group 0, Index 1, Command "NTPSERVER", Data "192.168.178.1"
00:01:06 MQT: stat/SonoffPow_7/RESULT = {"NtpServer1":"192.168.178.1"}
00:01:06 CFG: Saved to flash at FB, Count 1579, Bytes 4096
00:01:16 CMD: Time 0
00:01:16 SRC: WebConsole from 192.168.178.201
00:01:16 CMD: Group 0, Index 1, Command "TIME", Data "0"
00:01:17 MQT: stat/SonoffPow_7/RESULT = {"Time":"1970-01-01T00:01:17","Epoch":77}
00:01:26 WIF: Checking connection...
00:01:26 WIF: Connected

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-03:
Please, check if downgrading reverts your issue. We are aware that some ISP are blocking NTP servers, and if they block the one that uses your routers, Tasmota won't sync the Time. Tasmota uses the NTP provided by the router when using DHCP

arendst, 2019-12-03:
The current NTP implementation tries to get the correct time every second within the first seconds after restart. If it fails it retries every minute offset by the MAC address lowest nibble times 3 + 3. In your case that's every 48 seconds within a minute.

```
00:00:00 CFG: Loaded from flash at FB, Count 177
00:00:00 SRC: Restart
00:00:00 Project tasmota Wemos4 Version 7.1.1.1(tasmota)-2_6_1
00:00:00 DSB: Sensors found 1
00:00:00 I2C: BME680 found at 0x77
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP1, SSId indebuurt1, Channel 1, BSSId 24:D3:F2:97:C0:A1, RSSI -70, Encryption 1
00:00:04 WIF: Network 1, AP2, SSId indebuurt2, Channel 5, BSSId A0:AB:1B:7D:42:AC, RSSI -36, Encryption 1
00:00:04 WIF: Network 2, AP-, SSId indebuurt3, Channel 12, BSSId 60:E3:27:58:77:E6, RSSI -82, Encryption 1
00:00:04 WIF: Connecting to AP2 indebuurt2 in mode 11N as wemos4...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Connected
00:00:07 HTP: Web server active on wemos4 with IP address 192.168.2.223
NTP: Retry sync offset 7
00:00:08 APP: Boot Count 53
00:00:08 CFG: Saved to flash at FA, Count 178, Bytes 4096
NTP: Retry sync offset 8
00:00:09 MQT: Attempting connection...
00:00:09 MQT: Connected
00:00:09 MQT: tele/wemos4/LWT = Online (retained)
00:00:09 MQT: cmnd/wemos4/POWER = 
00:00:09 MQT: Subscribe to cmnd/wemos4/#
00:00:09 MQT: Subscribe to cmnd/tasmotas/#
00:00:09 MQT: Subscribe to cmnd/DVES_83BB10_fb/#
00:00:09 MQT: tele/wemos4/INFO1 = {"Module":"Generic","Version":"7.1.1.1(tasmota)","FallbackTopic":"cmnd/DVES_83BB10_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:09 MQT: tele/wemos4/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos4","IPAddress":"192.168.2.223"}
00:00:09 MQT: tele/wemos4/INFO3 = {"RestartReason":"Software/System restart"}
00:00:09 MQT: stat/wemos4/POWER = {"POWER":"ON"}
00:00:09 MQT: stat/wemos4/POWER = ON
NTP: Retry sync offset 9
NTP: Retry sync offset 10
NTP: Retry sync offset 11
NTP: Retry sync offset 12
00:00:13 MQT: tele/wemos4/STATE = {"Time":"1970-01-01T00:00:13","Uptime":"0T00:00:15","UptimeSec":15,"Heap":31,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":2,"SSId":"indebuurt2","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:09"}}
00:00:13 MQT: tele/wemos4/SENSOR = {"Time":"1970-01-01T00:00:13","DS18B20":{"Id":"0000048EC44C","Temperature":21.6},"BME680":{"Temperature":22.4,"Humidity":50.3,"Pressure":1027.2,"Gas":528.88},"PressureUnit":"hPa","TempUnit":"C"}
NTP: Retry sync offset 13
NTP: Retry sync offset 14
NTP: Retry sync offset 15
NTP: Retry sync offset 16
NTP: Retry sync offset 17
NTP: Retry sync offset 18
NTP: Retry sync offset 19
NTP: Retry sync offset 20
NTP: Retry sync offset 21
NTP: Retry sync offset 22
NTP: Retry sync offset 23
NTP: Retry sync offset 24
NTP: Retry sync offset 25
NTP: Retry sync offset 26
NTP: Retry sync offset 27
NTP: Retry sync offset 48
NTP: Retry sync offset 48
NTP: Retry sync offset 48
NTP: Retry sync offset 48
NTP: Retry sync offset 48
00:05:13 MQT: tele/wemos4/STATE = {"Time":"1970-01-01T00:05:13","Uptime":"0T00:05:15","UptimeSec":315,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":2,"SSId":"indebuurt2","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:09"}}
00:05:13 MQT: tele/wemos4/SENSOR = {"Time":"1970-01-01T00:05:13","DS18B20":{"Id":"0000048EC44C","Temperature":21.7},"BME680":{"Temperature":22.8,"Humidity":49.7,"Pressure":1027.1,"Gas":606.95},"PressureUnit":"hPa","TempUnit":"C"}
NTP: Retry sync offset 48
```
In the above example I removed all three NTPservers from the configuration an I know my gateway does not provide NTP.

In your situation it doesn't seem to find the correct NTP server.

Lu-Fi, 2019-12-03:
@ascillato 
Yes downgrade to 6.6 fixed the Issue.
Since i'am using my own NTP, there is no posibility to block the connection. I tested 2 different of my own NTP servers and also Public once.

@arendst 
Currently around 20 Tasmota Devices running here. 
Sonoff, Shelly, Blitzwolf, WeMos and MagicHome. Last Week i tried to upgrade to 7.1 first on all devices. After running into trouble on more than 5 of them i revert back to 6.6. and started Troubleshooting. Here i'am having one Fritzbox and on OpenWrt Router running and both can provide NTP. All NTP Servers ar up, running and reachable.
For now i have all Devices back on 6.6 and everything works as expected. Only one Device is running Version 7.1 for Testing. 

Current Startup Weblog (Level 4), should i also see NTP Requests here ? 
```
00:00:00 CFG: Loaded from flash at FA, Count 1588
00:00:00 QPC: Flag 0E
00:00:00 SRC: Restart
00:00:00 Project tasmota Sonoff Version 7.1.1(tasmota)-2_6_1
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP2 FB in mode 11N as POW-KE-SCHRANK...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on POW-KE-SCHRANK with IP address 192.168.178.29
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/SonoffPow_7/LWT = Online (retained)
00:00:06 MQT: cmnd/SonoffPow_7/POWER = 
00:00:06 MQT: Subscribe to cmnd/SonoffPow_7/#
00:00:06 MQT: Subscribe to cmnd/sonoffs/#
00:00:06 MQT: Subscribe to cmnd/SonoffPow_7_fb/#
00:00:06 MQT: tele/SonoffPow_7/INFO1 = {"Module":"Sonoff Pow","Version":"7.1.1(tasmota)","FallbackTopic":"cmnd/SonoffPow_7_fb/","GroupTopic":"cmnd/sonoffs/"}
00:00:06 MQT: tele/SonoffPow_7/INFO2 = {"WebServerMode":"Admin","Hostname":"POW-KE-SCHRANK","IPAddress":"192.168.178.29"}
00:00:06 MQT: tele/SonoffPow_7/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/SonoffPow_7/RESULT = {"POWER1":"on"}
00:00:06 MQT: stat/SonoffPow_7/POWER1 = on
00:00:06 QPC: Reset
00:00:07 HTP: Main Menu
00:00:08 APP: Boot Count 287
00:00:08 CFG: Saved to flash at F9, Count 1589, Bytes 4096
00:00:10 MQT: tele/SonoffPow_7/STATE = {"Time":"1970-01-01T00:00:10","Epoch":10,"Uptime":"0T00:00:12","UptimeSec":12,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"on","Wifi":{"AP":2,"SSId":"FB","BSSId":"36:B5:C2:75:78:75","Channel":9,"RSSI":62,"LinkCount":1,"Downtime":"0T00:00:06"}}
00:00:10 MQT: tele/SonoffPow_7/SENSOR = {"Time":"1970-01-01T00:00:10","Epoch":10,"ENERGY":{"TotalStartTime":"2019-01-03T21:39:37","Total":65.257,"Yesterday":0.386,"Today":0.921,"Period":0,"Power":16,"ApparentPower":32,"ReactivePower":28,"Factor":0.50,"Voltage":231,"Current":0.138}}
00:00:25 WIF: Checking connection...
00:00:25 WIF: Connected
00:00:45 WIF: Checking connection...
00:00:45 WIF: Connected
00:01:05 WIF: Checking connection...
00:01:05 WIF: Connected
00:01:25 WIF: Checking connection...
00:01:25 WIF: Connected
00:01:45 WIF: Checking connection...
00:01:45 WIF: Connected
00:02:05 WIF: Checking connection...
00:02:05 WIF: Connected
00:02:25 WIF: Checking connection...
00:02:25 WIF: Connected
00:02:45 WIF: Checking connection...
00:02:45 WIF: Connected
00:02:48 HTP: Console
00:03:05 WIF: Checking connection...
00:03:05 WIF: Connected
00:03:25 WIF: Checking connection...
00:03:25 WIF: Connected
00:03:45 WIF: Checking connection...
00:03:45 WIF: Connected
00:03:52 CMD: ntpserver
00:03:52 SRC: WebConsole from 192.168.178.201
00:03:52 CMD: Group 0, Index 1, Command "NTPSERVER", Data ""
00:03:52 MQT: stat/SonoffPow_7/RESULT = {"NtpServer1":"192.168.178.1"}
00:03:56 CMD: ntpserver2
00:03:56 SRC: WebConsole from 192.168.178.201
00:03:56 CMD: Group 0, Index 2, Command "NTPSERVER", Data ""
00:03:56 MQT: stat/SonoffPow_7/RESULT = {"NtpServer2":"nl.pool.ntp.org"}
00:04:01 CMD: ntpserver3
00:04:01 SRC: WebConsole from 192.168.178.201
00:04:01 CMD: Group 0, Index 3, Command "NTPSERVER", Data ""
00:04:01 MQT: stat/SonoffPow_7/RESULT = {"NtpServer3":"0.nl.pool.ntp.org"}
00:04:05 WIF: Checking connection...
00:04:05 WIF: Connected
00:04:11 CMD: ntpserver4
00:04:11 SRC: WebConsole from 192.168.178.201
00:04:11 CMD: Group 0, Index 4, Command "NTPSERVER", Data ""
00:04:11 MQT: stat/SonoffPow_7/RESULT = {"Command":"Error"}
00:04:18 CMD: Time
00:04:18 SRC: WebConsole from 192.168.178.201
00:04:18 CMD: Group 0, Index 1, Command "TIME", Data ""
00:04:18 MQT: stat/SonoffPow_7/RESULT = {"Time":"1970-01-01T00:04:18","Epoch":258}
00:04:25 WIF: Checking connection...
00:04:25 WIF: Connected
00:04:30 CMD: time 0
00:04:30 SRC: WebConsole from 192.168.178.201
00:04:30 CMD: Group 0, Index 1, Command "TIME", Data "0"
00:04:31 MQT: stat/SonoffPow_7/RESULT = {"Time":"1970-01-01T00:04:31","Epoch":271}
00:04:46 WIF: Checking connection...
00:04:46 WIF: Connected
```

i found one hint within the Tasmota Community to Reset to default can fix this, but i don't know if this hint is what helps to solve issues.
I can try this next days.
https://groups.google.com/forum/#!searchin/sonoffusers/NTP%7Csort:date/sonoffusers/-7HDRJzgj8c/BcNwPdlrBAAJ

arendst, 2019-12-03:
If you can monitor your network for NTP request you should expect requests from the device as soon as wifi is connected.

BTW: you won't see messages like ``NTP: Retry sync offset 22`` as they were only implemented in my local code. Problem with messages in NTP interrupt code is that they will cause exceptions due to tcp/udp.

mike2nl, 2019-12-04:
Only to give my 2 cents to this

- NTP on sonoff SV, sonoff SC, sonoff basic, sonoff iFAN2, sonoff B1
Used Tasmota version: 7.1.1.1
Setup: own compile test under platformio and adruino ide
my own NTP server 1 on a PLC 192.168.2.194
my own NTP server 2 on a windows ;-) PC 192.168.4.96 
(yes different networks for different hardware and levels of the house)
used commands to check all settings: ntpserver, timezone is 99, timedst, timestd
the ntpserver command gives me the right ip to the PLC

status 10 gives me the right time and date  from the ntp server on the plc which gets the right time from the internet via a firewall. and only the plc can get over it in this case (DMZ). because the plc can't be hacked about a special pvi interface.

arendst, 2019-12-04:
All my devices run fine with NTP. Always did and currently do. Either on pool.ntp.org or my local NTP servers.

So the issue is not that NTP doesn't work in Tasmota. The issue here is why doesn't NTP work in this specific case.

Jason2866, 2019-12-05:
@Lu-Fi  I do run my own NTP server too. The are running fine. The always did and the do with 7.1.x. 
Tried also with pool.ntp.org and ntp1.t-online.de. Worked too.

Lu-Fi, 2019-12-05:
Ok, sry no Idea what else to try.
Today i reset this Device to default. 
After Reset i configured WIFI, MQTT and entered old Calibration / History Values. Also "Time 0".
I tested my own NTP and also pool NTP Servers. 

Same result as before. No NTP. 

Then i used my Router (Fritzbox) to sniff Network Traffic.
1. Reboot non working Device (7.1.1) -> NO NTP Packages captured
2. Reboot Another Device (6.6) -> NTP Request captured
3. Downgrade non working Device to 6.6
4. Reboot non Working Device (now 6.6) -> NTP Request captured

Ok, since i'am the only one with this Problem. It's not important but strange.

Slevin1988, 2019-12-05:
Hallo,
I have the same issue (8 Devices). But i find out i have just the Problem if i use the fixed IPAddress.
With DHCP 0.0.0.0 it Could you Check it?

alofi198, 2019-12-05:
Hello, 
have the same problem, ntp not working, after remove the mac reservation on my dhcp, ntp is working again.

Lu-Fi, 2019-12-05:
Extrem crazy but clearing dhcp reservation helps here ........

ascillato2, 2019-12-05:
> Extrem crazy but clearing dhcp reservation helps here ........

Great to know that now it is working for you. :+1:

closed 2019-12-05 by ascillato2

Lu-Fi, 2019-12-06:
It's ok for me that you closed this, but don't forget, this may a Tasmota or Library bug. This is not a DHCP bug ! 
I think there are some informations cached which received from DHCP. And this still persists also after a reset. 
Is this possible ?

Jason2866, 2019-12-06:
@Lu-Fi  what makes you believe it can not be a bug in DHCP?

jnovack, 2020-09-05:
Should NTP be working by default on a `tasmota-minimal` installation of 8.4.0.3?  Sure, the timezone would be off (if you are not UTC+1, but the date should be set to sometime today.

```text
00:17:11 CMD: ntpserver
00:17:11 RSL: stat/tasmota/RESULT = {"NtpServer1":"poo.ntp.org","NtpServer2":"nl.pool.ntp.org","NtpServer3":"0.nl.pool.ntp.org"}
00:17:24 CMD: time
00:17:24 RSL: stat/tasmota/RESULT = {"Time":"1970-01-01T00:17:24"}
```

ascillato, 2020-09-05:
Tasmota minimal is not a firmware to be on normal use. It is ONLY for updating. Use Tasmota.bin instead.

Tasmota minimal is just for freeing flash space in order to do OTA to the full version. 

Anyway, it is not needed anymore because you can now do OTA with the compressed firmware Tasmota.bin.gz

jnovack, 2020-09-06:
That wasn't the point of the question.

The point of the question is it supposed to be working on a completely fresh install with a device completely reset?

ascillato, 2020-09-06:
Minimal IS NOT for normal usage. It is not for first flash neither. It is just for the updating process. It does not have any driver.

You should use **tasmota.bin** or **tasmota_lite.bin** (if you need a small flash usage firmware)

Jason2866, 2020-09-06:
Expect nothing to work with tasmota-minimal. The only use case is beeing small as possible
for updating.