arendst/Tasmota #7097: Home Assistant Integration - Custom discovery_prefix

xefil, 2019-12-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Hello, I've a problem setting up discovery of my two tasmota in Home Assistant. I've last versions and the problem (I think) is due the custom discovery_prefix I'm using in Home Assistant:

```
  discovery: true
  discovery_prefix: XefilHome
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_
Is there a way to change the tasmota discovery prefix? As soon I use the `SetOption19 1` I see the follow, which is not working:

```
13:38:02 MQT: tasmota01/tele/LWT = Online (retained)
13:38:02 MQT: tasmota01/cmnd/POWER = 
13:38:02 MQT: tasmota01/tele/INFO1 = {"Module":"Gosund SP1 v23","Version":"7.1.1(tasmota)","FallbackTopic":"cmnd/DVES_522851_fb/","GroupTopic":"tasmotas/cmnd/"}
13:38:02 MQT: tasmota01/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota01","IPAddress":"192.168.3.21"}
13:38:02 MQT: tasmota01/tele/INFO3 = {"RestartReason":"Software/System restart"}
13:38:02 MQT: tasmota01/stat/RESULT = {"POWER":"ON"}
13:38:02 MQT: tasmota01/stat/POWER = ON
13:38:03 MQT: homeassistant/sensor/522851_ENERGY_Yesterday/config = {"name":"Tasmota ENERGY Yesterday","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"522851_ENERGY_Yesterday","device":{"identifiers":["522851"],"connections":[["mac","84:0D:8E:52:28:51"]]},"~":"tasmota01/tele/","unit_of_meas":"kWh","val_tpl":"{{value_json['ENERGY'].Yesterday}}","dev_cla":"power"} (retained)
13:38:03 MQT: homeassistant/sensor/522851_ENERGY_Today/config = {"name":"Tasmota ENERGY Today","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"522851_ENERGY_Today","device":{"identifiers":["522851"],"connections":[["mac","84:0D:8E:52:28:51"]]},"~":"tasmota01/tele/","unit_of_meas":"kWh","val_tpl":"{{value_json['ENERGY'].Today}}","dev_cla":"power"} (retained)
13:38:04 MQT: homeassistant/sensor/522851_ENERGY_Power/config = {"name":"Tasmota ENERGY Power","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"522851_ENERGY_Power","device":{"identifiers":["522851"],"connections":[["mac","84:0D:8E:52:28:51"]]},"~":"tasmota01/tele/","unit_of_meas":"W","val_tpl":"{{value_json['ENERGY'].Power}}","dev_cla":"power"} (retained)
13:38:04 MQT: homeassistant/sensor/522851_ENERGY_ApparentPower/config = {"name":"Tasmota ENERGY ApparentPower","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"522851_ENERGY_ApparentPower","device":{"identifiers":["522851"],"connections":[["mac","84:0D:8E:52:28:51"]]},"~":"tasmota01/tele/","unit_of_meas":" ","val_tpl":"{{value_json['ENERGY'].ApparentPower}}"} (retained)
13:38:04 MQT: homeassistant/sensor/522851_ENERGY_ReactivePower/config = {"name":"Tasmota ENERGY ReactivePower","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"522851_ENERGY_ReactivePower","device":{"identifiers":["522851"],"connections":[["mac","84:0D:8E:52:28:51"]]},"~":"tasmota01/tele/","unit_of_meas":" ","val_tpl":"{{value_json['ENERGY'].ReactivePower}}"} (retained)
13:38:04 MQT: homeassistant/sensor/522851_ENERGY_Factor/config = {"name":"Tasmota ENERGY Factor","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"522851_ENERGY_Factor","device":{"identifiers":["522851"],"connections":[["mac","84:0D:8E:52:28:51"]]},"~":"tasmota01/tele/","unit_of_meas":" ","val_tpl":"{{value_json['ENERGY'].Factor}}"} (retained)
13:38:04 MQT: homeassistant/sensor/522851_ENERGY_Voltage/config = {"name":"Tasmota ENERGY Voltage","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"522851_ENERGY_Voltage","device":{"identifiers":["522851"],"connections":[["mac","84:0D:8E:52:28:51"]]},"~":"tasmota01/tele/","unit_of_meas":"V","val_tpl":"{{value_json['ENERGY'].Voltage}}","dev_cla":"power"} (retained)
13:38:04 MQT: homeassistant/sensor/522851_ENERGY_Current/config = {"name":"Tasmota ENERGY Current","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"522851_ENERGY_Current","device":{"identifiers":["522851"],"connections":[["mac","84:0D:8E:52:28:51"]]},"~":"tasmota01/tele/","unit_of_meas":"A","val_tpl":"{{value_json['ENERGY'].Current}}","dev_cla":"power"} (retained)
13:38:04 MQT: homeassistant/sensor/522851_status/config = {"name":"Tasmota status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"522851_status","device":{"identifiers":["522851"],"connections":[["mac","84:0D:8E:52:28:51"]],"name":"Tasmota","model":"Gosund SP1 v23","sw_version":"7.1.1(tasmota)","manufacturer":"Tasmota"},"~":"tasmota01/tele/"} (retained)
13:38:06 MQT: tasmota01/tele/STATE = {"Time":"2019-12-03T13:38:06","Uptime":"0T00:00:12","UptimeSec":12,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"XefilHQ","BSSId":"98:9B:CB:D2:50:2E","Channel":6,"RSSI":76,"LinkCount":1,"Downtime":"0T00:00:06"}}
13:38:06 MQT: tasmota01/tele/SENSOR = {"Time":"2019-12-03T13:38:06","ENERGY":{"TotalStartTime":"2019-12-02T21:57:44","Total":0.316,"Yesterday":0.000,"Today":0.316,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":214,"Current":0.000}}
13:42:56 MQT: tasmota01/tele/HASS_STATE = {"Version":"7.1.1(tasmota)","BuildDateTime":"2019-12-01T13:00:09","Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Module":"Gosund SP1 v23","RestartReason":"Software/System restart","Uptime":"0T00:05:02","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:06","MqttCount":1,"BootCount":16,"SaveCount":36,"IPAddress":"192.168.3.21","RSSI":"78","LoadAvg":19}
13:43:06 MQT: tasmota01/tele/STATE = {"Time":"2019-12-03T13:43:06","Uptime":"0T00:05:12","UptimeSec":312,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"XefilHQ","BSSId":"98:9B:CB:D2:50:2E","Channel":6,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:06"}}
13:43:06 MQT: tasmota01/tele/SENSOR = {"Time":"2019-12-03T13:43:06","ENERGY":{"TotalStartTime":"2019-12-02T21:57:44","Total":0.316,"Yesterday":0.000,"Today":0.316,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":214,"Current":0.000}}
13:47:54 APP: Serial logging disabled
```

Thanks for the help!

- [DONE ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ DONE] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [DONE ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ DONE] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____ Gosund SP1 v23 Module
- [ ] Tasmota binary firmware version number used: _____ 7.1.1
  - [XX] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Thanks for the help!

Simon

effelle, 2019-12-03:
The discovery prefix is hard-coded on the firmware and the name is of the topic is the one suggested by HA devs. 
If you want to change it you need to compile your own firmware:
```lua
#define USE_HOME_ASSISTANT                       // Enable Home Assistant Discovery Support (+7k code)
  #define HOME_ASSISTANT_DISCOVERY_PREFIX "homeassistant"  // Home Assistant discovery prefix
```
Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for **Tasmota software feature requests and bug reporting**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-12-03:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-03 by ascillato2

xefil, 2019-12-04:
Thanks a lot @effelle for the hint. I've compiled it succesfully and it works.
Now I've another little problem which I'll in case write in another thread.
Bye, simon