arendst/Tasmota #7102: Version 7.1.1(display)-2_6_1 - DisplayMode 0 on DisplayModel 4

itbeyond, 2019-12-04:
### PROBLEM DESCRIPTION
I have been working for a while on Displays and have been successful in making stuff work. I have been mainly working on 6.6.0(release-display) precompiled binary and have been able to make a number of displays work. Today I flashed a WEMOS MINI D1 with a LOLIN TFT-2.4 Shield to the latest 7.7.1(display) - precompiled binary, I was able to make this work in any DisplayMode 1-5 however on DisplayMode 0 I was unable to DisplayText (anything). I flashed back to 6.6.0 and without even a reconfig was able to use DisplayText.

Not currently being a self compiler of this library I am just wondering if there is something missing in the precompiled binary.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
18:52:36 MQT: stat/sonoff-6896/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
18:52:36 MQT: stat/sonoff-6896/RESULT = {"Module":{"18":"Generic"}}
18:52:36 MQT: stat/sonoff-6896/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"68":"Backlight"},"GPIO5":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"67":"SPI DC"},"GPIO16":{"66":"SPI CS"}}
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
18:53:17 MQT: stat/sonoff-6896/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff-6896","ButtonTopic":"","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,3,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:53:17 MQT: stat/sonoff-6896/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-display.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:14","StartupUTC":"2019-12-04T10:47:03","Sleep":50,"CfgHolder":4617,"BootCount":80,"SaveCount":1179,"SaveAddress":"F6000"}}
18:53:17 MQT: stat/sonoff-6896/STATUS2 = {"StatusFWR":{"Version":"7.1.1(display)","BuildDateTime":"2019-12-01T13:03:34","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
18:53:17 MQT: stat/sonoff-6896/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["54HA",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000880B","1405C8000100060000005AFF000000000000","00000800","00000000"]}}
18:53:17 MQT: stat/sonoff-6896/STATUS4 = {"StatusMEM":{"ProgramSize":546,"Free":456,"Heap":27,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FAAA19F","04049FA0","22B617C1","010008C0","00004080"],"Drivers":"1,2,4,5,8,9,10,13,16","Sensors":"1,2,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
18:53:17 MQT: stat/sonoff-6896/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6896","IPAddress":"192.168.100.215","Gateway":"192.168.100.250","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.62","Mac":"84:F3:EB:EE:1A:F0","Webserver":2,"WifiConfig":4}}
18:53:17 MQT: stat/sonoff-6896/STATUS6 = {"StatusMQT":{"MqttHost":"rpis.office.itbeyond.com.au","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EE1AF0","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
18:53:17 MQT: stat/sonoff-6896/STATUS7 = {"StatusTIM":{"UTC":"Wed Dec 04 10:53:17 2019","Local":"Wed Dec 04 18:53:17 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+08:00","Sunrise":"15:25","Sunset":"23:54"}}
18:53:17 MQT: stat/sonoff-6896/STATUS10 = {"StatusSNS":{"Time":"2019-12-04T18:53:17"}}
18:53:17 MQT: stat/sonoff-6896/STATUS11 = {"StatusSTS":{"Time":"2019-12-04T18:53:17","Uptime":"0T00:06:14","UptimeSec":374,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"54HA","BSSId":"F2:9F:C2:F7:17:CB","Channel":11,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:06"}}}

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
18:53:39 CMD: weblog 4
18:53:39 MQT: stat/sonoff-6896/RESULT = {"WebLog":4}
18:53:40 CFG: Saved to flash at F5, Count 1180, Bytes 4096
18:53:48 CMD: displaytext Hello
18:53:48 SRC: WebConsole from 192.168.100.209
18:53:48 CMD: Group 0, Index 1, Command "DISPLAYTEXT", Data "Hello"
18:53:48 MQT: stat/sonoff-6896/RESULT = {"DisplayText":"Hello"}
18:53:51 WIF: Checking connection...
18:53:51 WIF: Connected

18:55:34 CMD: display
18:55:34 SRC: WebConsole from 192.168.100.209
18:55:34 CMD: Group 0, Index 1, Command "DISPLAY", Data ""
18:55:34 MQT: stat/sonoff-6896/RESULT = {"Display":{"Model":4,"Width":240,"Height":320,"Mode":0,"Dimmer":15,"Size":2,"Font":2,"Rotate":0,"Refresh":2,"Cols":[20,8],"Rows":20}}
  ```

### TO REPRODUCE
I simply upgrade from 6.6.0 display precompiled to 7.1.1 display precompilied on a device and when using DisplayText Hello - get nothing on the display.

### EXPECTED BEHAVIOUR
I expected the Hello to appear on the display as it does with 6.6.0

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-12-04:
Have you switched the display on? You have a toggle on / off button in webfrontend.

itbeyond, 2019-12-04:
Yes I have done the process twice going up and down versions and have the display on for sure. As a test I toggle it both ways.

arendst, 2019-12-08:
Try to change color with ``DisplayText [C0]`` first before any other DisplayText commands

Apparently the DisplayText color seem to be swapped...

itbeyond, 2019-12-09:
Theo, Yes thanks that is the problem after the DisplayText [C0] the display is working as expected. I thought I had tried this earlier but I must not have done it right. Thanks for your continued work and efforts.

closed 2019-12-17 by arendst