arendst/Tasmota #7105: Home Assistant integration - retain question

xefil, 2019-12-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Hello!
After recompiling the latest Tasmota version to add a custom homeassistant_prefix (see https://github.com/arendst/Tasmota/issues/7097) I was able to let the devices discovery into Home Assistant.
My goal is to not configure anything in HA, so that the auto discovery and via GUI I'm able to accomplish all.
The problem is that, now, when I turh the relay ON via Home Assistant, the command is sent to the MQTT and Tasmota get is, as follow:

```
13:32:34 MQT: tasmota02/stat/RESULT = {"POWER":"ON"}
13:32:34 MQT: tasmota02/stat/POWER = ON (retained)
```

BTW the switch on Home Assistant side gets immediatly OFF.
I think it's related to retain settings, but, AFAYK, in HA, if nothing is specified, the default *should* be retained off. Those devices are not configured manually, but via auto discovery.

Does it depends on setting on Tasmota side or any suggestions?
I've followed this procedure as well, indicated in the forum, but same result.

```
switchretain off
buttonretain on
buttonretain off
poweronstate 3
powerretain on
```

Ideas?

Thanks!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
Defalt config, expect MQTT auto discovery enabled for Home Assistant.

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
13:36:54 CMD: STATUS 0
13:36:54 MQT: tasmota02/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota02","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
13:36:54 MQT: tasmota02/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T03:57:26","StartupUTC":"2019-12-04T08:39:28","Sleep":50,"CfgHolder":4617,"BootCount":15,"SaveCount":53,"SaveAddress":"F5000"}}
13:36:54 MQT: tasmota02/stat/STATUS2 = {"StatusFWR":{"Version":"7.1.1.1(tasmota)","BuildDateTime":"2019-12-04T09:27:22","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285"}}
13:36:54 MQT: tasmota02/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XefilHQ",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8029","2805C8000100060000005A00000000000000","00000200","00000000"]}}
13:36:54 MQT: tasmota02/stat/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
13:36:54 MQT: tasmota02/stat/STATUS5 = {"StatusNET":{"Hostname":"tasmota02","IPAddress":"192.168.3.22","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.3.1","Mac":"84:0D:8E:52:26:89","Webserver":2,"WifiConfig":2}}
13:36:54 MQT: tasmota02/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.13","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_522689","MqttUser":"mqtt_user","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
13:36:54 MQT: tasmota02/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Dec 04 12:36:54 2019","Local":"Wed Dec 04 13:36:54 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:25","Sunset":"16:54"}}
13:36:54 MQT: tasmota02/stat/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:36:54 MQT: tasmota02/stat/STATUS10 = {"StatusSNS":{"Time":"2019-12-04T13:36:54","ENERGY":{"TotalStartTime":"2019-12-02T22:43:02","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":231,"Current":0.000}}}
13:36:54 MQT: tasmota02/stat/STATUS11 = {"StatusSTS":{"Time":"2019-12-04T13:36:54","Uptime":"0T03:57:26","UptimeSec":14246,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":{"STATE":"ON"},"Wifi":{"AP":1,"SSId":"MYSSID","BSSId":"E0:28:6D:7B:CE:48","Channel":11,"RSSI":72,"LinkCount":1,"Downtime":"0T00:00:08"}}}

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2019-12-04:
No, there is something wrong on your HA configuration or you mess up with PowerRetain (must be `0`), because state change is not retained normally:
```json
13:49:52 MQT: *edited*/stat/SWITCH2 = {"STATE":"TOGGLE"}
13:49:52 MQT: *edited*/stat/RESULT = {"POWER":"OFF"}
13:49:52 MQT: *edited*/stat/POWER = OFF
```
If after changing PowerRetain you still see a `(retained)` message means you have some messages stuck on your broker and you need to delete them. Have a look [here](https://github.com/arendst/Tasmota/wiki/PowerOnState-Configuration) on documentation
> BTW the switch on Home Assistant side gets immediatly OFF.

Means is a button, not a switch. Have a look on [documentation](https://tasmota.github.io/docs/#/integrations/Home-Assistant?id=automatic-discovery) chapter about button and switches discovery. 

Thanks.

Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato, 2019-12-04:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-04 by ascillato2

xefil, 2019-12-09:
Hello @ascillato and @effelle 
I've some news about the "non working" MQTT on Tasmota 7.1 and the working version on 6.7.1
I was able to recompile the 6.7.1 (due I'm having custom homeassistant_discovery_prefix) and found the difference.
As you remember, on 7.1 the switch was turned ON and immediate after turned OFF. Now I've the debug showing me the command I get more on 6.7 version.
7.1 version of Tasmota on home assistant log:
```
DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on tasmota02/cmnd/POWER: ON
DEBUG (MainThread) [homeassistant.components.mqtt] Received message on tasmota02/tele/STATE: b'{"Time":"2019-12-05T13:28:47","Uptime":"1T03:49:19","UptimeSec":100159,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":14,"POWER":{"STATE":"ON"},"Wifi":{"AP":1,"SSId":"XefilHQ","BSSId":"E0:28:6D:7B:CE:48","Channel":11,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:08"}}'
```
6.7 version of Tasmota SonOff on homeassistant log:
```
2019-12-09 13:08:58 DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on tasmota02/cmnd/POWER: ON
DEBUG (MainThread) [homeassistant.components.mqtt] Received message on tasmota02/tele/STATE: b'{"Time":"2019-12-09T13:08:58","Uptime":"0T00:03:46","UptimeSec":226,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"ON","Wifi":{"AP":1,"SSId":"XefilHQ","BSSId":"E0:28:6D:7B:CE:48","Channel":11,"RSSI":76,"LinkCount":1,"Downtime":"0T00:00:05"}}'
DEBUG (MainThread) [homeassistant.components.mqtt] Received message on tasmota02/stat/RESULT: b'{"POWER":"ON"}'
WARNING (MainThread) [homeassistant.components.mqtt.binary_sensor] No matching payload found for entity: Tasmota Button1 with state topic: tasmota02/stat/RESULT. Payload: , with value template Template("{{value_json.BUTTON1}}")
```

On 6.7 version I'm getting a /stat/RESULT more, what on 7.1 is not sent. Now, is that expected? Is something I need to add on 7.1 as option ?

I've written on the CHAT as well, but posted here too, to understand if that's an expected behaviour or not.

At least we have the cause, I think

Thanks!!

xefil, 2019-12-09:
Ok, as in chat, solved with deve version, great!
Thanks to all!!

xefil, 2019-12-09:
@ascillato  and @effelle an update from Home Assistant, pointing us to a problem in the template. Read here:

https://community.home-assistant.io/t/tasmota-and-retained-messages/152639/11

Cheers, Simon

ascillato, 2019-12-09:
Hi, Thanks for sharing that article. That template POWER:{STATE:ON} was reverted. It is no used any longer. It was in the repo few hours and we reverted back due it was breaking backwards compatibility with several other tools. So, just update to latest Tasmota from the development branch to solve your issue. Thanks

xefil, 2019-12-10:
Thanks to you for the super-fast update and support ;-)