arendst/Tasmota #7110: Log character length limit for rule sets

meingraham, 2019-12-05:
**Have you looked for this feature in other issues and in the wiki?**  
N/A

**Is your feature request related to a problem? Please describe.**  
At times. When using a long set of rule in a rule set, one is unable to confirm that rules at the tail end are stored properly in the rule set because the resulting log is truncated. Also, any time one uses `Rule<x>` to inspect the rules, it is not possible to retrieve the full rule set.

**Describe the solution you'd like**  
Extend the character limit for log messages (or at least for these) so that all of the preamble fields plus the full rule set can be output in the log.

**Describe alternatives you've considered**  
Use decode_config.py. But for troubleshooting this is very cumbersome.

**Additional context**  
N/A

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-12-05:
How about using the console feature of TDM. This uses MQTT and should report the complete information.

meingraham, 2019-12-05:
So, usually when I post a feature request, typically it is not for something I'm implementing. Even my infamous "absolute PowerDelta" ;-) Ditto consolidation of SerialSend. This request is based on interactions with users where I found that the truncation of the rule impacted the troubleshooting.

Yes, TDM is another alternative. Nevertheless, using TDM (whether an executable, if available, or the installation of the required packages), would also be a tedious workaround... although admittedly probably less so than using decode_config.

arendst, 2019-12-05:
The current logsize is set to 520 characters. A moderate tradeoff between readable output and global memory use.

Looking at the rule will result in a text like:
```
16:05:22 MQT: stat/wemos4/RULE = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":0,"Rules":"12345678901any many more ending in ...
```
The max rule length is 511 characters. The loglength should cater for these 511 characters in addition to the time, topic and other rule information. As the topic can grow to 100 characters a nice logbuffer length would end up at around 185 + 511 = 696 characers.

So I'll increment log size to 700 characters to accomodate full rule logging ;-)

ascillato2, 2019-12-06:
Closing this issue as it has been added. Thanks.

closed 2019-12-06 by ascillato2