arendst/Tasmota #7113: implement option to present WiFi RSSI in dbm instead of percent

RoadRunnr, 2019-12-05:
**Have you looked for this feature in other issues and in the wiki?**  

Yes,

**Is your feature request related to a problem? Please describe.**  

Having the RSSI in percent instead of dbm is user friendly, but not very precise.

**Describe the solution you'd like**  

A SetOption to choose between the current percent presentation and dbm.

**(Please, remember to close the issue when the problem has been addressed)**  

I would implement this myself and send a PR, just wanting to know if this has a chance of being accepted before I invest the time.

ascillato, 2019-12-05:
Hi,

Dupplicated from https://github.com/arendst/Tasmota/issues/3532

andrethomas, 2019-12-05:
I think it should stay as % for the sake of simplicity to users.

meingraham, 2019-12-05:
It can stay as % as default... but allow users to change it with a SetOption if they desire? This way most users retain the current output but those that want dBm can change the setting?

Jason2866, 2019-12-05:
I would like to have dbm too. Should be possible https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/station-class.html#rssi

ascillato, 2019-12-05:
@Jason2866 here is the actual code in Tasmota:

https://github.com/arendst/Tasmota/blob/b347c53846c84f17f246ab25cb373c87a0c747b6/tasmota/support_wifi.ino#L55-L67

jziolkowski, 2019-12-06:
Instead of adding a SetOption to select one over another, IMO it's better to have RSSI always displayed and use a SetOption to optionally add dBm value as well. RSSI (while simplified) is a very useful on-glance quick diagnose wifi metric. Forcing external tools to poll the device for something so trivial is - IMO again - a bit overkill.

RoadRunnr, 2019-12-06:
> Instead of adding a SetOption to select one over another, IMO it's better to have RSSI always displayed and use a SetOption to optionally add dBm value as well.

My plan was to also switch the value of the RSSI in the MQTT topic when the SetOption is in use. Displaying both on the WebUI is fine, but adding another node to the MQTT json is IMHO overkill.

Anyhow, after looking a bit more at the RSSI code I've decided that the changes are going to be quite simple. So, I'm going to just build it over the weekend, send a PR and then we can discuss based on an actual implementation.

jziolkowski, 2019-12-06:
But in any scenario, RSSI won't be reported as -dBm?

Jason2866, 2019-12-06:
@jziolkowski  It is in dBm if the conversion is NOT done Adrian showed.
I deleted the conversion and i get `"RSSI dBm":-51`

jziolkowski, 2019-12-06:
RSSI dBm does not make sense, because RSSI by design is a unitless value. You have either RSSI, or dBm. Using "RSSI dBm" is like saying "percent Celsius"

Jason2866, 2019-12-06:
Whatever you want :-) Just edit the text in that line

jziolkowski, 2019-12-06:
The point is not how that node is named. What I'm saying is: keep RSSI 0-100. If user wants to display dBm value, don't present it as RSSI, because it is not.

Jason2866, 2019-12-06:
It is named correctly if you change the line. There is NO RSSI anymore...
`10:01:17 MQT: tele/sonoff-53CBF8/STATE = {"Time":"2019-12-06T10:01:17","Uptime":"0T00:00:12","UptimeSec":12,"Vcc":3.041,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"Wifi dBm":-54,"LinkCount":1,"Downtime":"0T00:00:06"}}`

jziolkowski, 2019-12-06:
dBm is enough, as it's a sub-dictionary of Wifi anyway :) However I still prefer having RSSI in place and dBm as an additional node, enabled with setoption

andrethomas, 2019-12-06:
This thread is indeed entertaining...

closed 2019-12-08 by arendst

jziolkowski, 2019-12-08:
RSSI and dBm, best of both worlds. +1

andrethomas, 2019-12-08:
Yup, now we can add all the rounding errors and find the missing wifi waves...

PR #7145 (2019-12-08): report signal strength in dBm over MQTT
## Description:

* report signal strength in dBm over MQTT
* include dBm in debug log and in WebIf information display.

**Related issue (if applicable):** fixes #7113

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development [... cut]