arendst/Tasmota #7117: Power Status LED does not switch when switching relay2 of a double socket

rqsnst, 2019-12-05:
### PROBLEM DESCRIPTION
i configured a Teckin SS42 with the template from the database but that did not work, so i
figured out a own template (see below)
the problem is, that if i switch the relay2 on the power status LED1 does not swicth on
i tried some things to switch the configuration of the LED1 and LEDlink as advised in the discord chat, but nothing worked. The LEDmask is default (0xffff)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Teckin SS42
- [ X] Tasmota binary firmware version number used: 7.1.1.1
  - [X ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X ] Flashing tools used: raspberry OTA Tuya device
- [X ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```22:20:54 MQT: stat/Aussen-Teckin-SS42/RESULT = {"NAME":"Teckin SS42","GPIO":[255,255,255,255,56,158,0,0,21,17,22,255,255],"FLAG":15,"BASE":18}
22:20:54 MQT: stat/Aussen-Teckin-SS42/RESULT = {"Module":{"0":"Teckin SS42"}}
22:20:54 MQT: stat/Aussen-Teckin-SS42/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"56":"Led1i"},"GPIO5":{"158":"LedLinki"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"17":"Button1"},"GPIO14":{"22":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
  Configuration output here:
ledstate 1

```
- [x ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
22:19:44 CMD: status 0
22:19:44 MQT: stat/Aussen-Teckin-SS42/STATUS = {"Status":{"Module":0,"FriendlyName":["Aussen1","Aussen2"],"Topic":"Aussen-Teckin-SS42","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:19:44 MQT: stat/Aussen-Teckin-SS42/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"https://thehackbox.org/tasmota/tasmota-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:16:00","StartupUTC":"2019-12-05T21:03:44","Sleep":50,"CfgHolder":4617,"BootCount":76,"SaveCount":571,"SaveAddress":"FA000"}}
22:19:44 MQT: stat/Aussen-Teckin-SS42/STATUS2 = {"StatusFWR":{"Version":"7.1.1.1(e51cefe-tasmota)","BuildDateTime":"2019.12.04 23:02:47","Boot":7,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285"}}
22:19:44 MQT: stat/Aussen-Teckin-SS42/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IR1","IR1"],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000008","00000000"]}}
22:19:44 MQT: stat/Aussen-Teckin-SS42/STATUS4 = {"StatusMEM":{"ProgramSize":563,"Free":440,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
22:19:44 MQT: stat/Aussen-Teckin-SS42/STATUS5 = {"StatusNET":{"Hostname":"SS42-1","IPAddress":"192.168.178.90","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.173","Mac":"D8:F1:5B:CB:26:1C","Webserver":2,"WifiConfig":2}}
22:19:44 MQT: stat/Aussen-Teckin-SS42/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.173","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CB261C","MqttUser":"homemqtt","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:19:45 MQT: stat/Aussen-Teckin-SS42/STATUS7 = {"StatusTIM":{"UTC":"Thu Dec 05 21:19:44 2019","Local":"Thu Dec 05 22:19:44 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"08:09","Sunset":"16:24"}}
22:19:45 MQT: stat/Aussen-Teckin-SS42/STATUS10 = {"StatusSNS":{"Time":"2019-12-05T22:19:45"}}
22:19:45 MQT: stat/Aussen-Teckin-SS42/STATUS11 = {"StatusSTS":{"Time":"2019-12-05T22:19:45","Uptime":"0T00:16:01","UptimeSec":961,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":{"STATE":"OFF"},"POWER2":{"STATE":"OFF"},"Wifi":{"AP":1,"SSId":"IR1","BSSId":"38:10:D5:2E:74:84","Channel":1,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
can some check this with a device with more than one relay ?

### EXPECTED BEHAVIOUR
i want that the power status LED goes on if i swicht relay 1 or relay 2

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

rqsnst, 2019-12-05:
i went on testing and found a solution on my own:

{"NAME":"Teckin SS42","GPIO":[255,255,255,255,56,57,0,0,21,17,22,255,255],"FLAG":15,"BASE":18}

rqsnst, 2019-12-06:
one point i forget and diden't solve,
with this template:
> {"NAME":"Teckin SS42","GPIO":[255,255,255,255,56,57,0,0,21,17,22,255,255],"FLAG":15,"BASE":18}

I now have the following behavior:
When I start the device, the blue LED (GPIO4) flashes until the WLAN / MQTT connection is established.
if I turn relay1 or 2 on, the green LED (GPIO5) will light up in both cases

How can I configure the blue LED (GPIO4) to indicate the connection status (WLAN / MQTT) ** and ** the current status of relay2
and the green LED (GPIO5) indicates the current status of relay1?

ascillato, 2019-12-06:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-06 by ascillato2

rqsnst, 2019-12-06:
in the chat the moderator said, that my wished configuration ist not possible, maybe then it's a feature request ;-)