arendst/Tasmota #7120: Missing OTA upgrade path from v6.4 to 7.1.1

lobocobra, 2019-12-06:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have 40+ shelly1 installed in the wall and most of them have the v6.4 loaded, which require 510kb.
=> when you want to upgrade them to the latest version you normally need to...
1) reset 5 (clear settings)
2) upload 7.1.1 minimal firmware
3) upload 7.1.1 firmware
Reset 5 works well, but after the upload of the minimal firmware shelly1 crashes (light can not be switched any longer manually nor by interface). 
=> The device is not pingable after the minimal upgrade and does also not offer any web-interface or AP connection. 
=> Only solution is to take the device out of the wall and to erase the flash and to start from scratch by serial.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): shelly1
- [x] Tasmota binary firmware version number used: 6.4
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
   not possible as the device is not responsive any longer and installed in the wall
  Configuration output here:
`13:42:52 MQT: stat/y_ug_heizung_licht/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
13:42:53 MQT: stat/y_ug_heizung_licht/RESULT = {"Module":{"46":"Shelly 1"}}
13:42:53 MQT: stat/y_ug_heizung_licht/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"82":"Switch1n"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}`

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
no rules used
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
13:44:32 MQT: stat/y_ug_heizung_licht/STATUS = {"Status":{"Module":46,"FriendlyName":["y_ug_heizung_licht"],"Topic":"y_ug_heizung_licht","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:44:32 MQT: stat/y_ug_heizung_licht/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:12:21","StartupUTC":"2019-12-06T12:32:11","Sleep":50,"CfgHolder":4617,"BootCount":7,"SaveCount":24,"SaveAddress":"FA000"}}
13:44:32 MQT: stat/y_ug_heizung_licht/STATUS2 = {"StatusFWR":{"Version":"7.1.1(tasmota)","BuildDateTime":"2019-12-05T17:11:33","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
13:44:33 MQT: stat/y_ug_heizung_licht/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["domaincats",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","000000C0","00000000"]}}
13:44:33 MQT: stat/y_ug_heizung_licht/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":27,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","8F80A187","040403A0","200217CD","010013C0","00004880"],"Drivers":"1,2,3,4,8,9,10,29","Sensors":"1,2,3,5,6,7,8,9,10,14,26"}}
13:44:33 MQT: stat/y_ug_heizung_licht/STATUS5 = {"StatusNET":{"Hostname":"y_ug_heizung_licht-1993","IPAddress":"192.168.20.126","Gateway":"192.168.20.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.20.2","Mac":"3C:71:BF:2B:E7:C9","Webserver":2,"WifiConfig":5}}
13:44:33 MQT: stat/y_ug_heizung_licht/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.20.90","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2BE7C9","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
13:44:33 MQT: stat/y_ug_heizung_licht/STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 06 12:44:33 2019","Local":"Fri Dec 06 13:44:33 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:57","Sunset":"16:35"}}
13:44:33 MQT: stat/y_ug_heizung_licht/STATUS10 = {"StatusSNS":{"Time":"2019-12-06T13:44:33","Switch1":"OFF"}}
13:44:33 MQT: stat/y_ug_heizung_licht/STATUS11 = {"StatusSTS":{"Time":"2019-12-06T13:44:33","Uptime":"0T00:12:22","UptimeSec":742,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"domaincats","BSSId":"14:CC:20:AC:2D:4D","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
 not possible, the device becomes unresponsible after the successfull upload of minimal firmware

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- have a shelly1 with firmware 6.4
- in console: reset 5 in order to clear memory
- upload by web the minimal firmware as the original 6.4 version needs 510kb (no direct upload possible of 7.1.1 normal build)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
An OTA ugrade path from v6.4 / 6.5 to the latest 7.1.1 needs to be ensured.
=> if devices like shelly 1 are installed in the wall, an upgrade needs to be possible by OTA
=> this worked always fine till version 6.6

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
n/a

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

lobocobra, 2019-12-06:
In order to make re-usable the shelly1 I need to do (was needed on 3 devices so far)
- erase shelly with esptool
- load over serial v 7.1.1
- connect to device over serial and set manually ssid and password

I do not understand why no AP is available or Webserver to change the domain info differently. I have never seen such behavior before. 
=> once done I can load minimal firmware or regular firmware as usual and have no more problems.

The real issue is that OTA upgrade becomes completely unreliable and it is not practicable now and for the future to unmount shelly from the wall each time an update is needed.

lobocobra, 2019-12-06:
#5239 shows the source of the problem. The minimal build crashes the devices.
=> either you are stuck in the minimal setup
=> or the device is not reacting any longer (also not on button push)

Can you please fix the minimal build? It is the same outcome if you load the compiled one or compile it yourself.

arendst, 2019-12-06:
Just for the fun I installed 6.4.1 on a wemos and tried to upgrade to 7.1.2. It succeeds as expected.

```
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff Version 6.4.1(release-sonoff)-2_3_0
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 indebuurt in mode 11N as wemos4-6928...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on wemos4-6928.local with IP address 192.168.2.223
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/wemos4/LWT = Online (retained)
00:00:05 MQT: cmnd/wemos4/POWER = 
00:00:05 MQT: Subscribe to cmnd/wemos4/#
00:00:05 MQT: Subscribe to cmnd/sonoffs/#
00:00:05 MQT: Subscribe to cmnd/DVES_83BB10_fb/#
00:00:05 MQT: tele/wemos4/INFO1 = {"Module":"Shelly 1","Version":"6.4.1(release-sonoff)","FallbackTopic":"cmnd/DVES_83BB10_fb/","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/wemos4/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos4-6928","IPAddress":"192.168.2.223"}
00:00:05 MQT: tele/wemos4/INFO3 = {"RestartReason":"Software/System restart"}
00:00:05 MQT: stat/wemos4/RESULT = {"POWER":"OFF"}
00:00:05 MQT: stat/wemos4/POWER = OFF
00:00:06 APP: (UTC) Fri Dec 06 15:40:43 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
16:40:47 APP: Boot Count 5
16:40:47 CFG: Saved to flash at F8, Count 12, Bytes 3584
16:40:51 MQT: tele/wemos4/STATE = {"Time":"2019-12-06T16:40:51","Uptime":"0T00:00:14","Vcc":2.726,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"POWER":"OFF","Wifi":{"AP":1,"SSId":"indebuurt","BSSId":"18:E8:29:CA:17:C1","Channel":9,"RSSI":100}}
16:40:51 MQT: tele/wemos4/SENSOR = {"Time":"2019-12-06T16:40:51","Switch1":"ON"}
16:40:53 CMD: status 0
16:40:53 SRC: Serial
16:40:53 RSL: Group 0, Index 1, Command STATUS, Data 0
16:40:53 MQT: stat/wemos4/STATUS = {"Status":{"Module":46,"FriendlyName":["Sonoff"],"Topic":"wemos4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:40:53 MQT: stat/wemos4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:16","StartupUTC":"2019-12-06T15:40:37","Sleep":50,"BootCount":5,"SaveCount":12,"SaveAddress":"F8000"}}
16:40:53 MQT: stat/wemos4/STATUS2 = {"StatusFWR":{"Version":"6.4.1(release-sonoff)","BuildDateTime":"2018-12-24T15:51:34","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:40:53 MQT: stat/wemos4/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["indebuurt",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
16:40:53 MQT: stat/wemos4/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
16:40:53 MQT: stat/wemos4/STATUS5 = {"StatusNET":{"Hostname":"wemos4-6928","IPAddress":"192.168.2.223","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"5C:CF:7F:83:BB:10","Webserver":2,"WifiConfig":4}}
16:40:53 MQT: stat/wemos4/STATUS6 = {"StatusMQT":{"MqttHost":"domus1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_83BB10","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:40:53 MQT: stat/wemos4/STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 06 15:40:53 2019","Local":"Fri Dec 06 16:40:53 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:28","Sunset":"16:54"}}
16:40:53 MQT: stat/wemos4/STATUS10 = {"StatusSNS":{"Time":"2019-12-06T16:40:53","Switch1":"ON"}}
16:40:53 MQT: stat/wemos4/STATUS11 = {"StatusSTS":{"Time":"2019-12-06T16:40:53","Uptime":"0T00:00:16","Vcc":2.726,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"indebuurt","BSSId":"18:E8:29:CA:17:C1","Channel":9,"RSSI":100}}}
16:41:06 CMD: otaurl
16:41:06 SRC: Serial
16:41:06 RSL: Group 0, Index 1, Command OTAURL, Data 
16:41:06 MQT: stat/wemos4/RESULT = {"OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin"}
16:41:15 CMD: otaurl http://thehackbox.org/tasmota/release/tasmota.bin
16:41:15 SRC: Serial
16:41:15 RSL: Group 0, Index 1, Command OTAURL, Data http://thehackbox.org/tasmota/release/tasmota.bin
16:41:15 MQT: stat/wemos4/RESULT = {"OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin"}
16:41:15 CFG: Saved to flash at F7, Count 13, Bytes 3584
16:41:23 CMD: reset 5
16:41:23 SRC: Serial
16:41:23 RSL: Group 0, Index 1, Command RESET, Data 5
16:41:23 MQT: stat/wemos4/RESULT = {"Reset":"Erase, Reset and Restarting"}
16:41:24 APP: Erase 898 sectors
16:41:24 CFG: Use defaults
16:41:31 CFG: Saved to flash at FB, Count 1, Bytes 3584
16:41:31 CFG: Saved to flash at FA, Count 2, Bytes 3584
16:41:31 CFG: Saved to flash at F9, Count 3, Bytes 3584
16:41:31 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
[... 91 lines cut ...]
00:00:00 Project tasmota Sonoff Version 7.1.2(tasmota)-2_6_1
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 indebuurt in mode 11N as sonoff-6928...
[... cut ...]
16:43:50 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 06 15:43:50 2019","Local":"Fri Dec 06 16:43:50 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:28","Sunset":"16:54"}}
16:43:50 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-12-06T16:43:50"}}
16:43:50 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-12-06T16:43:50","Uptime":"0T00:00:17","UptimeSec":17,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"indebuurt","BSSId":"18:E8:29:CA:17:C1","Channel":9,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```

lobocobra, 2019-12-06:
hmm that is strange.... you use shelly1 and everything looks the same.
Maybe the trick is to downgrade first to 6.4.1? 
I will try.

The only work around for me is.... and I know this sounds silly....
- 6.4.x.x reset 5
- reboot 
- sonoff minimal http://thehackbox.org/tasmota/tasmota-minimal.bin
   => from here the firmware is stuck and you can not upgrade anymore
- build sonoff minimal from latest git (if I use this directly, shelly not responsive)
- after that any firmware can be loaded w/o problem.

arendst, 2019-12-06:
After reboot do not load tasmota-minima.binl! Load tasmota.bin. This will auto load minimal if needed as shown in the log above.

lobocobra, 2019-12-06:
argh.....
found a shelly1 with Program Version 6.4.1.15(sonoff) and felt curageous....
- reset 5 (all good)
- reboot (all good)
- loaded directly the desired 7.1.1 master (Flash Size	2048kB / Program Flash Size	2048kB)

aaaand.... ended up with un-responsive shelly Nr6. The next one I find I will try with DEV. Maybe this works. But problem is that some need smaller than 500kb firmware to load it by OTA. 
lol

arendst, 2019-12-06:
You cannot ota or upload a binary compiled with a different memory model. This results in what you experience.

I suggest you try to use a wemos with 6.4 loaded and perform an ota upload with tasmota.bin

lobocobra, 2019-12-06:
Thanks for the hint. This must be the issue.
=> I close it here as the problem comes from using different memory models

The work - around I posted works for those with the same issue.

closed 2019-12-06 by lobocobra