arendst/Tasmota #7121: EXS-Dimmer (72) missing from modules and templates

jonathanschneider, 2019-12-06:
### PROBLEM DESCRIPTION
Since v7.1.x the supported module "EXS-Dimmer (72)" is missing from the drop-down menus for modules and templates on the configuration page in the release binaries. The module is still listed as supported and was included in v6.7.1.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): EXS-Dimmer (72)
- [x] Tasmota binary firmware version number used: 7.1.0 & 7.1.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
  Irrelevant

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
  Irrelevant

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
  Irrelevant

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
  Irrelevant

```

### TO REPRODUCE
Go to "Configuration > Configure Module" and try to select "EXS-Dimmer (72).
Go to "Configuration > Configure Other" and try to select "EXS-Dimmer (72).

### EXPECTED BEHAVIOUR
"EXS-Dimmer (72)" included in both drop-down menus.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-06:
Please, use the corresponding template for your device. Thanks

arendst, 2019-12-06:
Currently the EXS dimmer is only supported in the tasmota-sensor.bin

jonathanschneider, 2019-12-06:
@ascillato like I wrote, the corresponding template is missing from the list and just recreating the template would not be enough to make this module work.

@arendst thanks, I will try that binary then. Before v7.1.x a table of the binaries and the supported features was included on the release page. Where would I find that information now?

closed 2019-12-06 by jonathanschneider

arendst, 2019-12-06:
In the same releasenotes but now as a link somewhere in the middle.

jonathanschneider, 2019-12-06:
@arendst Found it. Thanks. The "x" for "USE_EXS_DIMMER" should then be in the column "sensors" and not "tasmota" and "knx", correct?

arendst, 2019-12-06:
Correct.