arendst/Tasmota #7122: WS21812 does not turn off when using a Scheme > 5

reimerp, 2019-12-06:
### PROBLEM DESCRIPTION
WS21812 does not turn off when using a Scheme > 5
Using ```Scheme 1``` turn on the device, ```Power OFF``` turns it off,
but
using ```Scheme 6```enables animation, ```Power OFF``` stops the animation but does NOT power off the device. It was working in older versions.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP8266 with WS2812 Module
- [x] Tasmota binary firmware version number used: 7.1.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
{"NAME":"RGB","GPIO":[255,56,7,255,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
{"Module":{"0":"RGB"}}
{"GPIO0":{"0":"None"},"GPIO1":{"56":"Led1i"},"GPIO2":{"7":"WS2812"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
STATUS = {"Status":{"Module":0,"FriendlyName":["SonoffRGB"],"Topic":"sonoff/rgb","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":1}}
STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"groups/wohnzimmer","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:22:54","StartupUTC":"2019-12-06T17:21:24","Sleep":50,"CfgHolder":4617,"BootCount":5,"SaveCount":28,"SaveAddress":"F8000"}}
STATUS2 = {"StatusFWR":{"Version":"7.1.1(tasmota)","BuildDateTime":"2019-12-01T13:00:09","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"hassbian","LogPort":514,"SSId":["WehLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8228","2805C8000100060000005A00000000000000","00000201","00000000"]}}
STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14400B","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
STATUS5 = {"StatusNET":{"Hostname":"sonoff-rgb","IPAddress":"192.168.176.25","Gateway":"192.168.176.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.176.1","Mac":"84:0D:8E:A4:93:6F","Webserver":2,"WifiConfig":2}}
STATUS6 = {"StatusMQT":{"MqttHost":"hassbian","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A4936F","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 06 17:44:18 2019","Local":"Fri Dec 06 18:44:18 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"08:19","Sunset":"16:01"}}
STATUS10 = {"StatusSNS":{"Time":"2019-12-06T18:44:18"}}
STATUS11 = {"StatusSTS":{"Time":"2019-12-06T18:44:18","Uptime":"0T00:22:54","UptimeSec":1374,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":50,"Color":"128,128,128","HSBColor":"0,0,50","Channel":[50,50,50],"Scheme":6,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"WehLAN","BSSId":"CC:B2:55:F3:4E:54","Channel":6,"RSSI":92,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Backlog; Scheme 6; Power OFF

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The device should power off.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-06:
Hi, this has been fixed in latest version of the development branch. Please update and re do your tests. Thanks

arendst, 2019-12-06:
That's why v7.1.2 was released today!

ascillato2, 2019-12-06:
Closing this issue as it is **duplicated**. Sorry. See https://github.com/arendst/Tasmota/issues/7090

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-06 by ascillato2

reimerp, 2019-12-06:
worls in 7,1,1,2, Thanks a lot.