arendst/Tasmota #7123: KNX binary contains Domoticz instead of KNX

ctr49, 2019-12-06:
### PROBLEM DESCRIPTION
The KNX binary contains Domoticz module instead of KNX module

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Nedis WIFIP110
- [x] Tasmota binary firmware version number used: 7.1.2
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
22:06:14 RSL: stat/tasmota/RESULT = {"NAME":"Nedis WIFIP110","GPIO":[17,0,0,0,134,132,0,0,131,56,21,0,0],"FLAG":0,"BASE":49}
22:06:14 RSL: stat/tasmota/RESULT = {"Module":{"0":"Nedis WIFIP110"}}
22:06:14 RSL: stat/tasmota/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"134":"BL0937 CF"},"GPIO5":{"132":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"56":"Led1i"},"GPIO14":{"21":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
22:07:07 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:07:07 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"https://github.com/arendst/Tasmota/releases/download/v7.1.2/tasmota-knx.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:05","StartupUTC":"2019-12-06T20:56:02","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":27,"SaveAddress":"FA000"}}
22:07:07 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019-12-06T11:22:22","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
22:07:07 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["redacted",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
22:07:07 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":0,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
22:07:07 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-redacted","IPAddress":"192.168.redacted","Gateway":"192.168.redacted","Subnetmask":"255.255.255.0","DNSServer":"192.168.redacted","Mac":"2C:F4:32:redacted","Webserver":2,"WifiConfig":2}}
22:07:07 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_redacted","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:07:07 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 06 21:07:07 2019","Local":"Fri Dec 06 22:07:07 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:28","Sunset":"16:54"}}
22:07:07 RSL: stat/tasmota/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
22:07:07 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2019-12-06T22:07:07","ENERGY":{"TotalStartTime":"2019-12-06T21:28:14","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
22:07:07 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2019-12-06T22:07:07","Uptime":"0T00:11:05","UptimeSec":665,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"redacted","BSSId":"24:A4:3C:redacted","Channel":11,"RSSI":56,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Load tasmota-knx.bin
Try to configure KNX

### EXPECTED BEHAVIOUR
A KNX Menu should appear

### SCREENSHOTS
n/a

### ADDITIONAL CONTEXT
n/a

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-12-07:
Hi,

Your device was not updating. In your Status 0, it says: `"Version":"7.1.2(tasmota)"` and should say: `"Version":"7.1.2.2(00fd7f2-knx)" if updating to latest`

Your OTAurl is wrong ( https://github.com/arendst/Tasmota/releases/download/v7.1.2/tasmota-knx.bin ) It can't be HTTPS. It requires to be a site with HTTP as the provided in the readme file ( http://thehackbox.org/tasmota/tasmota-knx.bin )

Please, change that OTAurl to http://thehackbox.org/tasmota/tasmota-knx.bin and try again.

I just try that and it works as expected:

[image]

closed 2019-12-07 by ascillato2

ctr49, 2019-12-07:
That's right. Apparently it never upgraded, but I already tried the thehackbox URL before and only went to the GitHub release page to double check. It seems I was just confused by Domoticz because I did not see it in the version before.
Nevertheless, I had to try four more times until the upgrade actually occured, never seeing an error message when it did not work.

Thanks for pointing me to the incomplete upgrade!

ascillato, 2019-12-07:
When using OTAUrl, Tasmota:

* downloads the minimal firmware (10-20 seconds depending on your internet connection),
* then it does the update on the flash (10 seconds approx and restarts)
* the it downloads the selected firmware (10-20 seconds depending on your internet connection),
* then it does the update on the flash (10 seconds approx and restarts)

So all the process takes some time to be performed. If in the middle you reboot the device, the Update process will be interrupted.