arendst/Tasmota #7125: Autoconfigured device state not recognised by Home Assistant

znanev, 2019-12-06:
### PROBLEM DESCRIPTION
Versions build from code from commit 45d5999 or later don't correctly report the POWER state when using autoconfigured MQTT devices in Home Assistant.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): BlitzWolf SHP2, Sonoff S20 (not device dependent)
- [x] Tasmota binary firmware version number used: 7.1.1.1(45d5999-tasmota) and later
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
22:03:14 CMD: Backlog Template; Module; GPIO 255
22:03:14 MQT: blitzwolf-shp2-1/stat/RESULT = {"NAME":"BlitzWolf SHP","GPIO":[57,255,56,255,0,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":45}
22:03:14 MQT: blitzwolf-shp2-1/stat/RESULT = {"Module":{"45":"BlitzWolf SHP"}}
22:03:14 MQT: blitzwolf-shp2-1/stat/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"None"},"GPIO2":{"158":"LedLinki"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"134":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"17":"Button1"},"GPIO14":{"132":"HLWBL CF1"},"GPIO15":{"21":"Relay1"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  No rules used / not relevant for this issue
```
- [x] Provide the output of this command: `Status 0`:
```
22:10:11 CMD: STATUS 0
22:10:11 MQT: blitzwolf-shp2-1/stat/STATUS = {"Status":{"Module":45,"FriendlyName":["MasterPlug"],"Topic":"blitzwolf-shp2-1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:10:11 MQT: blitzwolf-shp2-1/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T03:31:30","StartupUTC":"2019-12-06T18:38:41","Sleep":50,"CfgHolder":4617,"BootCount":69,"SaveCount":1167,"SaveAddress":"F7000"}}
22:10:11 MQT: blitzwolf-shp2-1/stat/STATUS2 = {"StatusFWR":{"Version":"7.1.1.1(45d5999-tasmota)","BuildDateTime":"2019-12-02T21:00:10","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
22:10:11 MQT: blitzwolf-shp2-1/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["<MySSID>",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["400A8009","2805C8000100060000005AFF000000170700","00000228","00000000"]}}
22:10:11 MQT: blitzwolf-shp2-1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440A1","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
22:10:11 MQT: blitzwolf-shp2-1/stat/STATUS5 = {"StatusNET":{"Hostname":"blitzwolf-shp2-1-4627","IPAddress":"192.168.120.147","Gateway":"192.168.120.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.120.1","Mac":"DC:4F:22:33:B2:13","Webserver":2,"WifiConfig":5}}
22:10:11 MQT: blitzwolf-shp2-1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.120.111","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_33B213","MqttUser":"","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:10:11 MQT: blitzwolf-shp2-1/stat/STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 06 22:10:11 2019","Local":"Fri Dec 06 22:10:11 2019","StartDST":"Sun Mar 31 01:00:00 2019","EndDST":"Sun Oct 27 02:00:00 2019","Timezone":99,"Sunrise":"07:51","Sunset":"15:55"}}
22:10:11 MQT: blitzwolf-shp2-1/stat/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
22:10:11 MQT: blitzwolf-shp2-1/stat/STATUS10 = {"StatusSNS":{"Time":"2019-12-06T22:10:11","ENERGY":{"TotalStartTime":"2019-06-26T21:15:40","Total":[6.458,0.371,4.150],"Yesterday":0.063,"Today":0.045,"Power":13,"ApparentPower":20,"ReactivePower":16,"Factor":0.63,"Voltage":245,"Current":0.082}}}
22:10:11 MQT: blitzwolf-shp2-1/stat/STATUS11 = {"StatusSTS":{"Time":"2019-12-06T22:10:11","Uptime":"0T03:31:30","UptimeSec":12690,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":{"STATE":"ON"},"Wifi":{"AP":1,"SSId":"<MySSID>","BSSId":"<MyAPMac>","Channel":4,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```

Pressing the TOGGLE button from OFF state:

22:05:31 SRC: WebGui from 192.168.100.228
22:05:31 MQT: blitzwolf-shp2-1/tele/STATE = {"Time":"2019-12-06T22:05:31","Uptime":"0T03:26:50","UptimeSec":12410,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":{"STATE":"ON"},"Wifi":{"AP":1,"SSId":"<MySSID>","BSSId":"<MyAPMac>","Channel":4,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:06"}}
22:05:31 MQT: blitzwolf-shp2-1/stat/RESULT = {"POWER":"ON"}
22:05:31 MQT: blitzwolf-shp2-1/stat/POWER = ON
22:05:31 CFG: Saved to flash at FA, Count 1164, Bytes 4096

Pressing the TOGGLE button from ON state:

22:05:57 SRC: WebGui from 192.168.100.228
22:05:57 MQT: blitzwolf-shp2-1/tele/STATE = {"Time":"2019-12-06T22:05:57","Uptime":"0T03:27:16","UptimeSec":12436,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"POWER":{"STATE":"OFF"},"Wifi":{"AP":1,"SSId":"<MySSID>","BSSId":"<MyAPMac>","Channel":4,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:06"}}
22:05:57 MQT: blitzwolf-shp2-1/stat/RESULT = {"POWER":"OFF"}
22:05:57 MQT: blitzwolf-shp2-1/stat/POWER = OFF
22:05:57 CFG: Saved to flash at F9, Count 1165, Bytes 4096

```

### TO REPRODUCE
1. Install a binary build from commit 45d5999 or later
2. Press the TOGGLE button in WebUI or send POWER ON / OFF command via MQTT
3. Note the value of the "POWER" field in %topic%/tele/STATE:

`"POWER":{"STATE":"ON"}`
or
`"POWER":{"STATE":"OFF"}`

### EXPECTED BEHAVIOUR
The value of the "POWER" field in %topic%/tele/STATE for releases, build before commit 45d5999, is (the one I used is from commit 0d138c8):

`"POWER":"ON"`
or
`"POWER":"OFF"`

### SCREENSHOTS
Following is the autoconfiguration payload for the device with firmware 45d5999:

```
{"name":"MasterPlug",
"cmd_t":"~cmnd/POWER",
"stat_t":"~tele/STATE",
"val_tpl":"{{value_json.POWER}}",
"pl_off":"OFF",
"pl_on":"ON",
"avty_t":"~tele/LWT",
"pl_avail":"Online",
"pl_not_avail":"Offline",
"uniq_id":"33B213_LI_1",
"device":{"identifiers":["33B213"],
"connections":[["mac","DC:4F:22:33:B2:13"]]},
"~":"blitzwolf-shp2-1/"}
```

The same device type, but flashed with earlier firmware - 7.0.0.5(0d138c8-tasmota), functions correctly. It seems that it has the same autoconfiguration payload:

```
{"name":"Back Light",
"cmd_t":"~cmnd/POWER",
"stat_t":"~tele/STATE",
"val_tpl":"{{value_json.POWER}}",
"pl_off":"OFF",
"pl_on":"ON",
"avty_t":"~tele/LWT",
"pl_avail":"Online",
"pl_not_avail":"Offline",
"uniq_id":"33EAED_LI_1",
"device":{"identifiers":["33EAED"],
"connections":[["mac","DC:4F:22:33:EA:ED"]]},
"~":"blitzwolf-shp2-2/"}
```

### ADDITIONAL CONTEXT
Home Assistant version is 0.102.3, although I suspect earlier releases will exhibit the same behaviour. It seems that the autoconfigured device denotes that the power state is extracted from %topic%/tele/STATE:

"pl_off":"OFF"
"pl_on":"ON"

which is changed in releases past commit 45d5999 as explained in steps to reproduce above.

**(Please, remember to close the issue when the problem has been addressed)**

znanev, 2019-12-06:
Sorry to be the first one to comment on my own issue, but is there a specific reason HA is instructed through the autoconfigured message to extract the power state from %topic%/tele/STATE and not from %topic%/stat/RESULT?

As I see it (at least on my end), those 3 messages are published when power state is changed:

%topic%/tele/STATE
%topic%/stat/POWER
%topic%/stat/RESULT

And power state is most "buried" in tele/STATE as opposed to the more simplistic stat/POWER and state/RESULT.

ascillato, 2019-12-06:
Reverting back this changes in Tasmota's status message due to it is breaking backwards compatibility with several tools and setups.

Fixed with #https://github.com/arendst/Tasmota/pull/7127

znanev, 2019-12-06:
@ascillato That was quick, thanks!

I'll test with a new version once your pull request is merged.

ascillato2, 2019-12-07:
Merged. Please, test. Thanks.

znanev, 2019-12-07:
Tested with binary built from f499373 - working again.

Thank you!

closed 2019-12-07 by znanev