arendst/Tasmota #7126: interlock fails in tuyamcu (with mqtt or on the console)

biotecov, 2019-12-06:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Tuya-MCU wall touch switch with 4 buttons, flashed with tasmota 7.1.1.1 and with 7.2.0.
I selected module 55 (tuya MCU), then:
backlog tuyamcu 12,2; tuyamcu 13,3; tuyamcu 14,4 (this activates successfully the four buttons)
and finally
backlog interlock 1,2,3,4; interlock onI selected module 55 (tuya MCU), then:
backlog tuyamcu 12,2; tuyamcu 13,3; tuyamcu 14,4
and finally
backlog interlock 1,2,3,4; interlock on

Command "powerX on" on the console or mqtt activates relayX and ledX (as expected), but then, "powerY on" switches off relayX and ledX but fails to activate relayY and ledY. Issuing a second instance of "powerY" does the expected action.

Interlock works as expected with the touch pad, but not with commands: command "powerX on" on the console or mqtt activates relayX and ledX (as expected), but then, "powerY on" switches off relayX and ledX but fails to activate relayY and ledY. Issuing a second instance of "powerY" does the expected action.

In other words, interlock switches off the relay that was on, but fails to switch on the new relay via console or mqtt commands. This happens with any combination of X and Y (via commands) and for any interlock definition.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): wall touch switch based on Tuya MCU
- [ ] Tasmota binary firmware version number used: 7.1.1.1 and 7.2.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Visual
- [x] Flashing tools used: tuya-convert
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

select module 55 (tuya MCU), then:
backlog tuyamcu 12,2; tuyamcu 13,3; tuyamcu 14,4
backlog interlock 1,2,3,4; interlock on
power1 on
power2 on

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

relay1 shoud switch off (this happens) and relay2 switch on (but this fails)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-12-07:
Hi, please, for more debug information, can you complete the template? STATUS 0 and the output of the console when you are performing those actions. Remember to set weblog to 4. Thanks.

biotecov, 2019-12-07:
> ### PROBLEM DESCRIPTION
> 
> _A clear and concise description of what the problem is._
> 
> Tuya-MCU wall touch switch with 4 buttons, flashed with tasmota 7.1.1.1 and with 7.2.0.
> I selected module 55 (tuya MCU), then:
> backlog tuyamcu 12,2; tuyamcu 13,3; tuyamcu 14,4 (this activates successfully the four buttons)
> and finally
> backlog interlock 1,2,3,4; interlock onI selected module 55 (tuya MCU), then:
> backlog tuyamcu 12,2; tuyamcu 13,3; tuyamcu 14,4
> and finally
> backlog interlock 1,2,3,4; interlock on
> 
> Command "powerX on" on the console or mqtt activates relayX and ledX (as expected), but then, "powerY on" switches off relayX and ledX but fails to activate relayY and ledY. Issuing a second instance of "powerY" does the expected action.
> 
> Interlock works as expected with the touch pad, but not with commands: command "powerX on" on the console or mqtt activates relayX and ledX (as expected), but then, "powerY on" switches off relayX and ledX but fails to activate relayY and ledY. Issuing a second instance of "powerY" does the expected action.
> 
> In other words, interlock switches off the relay that was on, but fails to switch on the new relay via console or mqtt commands. This happens with any combination of X and Y (via commands) and for any interlock definition.
> ### REQUESTED INFORMATION
> 
> _Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_
> 
>     * [x]  Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
> 
>     * [x]  Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
> 
>     * [x]  Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
> 
>     * [x]  Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
> 
>     * [ ]  Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
> 
>     * [x]  Device used (e.g., Sonoff Basic): wall touch switch based on Tuya MCU
> 
>     * [ ]  Tasmota binary firmware version number used: 7.1.1.1 and 7.2.1
>       
>       * [ ]  Pre-compiled
>       * [x]  Self-compiled
>         
>         * [x]  IDE / Compiler used: Visual
> 
>     * [x]  Flashing tools used: tuya-convert
> 
>     * [x]  Provide the output of command: `Backlog Template; Module; GPIO 255`:
> 
> 
> ```
>   Configuration output here:
> ```
> 
>     * [ ]  If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
> 
> 
> ```
>   Rules output here:
> ```
> 
>     * [ ]  Provide the output of this command: `Status 0`:
> 
> 
> ```
>   STATUS 0 output here:
> ```
> 
>     * [ ]  Provide the output of the Console log output when you experience your issue; if applicable:
>       _(Please use_ `weblog 4` _for more debug information)_
> 
> 
> ```
>   Console output here:
> ```
> 
> ### TO REPRODUCE
> 
> _Steps to reproduce the behavior:_
> 
> select module 55 (tuya MCU), then:
> backlog tuyamcu 12,2; tuyamcu 13,3; tuyamcu 14,4
> backlog interlock 1,2,3,4; interlock on
> power1 on
> power2 on
> ### EXPECTED BEHAVIOUR
> 
> _A clear and concise description of what you expected to happen._
> 
> relay1 shoud switch off (this happens) and relay2 switch on (but this fails)
> ### SCREENSHOTS
> 
> _If applicable, add screenshots to help explain your problem._
> ### ADDITIONAL CONTEXT
> 
> _Add any other context about the problem here._
> 
> **(Please, remember to close the issue when the problem has been addressed)**

biotecov, 2019-12-07:
Hello, I updated the info as you requested. On the log, look in particular at these:

at 10:01:47 I issued "power1 on" (physical result: led1 and relay1 turned on)
at 10:01:56 I issued "power3 on" (result: led1 and relay1 turned off, but led3 and relay3 remained physically off [=not correct])

at 10:03:00 I pushed manually button 1 (result: led1 and relay1 turned on)
at 10:03:04 I pushed manually button 3 (result: led1 and relay1 turned off, and led3 and relay3 turned on [=correct])

Tommy-LSA, 2019-12-07:
Interlock seems not to set within a backlog queue. I‘ve seen this also. Setting it as solo command works

biotecov, 2019-12-07:
@Tommy-LSA thanks for the clue, but I'm affraid is not my case. I tried that too, several times (activating interlock as a solo command too), and to be sure, I just repeated now the test before writing back, and results are the same. Interlock works in all cases for the touch pad but fails in all cases with commands.

In my working setup I am using "interlock 1,3 2,4" instead of "interlock 1,2,3,4" because I have dedicated this switch to control two shutters, and once more, the same is observed, again: buttons 1 and 3, and buttons 2 and 4 behave as expected from the touchpad, but fail with commands. I have implemented a work-around with rules that works well, but this is beyond the bug report. For the bug report I reset one of the devices and did exactly what is explained above.

Ricardo.

biotecov, 2019-12-09:
@ascillato2 Hello, you marked this as "awaiting feedback". I wonder if you are still awaiting for more feedback from me. Maybe you did not notice that I completed all the missing info following your request. Please tell me if you want me to make more test. I am using a workaround but I believe what I reported is a bug in the communication between the ESP and the Tuya MCU.

biotecov, 2020-01-05:
@ascillato2 Good year to all, and especially to the devels. This issue is a bug from my point of view, and can be a potential source of troubles for unaware users. Although I have solved the practical problem with a workaround that works for me, I believe it should not be closed without some attention from the devels.

ascillato2, 2020-01-15:
@biotecov

Hi, any news on this? Have you tried latest version from development branch? Some fixes has been added to the codebase.

ascillato2, 2020-01-22:
@biotecov

Hi, any news on this? Have you tried latest version from development branch? Some fixes has been added to the codebase.

Please, when you have time report back if your issue is solved. Thanks.

closed 2020-01-22 by ascillato2