arendst/Tasmota #7131: 18x20 not working when GPIO as "Switch 1"is configured in Tasmota V 7.1.2. Bug?

helmar74, 2019-12-07:
Hi,

I have the following issue. My Wemos Mini D is configured this way:
[image]

With Version 7.1.1. everything is working fine.
In Version 7.1.2 the 18x20 1-wire isn't working (no Temperature displayes and sent via MQTT.=

If I change GPIO D5 (GPIO14) to "none", the temperature is shown.

The sensor was working for years with different tasmota versions without an issue.
So I think, there's a problem, when a tasmota device is configured like this.

I have done a complete delete of the flash with esptool, which also had any impact.
I changed 18x20 sensor and replaced wemos mini D with a new one.

Reverting firmware back to V7.1.1 brought finally back the complete functionality.

Maybe there's a bug in version 7.1.2?

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): Wemos Mini D with 1-wire 18x20 sensor
- [X] Tasmota binary firmware version number used: V7.1.2
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Arduino
-  X] Flashing tools used: ESPFlash
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
08:30:29 CMD: status 0
08:30:29 MQT: stat/sonoffWemosMini_4/STATUS = {"Status":{"Module":18,"FriendlyName":["sonoffWemosMini4"],"Topic":"sonoffWemosMini_4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:30:29 MQT: stat/sonoffWemosMini_4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffStandardGruppe","OtaUrl":"http://192.168.1.2:5555/data/firmwares/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:01","StartupUTC":"2019-12-07T07:29:28","Sleep":50,"CfgHolder":4617,"BootCount":22,"SaveCount":46,"SaveAddress":"F8000"}}
08:30:29 MQT: stat/sonoffWemosMini_4/STATUS2 = {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019.12.06 19:24:32","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(bb83b9b)","Hardware":"ESP8266EX"}}
08:30:29 MQT: stat/sonoffWemosMini_4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WIFISSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
08:30:29 MQT: stat/sonoffWemosMini_4/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000407","8FAAE19F","00278FA0","22BE1FCF","012033C0","00007081"],"Drivers":"1,2,3,4,5,6,8,9,10,13,16,17,18,19,22,24,26","Sensors":"1,3,4,5,6,7,8,9,10,14,15,16,17,18,20,22,26,35,41"}}
08:30:29 MQT: stat/sonoffWemosMini_4/STATUS5 = {"StatusNET":{"Hostname":"sonoffWemosMini-4","IPAddress":"192.168.1.177","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"DC:4F:22:1D:89:7E","Webserver":2,"WifiConfig":4}}
08:30:29 MQT: stat/sonoffWemosMini_4/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"sonoffWemosMini-4","MqttClient":"sonoffWemosMini-4","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
08:30:29 MQT: stat/sonoffWemosMini_4/STATUS7 = {"StatusTIM":{"UTC":"Sat Dec 07 07:30:29 2019","Local":"Sat Dec 07 08:30:29 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:46","Sunset":"16:13"}}
08:30:29 MQT: stat/sonoffWemosMini_4/STATUS10 = {"StatusSNS":{"Time":"2019-12-07T08:30:29","Switch1":"ON"}}
08:30:29 MQT: stat/sonoffWemosMini_4/STATUS11 = {"StatusSTS":{"Time":"2019-12-07T08:30:29","Uptime":"0T00:01:01","UptimeSec":61,"Vcc":2.980,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WIFISSID","BSSId":"E8:DF:70:67:B9:00","Channel":6,"RSSI":48,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-12-07:
Mine works as expected

[image]

Using this config

[image]

Using GPIO02 for a sensor on a Wemos is a bit of a challenge as the same GPIO is also used for the onboard LED. This will interfere with the DS18x20 hardware.

I see you made a lot of drivers/sensor changes in your configuration which I'm not planning to test. Rest assured the default binary works fine. You'll have to debug this yourselves.

helmar74, 2019-12-07:
Tried the "sensors" image from you, same problem.
Configuration:
[image]

[image]

On a other Wemos Mini, with my Image (where no switch is configured, it's working as expected):
[image]

[image]

Strange thing...

Aint-Nobody, 2019-12-07:
Are both boards from the same manufacurer ?

helmar74, 2019-12-07:
Shame on me!
It was my error. I took over my_user_config.h from Version 7.1.1
-> Then the error was there

Now I created a new my_user_config.h, made my changes (for additional sensors), compiled new... -> running

closed 2019-12-07 by helmar74