arendst/Tasmota #7133: Rollershutters: Configurable pause between change of direction of a moving rollershutter

wbuh123, 2019-12-07:
**Have you looked for this feature in other issues and in the wiki?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  
Dear dev-team of the rollershutter feature in tasmota,
this is a feature request:
it would be very nice to get an option to set a duration in seconds for a pause in case the user wants to change the direction of a moving rollershutter.
If you have heavy rollershutters and it is moving already and the operator wants change the direction it's better to stop moving of the rollershutter, wait a configurable duration and afterwards move into the new direction.

Do you see a chance to implement this?
Yours sincerly
wbuh

stefanbode, 2019-12-17:
This is already implemented but hard coden with 500ms. You need to change in the source code the variable Shutter_motor_delay from 500 to more if you need. If you find a situation where it does not work please report here

ascillato2, 2019-12-17:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-17 by ascillato2

wbuh123, 2020-01-06:
Dear @stefanbode,
thank you for your quick replay.
Unfortunately I'm not able to find the file with Shutter_motor_delay in the repository?
Thank you for any hint.
wbuh

wbuh123, 2020-02-16:
> 
> 
> This is already implemented but hard coden with 500ms. You need to change in the source code the variable Shutter_motor_delay from 500 to more if you need. If you find a situation where it does not work please report here

I can't find "Shutter_motor_delay". What's the file of this variable?
There is a MOTOR_STOP_TIME in xdrv_27_shutter.ino but I don't think it's the same?

taste66, 2020-02-16:
Looks like it is MOTOR_STOP_TIME
`const uint16_t MOTOR_STOP_TIME = 500;  // in mS`

See function:
```
void ShutterDelayForMotorStop(void)
{
  AddLog_P2(LOG_LEVEL_INFO, PSTR("SHT: Wait for Motorstop %d"), MOTOR_STOP_TIME);
  delay(MOTOR_STOP_TIME);
}
```

stefanbode, 2020-02-16:
Yes this was implemented to give the motor a small pause before changing directions. The main cause was just to ensure to prevent abrupt changes. Therefore this was hardcoded because there is normally no need to have it flexible.

taste66, 2020-02-16:
> Yes this was implemented to give the motor a small pause before changing directions. 

@stefanbode Thank you for this explanation. Turns-out my problem reported in #216 what not caused by a too short delay but by the fact the processor is really waiting and does not excuted other tasks. Any hints how I could fix this?