arendst/Tasmota #7135: Make config human readable/pritable

Aint-Nobody, 2019-12-07:
Dear dev-team,

it would be very nice, to read out the whole configuration as a readable/printable file.
Actual the backup is coded.

This would be very helpful to see what value has which parameters/options and has it been changed from default.
For example, if there is rule defined i dont take notice about it. So i must call every rule to see what is in there.
Another, log/lat is set by default to 48.858360, 2.294442 (Paris/Eiffeltowers top) i must call both to see if the value has changed to local.
and all the other options...

There is a way to do this with decode-config.py
but it seems only for freaks to do this in a commandline after installing phyton ect... its not a practice way for dummies :-).

May you can generate a option as a flag in the config that the backup can be generated as a readable ?

**Have you looked for this feature in other issues and in the wiki?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

jziolkowski, 2019-12-07:
Hi again.
As I said to you on Discord regarding this very inquiry, the only thing you can use right now is the decode-config tool from Tasmota tools repo. At this moment we don't have anything else. Yes, there are plans to make a GUI version of that tool, but for now we're a) spread too thin already, b) have current ongoing projects with higher priority.

Thanks for understanding.

Jason2866, 2019-12-07:
@Aint-Nobody  The available tool for decoding the configuration is here https://github.com/tasmota/decode-config

Aint-Nobody, 2019-12-07:
@Jason2866, a exceuable file for Windows is quiet new and very cool.
I try it and get a error. Do i make it wrong or is there a bug?
Can you help ?

D:>decode-config -f Config_sonoff_6735_7.0.0.6.dmp
Traceback (most recent call last):
  File "decode-config.py", line 3324, in <module>
  File "decode-config.py", line 2743, in Bin2Mapping
  File "decode-config.py", line 2407, in GetField
  File "decode-config.py", line 2397, in GetField
  File "decode-config.py", line 2416, in GetField
  File "decode-config.py", line 2290, in GetFieldValue
UnicodeDecodeError: 'ascii' codec can't decode byte 0xa5 in position 0: ordinal not in range(128)
[3376] Failed to execute script decode-config

ascillato2, 2019-12-07:
Closing this issue as it is **duplicated**. Sorry. For support, please use the Tasmota support chat. 

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-07 by ascillato2

jziolkowski, 2019-12-07:
@Jason2866 @curzon01 the exe file needs to be pulled from the repo, as it bundles python2 which doesn't work with unicode. I tried it few minutes ago on both my tasmota dump and of issue author. On the other hand, trying to use the module with python3 throws error about missing `past` module, which is not available to install.

curzon01, 2019-12-07:
pls move decode-config related issue to https://github.com/tasmota/decode-config
and pls add the version you are using (decode-config -V) as info, thanks

Aint-Nobody, 2019-12-07:
right now,there is no .exe for x86 anymore to find.

----- Original Message ----- 
  From: Norbert Richter
  To: arendst/Tasmota
  Cc: Aint-Nobody ; Mention
  Sent: Saturday, December 07, 2019 7:28 PM
  Subject: Re: [arendst/Tasmota] Make config human readable/pritable (#7135)

  pls move decode-config related issue to 
https://github.com/tasmota/decode-config
  and pls add the version you are using (decode-config -V) as info, thanks

Jason2866, 2019-12-07:
@Aint-Nobody  @curzon01 @jziolkowski 
Fixed the defective 32 bit exe version.

Aint-Nobody, 2019-12-07:
it works by the way.

but there is no format in it.
by the way, i will find a way to format the file.

thanks for your time.

cheers

----- Original Message ----- 
  From: Jason2866
  To: arendst/Tasmota
  Cc: Aint-Nobody ; Mention
  Sent: Saturday, December 07, 2019 7:57 PM
  Subject: Re: [arendst/Tasmota] Make config human readable/pritable (#7135)

  @Aint-Nobody @curzon01 @jziolkowski
  Fixed the defective 32 bit exe version.

jziolkowski, 2019-12-07:
Yes, the format is there but you must give it an option in the command line
run the .exe without parameters and it will show you help
there is JSON output, for example

Jason2866, 2019-12-08:
@Aint-Nobody  Decode-Config.exe does output now not only JSON
It is a mess to build the 32 bit version....  64bit works always!
```d:\>decode-config.exe -d 192.168.2.100 -g mqtt --output-format cmnd

# Mqtt:
  ButtonRetain 0
  ButtonTopic 0
  FullTopic %prefix%/%topic%/
  FullTopic sonoff-084CBE
  GroupTopic sonoffs
  MqttClient sonoff-084CBE
  MqttFingerprint1 39 4f 23 65 0b 32 36
  MqttFingerprint2 39 4f 23 65 0b 32 36
  MqttHost 192.168.2.114
  MqttPassword pass
  MqttPort 1883
  MqttRetry 10
  MqttUser user
  PowerRetain 0
  Prefix1 cmnd
  Prefix2 stat
  Prefix3 tele
  SensorRetain 0
  StateText1 OFF
  StateText2 ON
  StateText3 TOGGLE
  StateText4 HOLD
  SwitchRetain 0
  SwitchTopic 0
  TelePeriod 300```

Aint-Nobody, 2019-12-09:
@Jason2866
Nice to hear about a modified version. But ...
I do'nt want you drive crazy :-) 
But i hope you find the solution. May you upload a different but wrong version ?
Thanks for your work !

This is my string:
decode-config -d 10.1.1.50 -p 1234  --json-indent 6 --output-format command -g Mqtti -o @f_@v.txt 

And here the result:

{
      "SensorBits1": {}, 
      "button_topic": "0", 
      "cfg_crc": "0x9fac", 
      "cfg_crc32": "0xf55a6d72", 
      "cfg_size": 4096, 
      "cfg_timestamp": 1575813438, 
      "flag": {
            "mqtt_button_retain": 0, 
            "mqtt_power_retain": 0, 
            "mqtt_sensor_retain": 0, 
            "mqtt_switch_retain": 0
      }, 
      "flag2": {}, 
      "flag3": {}, 
      "header": {
            "data": {
                  "crc": "0x9fb9", 
                  "crc32": "0xf55a6d72", 
                  "size": 4096, 
                  "version": "0x7010203"
            }, 
            "format": {
                  "jsoncompact": false, 
                  "jsonhidepw": false, 
                  "jsonindent": 6, 
                  "jsonsort": true
            }, 
            "os": [
                  "AMD64", 
                  "Windows", 
                  "7", 
                  "6.1.7601", 
                  "Windows-7-6.1.7601-SP1"
            ], 
            "python": "3.7.5", 
            "script": {
                  "name": "decode-config.py", 
                  "version": "7.1.1.1 [00044]"
            }, 
            "template": {
                  "crc": "0x9fac", 
                  "crc32": "0xf55a6d72", 
                  "size": 4096, 
                  "version": "0x7010203"
            }, 
            "timestamp": "2019-12-09 09:40:48"
      }, 
      "mqtt_client": "DVES_%06X", 
      "mqtt_fingerprint": [
            "9O#e\u000b26", 
            "9O#e\u000b26"
      ], 
      "mqtt_fulltopic": "%prefix%/%topic%/", 
      "mqtt_grptopic": "sonoffs", 
      "mqtt_host": "domus1", 
      "mqtt_port": 1883, 
      "mqtt_prefix": [
            "cmnd", 
            "stat", 
            "tele"
      ], 
      "mqtt_pwd": "DVES_PASS", 
      "mqtt_retry": 10, 
      "mqtt_topic": "sonoff", 
      "mqtt_user": "DVES_USER", 
      "state_text": [
            "OFF", 
            "ON", 
            "TOGGLE", 
            "HOLD"
      ], 
      "switch_topic": "0", 
      "tele_period": 300, 
      "version": "0x7010203"
}

Aint-Nobody, 2019-12-09:
@jason2866
The 64Bit version bring out the same result.

Aint-Nobody, 2019-12-09:
@jason2866
important information for you,
when i do'nt use the -o parameter it works !
Writing to screen is ok, to a file not.

Jason2866, 2019-12-09:
Mhh, i do `decode-config.exe -d 192.168.2.130 --json-indent 6 --output-format command -g Mqtt -o Config_@f_@v.txt`
and i get 
[Config__7.1.2.4.txt](https://github.com/arendst/Tasmota/files/3939501/Config__7.1.2.4.txt)

Aint-Nobody, 2019-12-09:
That is exactly the same here.
A JSON formated list.

meingraham, 2019-12-09:
@Aint-Nobody 

To make sure we're on the same page...

The configuration backup Tasmota creates is a binary file with all the configuration information. `decode-config` has the option to read that file and generate a JSON file that is a text file that a human can inspect for reference. This file can be edited and loaded back to the device.

Is there something different you desire?

Aint-Nobody, 2019-12-09:
@meingraham

yes we are on the same page.
The JSON Format dont show the parameters clear.
For example, the param "Tempoffset" will be find as clear text in the command sorted list.
In the JSON formated list you cant find "Tempoffset"
In the commands sorted list, are all params for a theme together exmpl: mqtt, wifi ect.

But i read the comment from Norbert about the decode-config, that it makes no sense to spend to much brain to make
it work perfect. Cause every release you must also make a new decode-config.
Thats make no sense for sure.

ok, lets get away the dream of a human readable clear list.

I helped myself with the dos command: 
decode-config -d 10.1.1.50 -p 1234  --json-indent 6 --output-format command  > IP50.txt
That works fine to generate a clear file.

So, let me ask a question in the round about JSON,
If i like to change some basic params like logitude, latitude, timezone, ntpserver and so on, 
they are all have to be changed and all have the same value,
is it possible to write it in a self generated, small JSON file like:

{
"longitude": 6.931524,
"latitude": 50.942772
}
(This dont work)

to upload it to all devices?

Thank you all for your attention.

curzon01, 2019-12-09:
@Aint-Nobody - thanks. 
The difference between both text formats are, that the Tasmota command output format (`--output-format command`) sometimes is very complicated, because in that case decode-config has to understand how the parameter of the Tasmota command needs to work.

The JSON output uses the same names like in tasmota/settings.h source code as identifier, so to change something on that text based config you need to know the internal source name (for .e.g  `TempOffset` Tasmota command it's the id `temp_comp`). But you can find out read the decode-config.py code at the beginning - search for TempOffset and you will have the internal Tasmota var name).

BTW your last question: You can change whatever you want within JSON, only the value which is different to your current config will be change. You can also use a subset (delete the unnessesary value) of JSON to read back, this works also (only JSON structure must be valid).
So you can use a JSON which contains only the value you want to change, that's ok

Aint-Nobody, 2019-12-09:
@curzon01

Hi,
i understand your explanation about the differences between the param names in the two kinds of files.
I have no knowledge in PY or C++ .So i can read this Spaghetti Code :-)
I cant compile a file for my own desire. I allways need to use the precompiled files.
But i understand your description about it.
Programming is not a strange thing for me, but only assembler on AVR, 6502, else also php, bash, batch, vba, ect.

You wrote, that it is possible to load up a small JSON file with only the params i need.
Thats great.
But u also mention, that the JSON structure must be valid. Thats what i also think.
I have no idea of the right JSON structure, sorry.
Can you please show/place a empty JSON text, where only the params have to place at the right position ? 

You are not Theo, but may u have a explanation about this ?

I have a generally question,
why must the backup file be crypted?
Why must it be decoded as a JSON file?
On a cisco device you can type "show run", what means show running config,
and you get a output that shows what actual is working on this device in the configuration as a clear text.
That u can use to upload to others or backup if need.

Thanks for listening.

curzon01, 2019-12-10:
JSON format is well desribed on https://www.json.org/ and https://en.wikipedia.org/wiki/JSON - keep the structure as described, otherwise decode-config will try to report the position of the error.

A simple JSON which sets only a view settings is:
```
{
  "altitude": 10, 
  "flag": {
    "temperature_conversion": 0
  }, 
  "latitude": 48.85836, 
  "longitude": 2.294442,
  "header": {
  }
}

```
This is more or less the same as the Tasomta commands  
`Backlog Altitude 10;SetOption8 0;Latitude 48.85836;Longitude 2.294442`  
with the difference that only changed values will be change.

Notes:
- keep `header` object, it can also be empty {} like in the example
- the order of the parameters are unimportant
- try change some parameter and use decode-config with `-v` to see what will be changed in destination (device or file)

For further decode-config related issue pls move to repository https://github.com/tasmota/decode-config

Aint-Nobody, 2019-12-10:
@curzon01
Thank you very much for your explanation.
Now my litle JSON file with all the basics works fine to upload.

There comes a error at the end, but all values has changed.
FYI: ERROR 11 (@3032): Config data upload failed - <div style='text-align:center;'>Ge
rÃ¤t wird jetzt neu gestartet</div><br></div>.