arendst/Tasmota #7138: Another Feit Bulb Variation OM60/RGBW/CA/AG (P)

blakeblackshear, 2019-12-07:
### PROBLEM DESCRIPTION
It appears that I have a variation of the Feit bulb that is somewhere in between the two available templates.

The [Feit Electric OM60/RGBW/CA/AG RGBW](https://blakadder.github.io/templates/feit_electric-OM60RGBWCAAG.html) template behaves similar to the way others described in #6534 

The [Feit Electric BPA800/RGBW/AG/2 RGBW](https://blakadder.github.io/templates/feit_electric-BPA800RGBWAG2.html) template won't power on or respond to any of the color commands.

My FCC ID is `SYW-A19RGBWAGT2`, but my model number is `OM60/RGBW/CA/AG (P)`

[Energystar.gov](https://www.energystar.gov/productfinder/product/certified-light-bulbs/details/2347609) has it listed separately and as available on the market 9/2/2019. The date code for wifi is 228-19-45.

If needed, I am willing to break open one of the bulbs.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 7.1.2 and 6.7.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: upload
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
15:31:24 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:31:24 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:15","StartupUTC":"2019-12-07T14:31:09","Sleep":50,"CfgHolder":4617,"BootCount":8,"SaveCount":47,"SaveAddress":"FA000"}}
15:31:24 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"6.7.1(basic)","BuildDateTime":"2019-10-26T13:20:09","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
15:31:24 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["blakeshome-iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000136060000005A00000000000000","00000000"]}}
15:31:24 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":456,"Free":544,"Heap":31,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8F082383","041083A0","20021706","010001C0","00000000"],"Drivers":"1,2,3,4,9,10,16,20,21","Sensors":"3"}}
15:31:24 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-7064","IPAddress":"192.168.2.37","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.2","Mac":"60:01:94:E7:BB:98","Webserver":2,"WifiConfig":4}}
15:31:24 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E7BB98","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:31:24 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Sat Dec 07 14:31:24 2019","Local":"Sat Dec 07 15:31:24 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:29","Sunset":"16:53"}}
15:31:24 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2019-12-07T15:31:24"}}
15:31:24 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2019-12-07T15:31:24","Uptime":"0T00:00:15","UptimeSec":15,"Vcc":3.579,"Heap":31,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":59,"MqttCount":0,"POWER":"ON","Dimmer":89,"Color":"E31E","HSBColor":"0,0,0","Channel":[89,12],"CT":194,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"blakeshome-iot","BSSId":"F2:9F:C2:2D:CD:1E","Channel":11,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:05"}}}
15:31:25 RSL: tele/tasmota/STATE = {"Time":"2019-12-07T15:31:25","Uptime":"0T00:00:16","UptimeSec":16,"Vcc":3.581,"Heap":30,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":98,"MqttCount":0,"POWER":"ON","Dimmer":89,"Color":"E31E","HSBColor":"0,0,0","Channel":[89,12],"CT":194,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"blakeshome-iot","BSSId":"F2:9F:C2:2D:CD:1E","Channel":11,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:05"}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-12-07:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-07 by ascillato2

osiris247, 2019-12-16:
Did you find a way to make color work on these new bulbs?

blakeblackshear, 2019-12-16:
Not yet.

blakeblackshear, 2019-12-18:
Given the number of people who are referencing this issue on Discord, I am posting these images to save others from taking their bulbs apart. 

[image]

[image]

[image]

[image]

wtsoper, 2020-05-08:
Hi Blake,   
I have the same bulb, did you find a solution?

instantchow, 2020-05-21:
I recently purchased these as a four pack from Costco (2020 May) and found this template to work:

{"NAME":"FE-OM60-15K-AG","GPIO":[0,0,0,0,141,140,0,0,37,142,38,0,0],"FLAG":0,"BASE":18}' 

Found this via https://github.com/blakadder/templates/tree/master/_templates

however, my bulb is marked with the model OM60/RGBW/CA/AG (P) exactly as this thread is titled, but the template that worked was from the 15K AG bulb listed on the blackadder repo...

possibly useful marking on the bulb is: 228-19-46

not sure what this indicates, but in https://github.com/blakadder/templates/blob/master/_templates/feit_electric-OM60RGBWCAAG there is a note indicating that 228-20-04 does not work with Tasmota

will2568, 2020-05-30:
> possibly useful marking on the bulb is: 228-19-46

I've been told this is a date code but I'm not sure how to decipher it.

cgull, 2020-06-24:
I have the 228-20-04 bulb, from my utility's energy efficiency program.  It's got the same model name and FCC ID, but is *definitely* a different bulb-- it has a different module.  I think it's a Tuya WB2L module, which is ARM-based.  The boards are just a later revision though, it might be possible to just install a TYWE2L module after you rip the bulb apart.  Photos here, more description over at https://github.com/ct-Open-Source/tuya-convert/issues/675.

[image]
[image]
[image]
[image]
[image]
[image]

JaimeZX, 2020-11-29:
Can confirm that OM60/RGBW/CA/AG(P), FCCID: SYW-A19RGBWAGT2, 228-20-32, also fails to work with Tuya-Convert / Tasmota. Purchased from Costco online while on sale in August 2020. No surprise, just adding data.

takov751, 2020-12-12:
@cgull i have the exact same interior ,just different brand called `tcp smart` i have found this as well
https://developer.tuya.com/en/docs/iot/device-development/module/wifibt-dual-mode-module/wb-series-module/wb2l-datasheet?id=K9duegc9bualu

which explains that the dots on the back on your last pucture from the RST text UART0_RXD, UART0_TXD , UART1_RXD, UART1_TXD,GND . Uart0 is user interface Uart1 is log interface. I will proceed on with my findings.

takov751, 2020-12-12:
And so far it seems tasmota has no working version for this chip so far

nospam2k, 2020-12-15:
@takov751 - I have a similar bulb with the same chip. Were you able to flash it? I can't find an external access to io0. Is this an esp8266?

takov751, 2020-12-15:
@nospam2k No this is a somewhat modified closed source version possibly based on esp32 , and so far there is no capability to flash it, even if you find a way ,there is no available tasmota image as what's inside the chip  is unknown. Sadly the best we can do is somehow replace this chip with a esp32 or esp8266 . I have few devkits so i will be able ,but it will be ugly AF, as the dev boards are larger peaces and still have to figure out the connection layout . Hopefully if i have time i will map the connections and will make a schematic to make it easier for those whom comes after with similar issues. The more troublesome thing that its seems from this point onward all new tuya based products coming with these chips, so we really need to debug them and reverse whats inside .

takov751, 2020-12-15:
Someone pointed me to this project https://github.com/rospogrigio/localtuya , which is a home assistant plugin to locally govern all these tuya devices worth a try

nospam2k, 2020-12-15:
Bummer. These bulbs I have (3Stone) are 10w. Really nice. I'll send them back though.

jprvita, 2021-10-18:
> [image]

How did you all manage to remove this LED board from the enclosure, to access the next board with the chip? I'm having a hard time with that step. Did you have to destroy the enclosure? (I'm aware that even after getting to the chip it is not possible to flash a custom firmware at this time)

jprvita, 2021-10-20:
> @jprvita did you manage to remove the PCB? It seems to be glued to an underlying ring of metal which is probably used as a heatsink and I am also not sure how to remove it nondestructively.

No, but I have not tried much more after my last comment. I think maybe heating up the base (perhaps the heat from running the light for a while might be enough) and trying to pull the PCB with something may work, although I can't see an good way to grip the board and pull it.

Srslyrelax, 2021-11-27:
the bottom side of the LED board is all metal heatsink and the only thing keeping it on is the adhesive and the pin connector, I just stuck a 90 degree bent metal pick under it and pulled straight up. came out with very little trouble, slightly bent one of the pins but totally intact.

nx10, 2021-11-27:
> the bottom side of the LED board is all metal heatsink and the only thing keeping it on is the adhesive and the pin connector, I just stuck a 90 degree bent metal pick under it and pulled straight up. came out with very little trouble, slightly bent one of the pins but totally intact.

Could you maybe add a photo of the underside of the board if it is not too much trouble?

rickyelopez, 2022-01-28:
I got some of these bulbs as well, looks like they use the WB2L. Specifically, I think the four PWN pins on the bottom are key here. Can't find any pin-compatible ESP modules, anybody have any suggestions?