arendst/Tasmota #7139: Tasmota 7.1.2.2. stuck in minimal after upgrade.

theorician, 2019-12-07:
### PROBLEM DESCRIPTION
After upgrading to 7.1.2.2. my lightbulb socket is stuck in 7.1.2.2. minimal. 

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Elegant Choice E27/E26 Socket (https://blakadder.github.io/templates/elegant_choice.html) - some variant thereof
- [x] Tasmota binary firmware version number used: 7.0.0.6 upgrading to 7.1.2.2.
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Web
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
15:38:05 MQT: light_hallway/stat/RESULT = {"NAME":"Bulb Socket","GPIO":[0,0,0,0,56,0,0,0,21,17,0,0,0],"FLAG":0,"BASE":18}
15:38:05 MQT: light_hallway/stat/RESULT = {"Module":{"0":"Bulb Socket"}}
15:38:05 MQT: light_hallway/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"56":"Led1i"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"17":"Button1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  N/A
```
- [x] Provide the output of this command: `Status 0`:
```
15:38:42 MQT: light_hallway/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Hallway Light"],"Topic":"light_hallway","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":1}}
15:38:42 MQT: light_hallway/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"light_sockets","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:09","StartupUTC":"2019-12-07T14:30:33","Sleep":50,"CfgHolder":4617,"BootCount":24,"SaveCount":98,"SaveAddress":"FB000"}}
15:38:42 MQT: light_hallway/stat/STATUS2 = {"StatusFWR":{"Version":"7.1.2.2(758b255-minimal)","BuildDateTime":"2019-12-07T13:00:55","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285"}}
15:38:42 MQT: light_hallway/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxx","xxxxxxx"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["400A80A9","2805C8000100060000005AFF000000000000","00000200","00000000"]}}
15:38:42 MQT: light_hallway/stat/STATUS4 = {"StatusMEM":{"ProgramSize":369,"Free":632,"Heap":29,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","80082181","040003A2","20021702","010000C0","00000000"],"Drivers":"1,2","Sensors":""}}
15:38:42 MQT: light_hallway/stat/STATUS5 = {"StatusNET":{"Hostname":"light_hallway","IPAddress":"xxxxx","Gateway":"xxxxx","Subnetmask":"255.255.255.0","DNSServer":"xxxxx","Mac":"xxxxxxx","Webserver":2,"WifiConfig":2}}
15:38:42 MQT: light_hallway/stat/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxx","MqttPort":1883,"MqttClientMask":"light_hallway","MqttClient":"light_hallway","MqttUser":"xxxxxxxx","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:38:42 MQT: light_hallway/stat/STATUS7 = {"StatusTIM":{"UTC":"Sat Dec 07 14:38:42 2019","Local":"Sat Dec 07 15:38:42 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00"}}
15:38:42 MQT: light_hallway/stat/STATUS10 = {"StatusSNS":{"Time":"2019-12-07T15:38:42"}}
15:38:42 MQT: light_hallway/stat/STATUS11 = {"StatusSTS":{"Time":"2019-12-07T15:38:42","Uptime":"0T00:08:09","UptimeSec":489,"Vcc":3.435,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxx","BSSId":"xxxxxxx","Channel":11,"RSSI":98,"LinkCount":1,"Downtime":"0T00:00:06"}}}
15:38:42 MQT: light_hallway/stat/STATUS12 = {"StatusSTK":{"call_chain":[]}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  N/A
```

### TO REPRODUCE
🤷‍♂ Flash from 7.0.0.6 to 7.1.2.2 and have it fail? I have a few more of these but am not about to experiment...

### EXPECTED BEHAVIOUR
I expect it to flash via the minimal firmware to the actual firmware, but it's stuck on the minimal. 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
Now that it's stuck, I've tried flashing via the OtaUrl and via the web upload. I've tried various versions. None have any effect. It says the upload has succeeded and the device will restart momentarily. When it does, it comes back in minimal and just stops there. No further restarts happen and the device does not finish flashing.

I've also tried doing a `reset 5`, hasn't helped at all.

**(Please, remember to close the issue when the problem has been addressed)**

theorician, 2019-12-07:
So it's reporting as minimal firmware but it connects to MQTT and does autodiscovery in Home Assistant. It _works_, but the web frontend displays as minimal. Every feature works though. This is confusing.

arendst, 2019-12-07:
I guess it's sometung to do with commit https://github.com/arendst/Tasmota/commit/758b25507822ad3c862216feb4c1af4fd5d53e78

I'm also unable to load anything new.

Investigating.

theorician, 2019-12-07:
Thanks for the quick reply!

kristofejro, 2019-12-07:
I have the same issue. All of my sonoffs are on minimal now.
[image]

I was OTA upgrading from 7.1.1

Xsev1, 2019-12-07:
Same problem here. I was upgrading from 7.0.3 i think...

scargill, 2019-12-07:
I too have issues with 7.1.2.2 - yesterday using the latest TDM I updated a boatload of different devices - all core 2.6.1 according to TDM, all 7.x - to 7.1.2.2 - then his morning, one which was still on 7.1.2 I upgraded to 7.1.2.2 (I'm on Windows 10 64 bit Pro) and this last one would only go so far as minimal - it all seems to work but despite several refreshes, it still says Minimal.  I ALSO took an old Sonoff DUAL which had not been on the system and loaded (using Tasmotize) 7.1 onto it. That worked but when I tried to upgrade it to 7.1.2.2 OTA (dev) I got the same - Minimal. As it was (is) still connected to serial I grabbed tasmota.bin (7.1.2.2) from the web and that one has now using Tasmotize upgraded to 7.1.2.2 but the first one I mentioned - which is not on serial as it is in my loft - is still displaying "Minimal".  So as of last night/this morning it seems something has gone wrong.

I'm using TDM 2.5 (exe version) and there are other issues and the author says it is down to Tasmota. For example, ONE of my boards refuses to show the WebUI in TDM even though when using a browser the WebUI is just fine a- nd my other boards show WebUI both in TDM and stand alone in Chrome.

ALL my boards are on Core 2.6.1 and 7.1.2.2

arendst, 2019-12-07:
As said before a recent commit is causing this issue.

I just reverted the commit. Expect functional binaries within two hours. 

There is a workaround but you only have two seconds to act at the correct moment otherwise flash will corrupt (to early) or you'll be back in the same situation (to late).

So if you can serial flash I would recommend that.

If you can perform a reset (not a power cycle) then perform the following from the console:
```
17:10:04 CMD: upload 1
17:10:04 SRC: WebConsole from 192.168.2.1
17:10:04 CMD: Group 0, Index 1, Command "UPLOAD", Data "1"
17:10:04 MQT: stat/wemos2/RESULT = {"Upload":"Version 7.1.2.2 from http://domus1:80/api/arduino/tasmota.bin"}
17:10:05 CFG: Saved to flash at FB, Count 303, Bytes 4096
17:10:07 HTP: Web server stopped
17:10:07 UPL: http://domus1:80/api/arduino/tasmota.bin
17:10:19 HTP: Web server active on wemos2 with IP address 192.168.2.191
17:10:19 MQT: Attempting connection...
17:10:19 MQT: Connected
17:10:19 MQT: tele/wemos2/LWT = Online (retained)
17:10:19 MQT: cmnd/wemos2/POWER = 
17:10:19 MQT: Subscribe to cmnd/wemos2/#
17:10:19 MQT: Subscribe to cmnd/tasmotas/#
17:10:19 MQT: Subscribe to cmnd/DVES_15568C_fb/#
17:10:20 MQT: stat/wemos2/UPGRADE = Successful. Restarting
```
Now you have two seconds max to reset the device.

After the reset the device seems to stall a little but will then continue executing the uploaded code.

Sorry for any inconvenience this may cause.

See https://github.com/arendst/Tasmota/pull/7134 for background information.

arendst, 2019-12-07:
Keep in mind this issue only occurs with the development version.

All releases do not have this issue so if you do not need the latest and greatest stick to the release version.

Xsev1, 2019-12-07:
Thank you for information and your work!!

scargill, 2019-12-07:
I’ll go find something else to do for a couple of hours 😊 I just uploaded a YouTube video to say how great 7.1.2.2 is (and didn’t bother to keep the project ) – oh, well, I guess I’ll put a comment in and say how great 7.1.2.3 is 😊

 

For clarity will all 7.1.2.2 boards need a serial reflash when this is resolved or are we just taking about being patient for a few hours or so?

 

Regards

 

Pete

arendst, 2019-12-07:
All boards with 7.1.2.2 AND commit https://github.com/arendst/Tasmota/commit/758b25507822ad3c862216feb4c1af4fd5d53e78 (datetime 20191207-13xxxx) have this issue. All others are fine.

scargill, 2019-12-07:
In my case too late – I upgraded all my boards (20 or so) (thankfully only the ones in THIS building)…. I’m afraid I’m a little sad that I usually go for the very latest.. while I’m here – and as I guess most of us call in only when there’s a problem – this is CRACKINGLY GOOD generally – I’m looking forward to trying out Sonoff WiFi/Zigbee Gateway maybe next week along with a Zemismart WiFi/Zigbee gateway – not yet checked if they are supported but I will.

 

Pete

theorician, 2019-12-07:
@arendst 

First of all thank you for not just the very quick response on this but also your hard work in general on this project. It really is amazing.

I'd like to try the workaround. I'm not exactly clear on the sequence though. Could you elaborate?

I issue the command `upload 1` and then wait for it to finish with 
> stat/wemos2/UPGRADE = Successful. Restarting

Then you say I have two seconds to "reset" the device. What exactly do you mean?

arendst, 2019-12-07:
A wemos has a reset button. Other devices I don't know. So pressing this reset button is what I meant.

theorician, 2019-12-07:
@arendst Aha, right. My bulb socket just has regular button which is just an on/off. I guess I'm stuck taking it apart for a serial flash...

theorician, 2019-12-07:
@arendst Hm, I just tried taking it apart - my device is this thing: https://blakadder.github.io/templates/elegant_choice.html - and I don't think I can take it apart without breaking it entirely. It also doesn't have a reset button, just the regular hardware on/off button. 

My device is therefore more or less bricked - it works in the current network configuration (server IP, MQTT credentials, etc) but any change will render it unusable. @scargill might be in a similar position, I'm not sure. A replacement wouldn't be too bad, since they're cheap, but it's a shame to waste the hardware.

If there's any other workaround you can give us, that would be immensely appreciated. Thanks again for your efforts!

scargill, 2019-12-07:
I'm still seeing 7.1.2.2 in dev ??

ascillato, 2019-12-07:
Yes, that is the actual version of the development branch

joder79, 2019-12-07:
I'm also stuck here with two newly bought Gosund SP111. I just wanted to update the v6 Sonoff-Tasmota that was already pre-installed. As the noob I am, I don't have any clue how to get them unbricked again, I have never soldered before... any help on how to get them back to work without disassembling them would be highly appreciated. Thanks in advance and keep up the great work!

marcgrc, 2019-12-07:
i have got same problem.
i attend resolve bug

Tommy-LSA, 2019-12-08:
As a noob I would only install release versions, no development versions. Development versions are for testing before release, to find exactly such issues. With Google you should find some pages which document serial flashing. E.g. https://www.bastelbunker.de/gosund-sp111-mit-tasmota/

scargill, 2019-12-08:
I hate to say this as a veteran but I didn't expect updatin issues with development versions... at least I know how to solder, in my cae life is more complicate as some of my kit is in another country - thankfully I didn't update any of those. To be fair it's a while since OTA has been an issue...but you're right, Tommy.. being patient and avoiding development versions except for handy units sitting on the workbench might be a sensinble thing - he says in hindsight. It is awfully tempting to want the very latest though...

sinesc, 2019-12-08:
Is it possible the upgrade URL still points to a broken version? Just had this issue after upgrading via thehackbox.org/tasmota/release/tasmota.bin.  In my case a reset 5 seems to have fixed the issue though.
Note: super new to this, don't know what I'm doing.

scargill, 2019-12-08:
Didnt work for me. I grabbed the release version of Tsamota.bin and stoerd it locally and tried that. the WebUI said "upload successful" but no change - still stuck on minimal after refreshing the browser. I even tried that twice. It's looking mor and more like a trip into the loft.

joder79, 2019-12-08:
> As a noob I would only install release versions, no development versions. Development versions are for testing before release, to find exactly such issues.

Yeah, I know. Too bad I had to change the OTA-URL after the change from "sonoff-tasmota" to "tasmota" and I managed to enter "http://thehackbox.org/tasmota/tasmota-DE.bin" instead of "http://thehackbox.org/tasmota/release/tasmota-DE.bin"...

mbomhof, 2019-12-08:
I have the same issue. All my devices are on minimal. Flashing them serial will be a pain because I have several units in the wall. Hope there will be a OTA solution soon...

s-hadinger, 2019-12-08:
I deeply apologize for the commit that caused the OTA to fail...

Does anyone know a command that would generate a crash of the ESP8266? That would make it possible to force reboot, equivalent to the reset button, without RTC memory being mistakenly wiped.

arendst, 2019-12-08:
But that command would write RTC call chain memory overwriting the ota/upload bootloader data ;-(

s-hadinger, 2019-12-08:
Good point... Sadly...

s-hadinger, 2019-12-08:
What about a quick sub-second power off-power on during that 2 seconds window? Would RTC memory survive a quick shutdown?

scargill, 2019-12-08:
> What about a quick sub-second power off-power on during that 2 seconds window? Would RTC memory survive a quick shutdown?

That assumes access and if you have that you may as well program serially. I'm trying to avoid dismantling a unit in the loft..... ideas for anything that can be done from a remote PC?

s-hadinger, 2019-12-08:
I'm assuming it's easier to shut down power from the fuse panel than opening the device. It's not ideal but the best option I can think about.

scargill, 2019-12-08:
Well guys - it turns out my unit wasn't in the loft so I used Tasmotizer to flash it serially. That worked so while connected I tried updating to dev version 7.1.2.2 (nothing to lose) and I can confirm that now works.... just a matter of sticking those screws back in now and this time I've given the unit a sensible name reflecting where it actually is :-).

theorician, 2019-12-08:
@s-hadinger No worries, as a software dev I know that these things can happen. I usually flash the release version; for some reason the OtaUrl was set incorrectly for this device and I didn't check before hitting upgrade. 

You think that a quick off-on during that two second window will do the trick? I can try that by plugging my device into another tasmota and telling it to restart.

theorician, 2019-12-08:
No - that doesn't seem to do the trick unfortunately. Oh well. :)

joder79, 2019-12-08:
Any possibility to have a "fresh" restart via https://github.com/ct-Open-Source/tuya-convert maybe somehow?

s-hadinger, 2019-12-08:
@joder79 No, it won't work with tuya-convert.

bburkett52, 2019-12-08:
Serial re-flash good! Subsequent OTA good! Thanks for all the work!

scargill, 2019-12-09:
Erm all of my boards were earlier upgraded to 7.1.2.2 - and then I thought I may as well bring them bang up to date with 7.1.2.4 - everyone worked - except for one - an original Sonoff DUAL - which has stubbornly stayed at 7.1.2.2 - not minimal I may add, the full 7.1.2.2 - I've tried booting the board and turning TDM off and on then starting all over - it just will not move... I'm not suer I fully undterstand the console output but... 

[01:20:46] stat/otherdual/RESULT {"OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin"}
[01:20:46] stat/otherdual/RESULT {"Upgrade":"Version 7.1.2.2 from http://thehackbox.org/tasmota/tasmota.bin"}
[01:20:51] tele/otherdual/LWT Offline

It LOOKS like the DUAL has missed out in the current round of upgrades - am I right?

stevevg, 2019-12-09:
I have the same issue (stuck on minimal firmware - please upgrade) trying to upgrade a teckin sp22 plug from 6.6.0 to latest 7.1.2
I had already done upgrades before (all in the 6.x major), which went smoothly, but now I'm stuck. Already tried to upload tasmota.bin via file upload, with no luck.

Xsev1, 2019-12-09:
Will we get an OTA update to fix the problem or is a new seriall flash the only way?

arendst, 2019-12-09:
No OTA. Pls read above why.

Xsev1, 2019-12-09:
Thank you for info.

danbamford, 2019-12-15:
I've just tried updating from 6.3.0 to 7.1.2 (release version) and had the exact same problem.
2x S26 plugs stuck on minimal. Is there anyway around this or is the only option to take them apart? 
This thread seems to indicate it's only the development version effected but like I say this was from the release page. Any help welcome, thanks.

sinesc, 2019-12-16:
@danbamford Have you tried waiting a minute or so after the upgrade? I thought I had this issue, but simply waiting a bit longer for the bulbs to fully reboot seemed to have fixed the minimal state.

danbamford, 2019-12-16:
> @danbamford Have you tried waiting a minute or so after the upgrade? I thought I had this issue, but simply waiting a bit longer for the bulbs to fully reboot seemed to have fixed the minimal state.

Yes I waited and the device reset after the upgrade but just sat there on the minimal firmware, nothing changed after 10 minutes.

ascillato2, 2019-12-17:
Closing this issue as the problem in the firmware was fixed.

As explained in previous comments, if you have your device stuck on minimal due to this bug, you will need to flash by serial.

If you need assistant on that, please address this to the Tasmota support chat. Thanks.

closed 2019-12-17 by ascillato2

Simon-N-Thornton, 2024-06-07:
Device:  NOUS A1T Smart Socket with power monitoring
Original Verion: 13.0.0
Original features enabled: MQTT, Syslog
Chipset: 8266, 1M Flash
Upgrade path: 13.0.0-sensors -> 14.1.0 minimal -> 14.1.0 sensors (hopefully)

I have a lot of these smart sockets and the majority updated correctly apart from two; one is bricked and requires flashing via serial port and the other refuses to upgrade beyond minimal.

I've tried the upgrade via the web interface and from the console:

`OtaUrl http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz
Upgrade 1`

It says it is upgrading and will reset but after reset it is still minimal. Upgrading via file upload does not work because of inadequate space.

`Jun  7 11:50:45 192.168.200.200 tasmota-XXXXXX ESP-MQT: pdu/stat/tasmotaXXXXXX/RESULT = {"OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz"}
Jun  7 11:51:18 192.168.200.200 tasmota-XXXXXX ESP-MQT: pdu/tele/tasmotaXXXXXX/STATE = {"Time":"2024-06-07T10:51:18","Uptime":"0T00:20:10","UptimeSec":1210,"Vcc":3.667,"Heap":33,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"WiFi-2.4-DFF6","BSSId":"D6:21:F9:37:F6:EF","Channel":6,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:04"}}
Jun  7 11:51:24 192.168.200.200 tasmota-XXXXXX ESP-CMD: otaurl http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz
Jun  7 11:51:24 192.168.200.200 tasmota-XXXXXX ESP-MQT: pdu/stat/tasmotaXXXXXX/RESULT = {"OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz"}
Jun  7 11:51:39 192.168.200.200 tasmota-XXXXXX ESP-CMD: upgrade 1
Jun  7 11:51:39 192.168.200.200 tasmota-XXXXXX ESP-MQT: pdu/stat/tasmotaXXXXXX/RESULT = {"Upgrade":"Version 14.1.0 from http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz"}
Jun  7 11:52:34 192.168.200.200 tasmota-XXXXXX ESP-MQT: Attempting connection...
Jun  7 11:52:34 192.168.200.200 tasmota-XXXXXX ESP-MQT: Connected
Jun  7 11:52:34 192.168.200.200 tasmota-XXXXXX ESP-MQT: pdu/tele/tasmotaXXXXXX/LWT = Online (retained)
Jun  7 11:52:34 192.168.200.200 tasmota-XXXXXX ESP-MQT: pdu/cmnd/tasmotaXXXXXX/POWER = 
Jun  7 11:52:34 192.168.200.200 tasmota-XXXXXX ESP-MQT: pdu/stat/tasmotaXXXXXX/RESULT = {"Command":"Unknown"}
Jun  7 11:55:14 192.168.200.200 tasmota-XXXXXX ESP-CFG: Loaded from flash at FB, Count 192
Jun  7 11:55:14 192.168.200.200 tasmota-XXXXXX ESP-Project tasmota - sonoff11 Version 14.1.0(release-minimal)-2_7_6(2024-06-03T11:53:06)
Jun  7 11:55:14 192.168.200.200 tasmota-XXXXXX ESP-WARNING This version does not support persistent settings
Jun  7 11:55:14 192.168.200.200 tasmota-XXXXXX ESP-WIF: Connecting to AP1 YYYYYYY Channel 6 BSSId AA:BB:CC:DD:EE:FF in mode 11n as tasmota-XXXXXX...
Jun  7 11:55:14 192.168.200.200 tasmota-XXXXXX ESP-WIF: Connected
Jun  7 11:55:14 192.168.200.200 tasmota-XXXXXX ESP-HTP: Web server active on tasmota-XXXXXX with IP address 192.168.200.200
Jun  7 11:55:15 192.168.200.200 tasmota-XXXXXX ESP-MQT: Attempting connection...
Jun  7 11:55:15 192.168.200.200 tasmota-XXXXXX ESP-MQT: Connected
Jun  7 11:55:15 192.168.200.200 tasmota-XXXXXX ESP-MQT: pdu/tele/tasmotaXXXXXX/LWT = Online (retained)
Jun  7 11:55:15 192.168.200.200 tasmota-XXXXXX ESP-MQT: pdu/cmnd/tasmotaXXXXXX/POWER = 
Jun  7 11:55:15 192.168.200.200 tasmota-XXXXXX ESP-MQT: pdu/tele/tasmotaXXXXXX/INFO1 = {"Info1":{"Module":"NOUS A1T","Version":"14.1.0(release-minimal)","FallbackTopic":"cmnd/DVESXXXXXX_fb/","GroupTopic":"pdu/cmnd/tasmotas/"}}`

What is curious is that it has retained the syslog and MQTT settings, I'm still getting these messages evn though there isn't a config option for it.

Does anyone have any advice before I dismantle the socket and flash them via the serial port?

Thanks,

Simon