arendst/Tasmota #7142: Report White when using RGBW (H801 Device)

florian-asche, 2019-12-08:
**Have you looked for this feature in other issues and in the wiki?**  
yes
**Is your feature request related to a problem? Please describe.**  
When using RGBW with H801 Device the White Status is not reported back in MQTT.

**Describe the solution you'd like**  
Remote the White Status.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
If you use RGBW and FHEM the Color cannot be used due to its added lengh by white stripe leds. I want to use HSB and White seperated like its currently working in Tasmota. But HSB is just RGB without the White color. 

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2019-12-08:
I'm not sure to understand what you want. Can you please post the MQTT message you are receiving and what field you want to be added.

Also, if you need to control separately RGB and White, use `SetOption37 128`.

florian-asche, 2019-12-08:
Hi, if you are using a RGB Stripe that has also Warm or Cold white LEDs, you have 4 or 5 channels. Lets try my example with 4 channels. 

I can cantrol the RGB leds by sending RRGGBB to ColorX. Also i know i could send RRGGBBWW to ColorX. But that is not supported by fhem. 

Because of that i am sending HSBColor to HSBColorX command. And if i want to control the warmwhite or coldwhite LEDs i send White 50 or White 100. 

The problem here is, that someone implemented the "White 100" command without it sending back information by MQTT. 

Setup:
`{"Time":"2019-12-08T14:37:33","Uptime":"7T01:27:37","UptimeSec":610057,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER":"OFF","Dimmer":34,"Color":"0,0,0,87","HSBColor":"0,0,0","Channel":[0,0,0,34],"Scheme":0,"Fade":"ON","Speed":4,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"ATLANTIS_IOT","BSSId":"64:66:B3:79:DE:58","Channel":9,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:06"}}`

s-hadinger, 2019-12-10:
Still on my todo list...

closed 2019-12-28 by arendst

florian-asche, 2020-01-16:
Hi, 

just as feedback, many thanks. Its working perfectly.

Regards, Florian

florian-asche, 2020-01-16:
But i found one more, if the RGB Stripe is state OFF and someone sets Color to green the state is not reported back as ON.

PR #7358 (2019-12-28): Fix: White added to light status (#7142)
## Description:

Multiple fixes with `White` command and status:

- Add `White` attribute to light status
- Enable `White` command for RGBW+RGBCW bulbs, was only for RGBW
- Disable `CT` and `White` commands with `SetOption68 1`
- Fix: `White` command would wrongly switch to decimal mode `SetOption17`
- Fix: `White` command would disable RGB channel even with `SetOption37 128`
- Fix: `HSBColor` would send status twice

**Related issue (if applicable):** fixes #7142

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only [... cut]