arendst/Tasmota #7149: H801 dimmer HomeAssistant autoconfig not working

namadori, 2019-12-08:
### PROBLEM DESCRIPTION
H801 5 channel dimmer configured as independent channels with SetOption68 1.

HomeAssistant autoconfiguration sees 3 switches and 1 light instead of 5 lights. Tasmota interface is working as expected.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): H801
- [ ] Tasmota binary firmware version number used: 7.1.2
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
00:34:53 CMD: Backlog Template; Module; GPIO 255
00:34:53 MQT: sonoffled-4/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
00:34:53 MQT: sonoffled-4/stat/RESULT = {"Module":{"20":"H801"}}
00:34:54 MQT: sonoffled-4/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"52":"Led1"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"41":"PWM5"},"GPIO5":{"57":"Led2i"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"39":"PWM3"},"GPIO13":{"38":"PWM2"},"GPIO14":{"40":"PWM4"},"GPIO15":{"37":"PWM1"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
00:35:24 MQT: sonoffled-4/stat/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
00:35:24 MQT: sonoffled-4/stat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
00:35:24 MQT: sonoffled-4/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
```
- [ ] Provide the output of this command: `Status 0`:
```
00:35:49 MQT: sonoffled-4/stat/STATUS = {"Status":{"Module":20,"FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4"],"Topic":"sonoffled-4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:35:49 MQT: sonoffled-4/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:32:15","StartupUTC":"2019-12-08T23:03:34","Sleep":50,"CfgHolder":4617,"BootCount":4,"SaveCount":12,"SaveAddress":"F8000"}}
00:35:49 MQT: sonoffled-4/stat/STATUS2 = {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019-12-06T11:22:22","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
00:35:49 MQT: sonoffled-4/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Spongebob",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["400A8009","2805C8000100060000005A00000000000000","00040200","00000000"]}}
00:35:49 MQT: sonoffled-4/stat/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
00:35:49 MQT: sonoffled-4/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoffled-4-4164","IPAddress":"192.168.2.70","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.254","Mac":"3C:71:BF:31:90:44","Webserver":2,"WifiConfig":4}}
00:35:49 MQT: sonoffled-4/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.33","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_319044","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:35:49 MQT: sonoffled-4/stat/STATUS7 = {"StatusTIM":{"UTC":"Sun Dec 08 23:35:49 2019","Local":"Mon Dec 09 00:35:49 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:31","Sunset":"16:53"}}
00:35:49 MQT: sonoffled-4/stat/STATUS10 = {"StatusSNS":{"Time":"2019-12-09T00:35:49"}}
00:35:49 MQT: sonoffled-4/stat/STATUS11 = {"StatusSTS":{"Time":"2019-12-09T00:35:49","Uptime":"0T00:32:15","UptimeSec":1935,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","Channel1":100,"POWER2":"OFF","Channel2":100,"POWER3":"OFF","Channel3":100,"POWER4":"OFF","Channel4":100,"POWER5":"OFF","Channel5":100,"Color":"255,255,255,255,255","Wifi":{"AP":1,"SSId":"Spongebob","BSSId":"78:8A:20:57:28:86","Channel":13,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
00:09:34 CMD: setoption19 1
00:09:34 MQT: sonoffled-4/stat/RESULT = {"SetOption19":"ON"}
00:09:35 MQT: homeassistant/light/319044_LI_2/config =  (retained)
00:09:35 MQT: homeassistant/switch/319044_RL_2/config = {"name":"Tasmota2","cmd_t":"~cmnd/POWER2","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER2}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"319044_RL_2","device":{"identifiers":["319044"],"connections":[["mac","3C:71:BF:31:90:44"]]},"~":"sonoffled-4/"} (retained)
00:09:35 MQT: homeassistant/light/319044_LI_3/config =  (retained)
00:09:35 MQT: homeassistant/switch/319044_RL_3/config = {"name":"Tasmota3","cmd_t":"~cmnd/POWER3","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER3}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"319044_RL_3","device":{"identifiers":["319044"],"connections":[["mac","3C:71:BF:31:90:44"]]},"~":"sonoffled-4/"} (retained)
00:09:35 MQT: homeassistant/light/319044_LI_4/config =  (retained)
00:09:35 MQT: homeassistant/switch/319044_RL_4/config = {"name":"Tasmota4","cmd_t":"~cmnd/POWER4","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER4}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"319044_RL_4","device":{"identifiers":["319044"],"connections":[["mac","3C:71:BF:31:90:44"]]},"~":"sonoffled-4/"} (retained)
00:09:35 MQT: homeassistant/switch/319044_RL_5/config =  (retained)
00:09:35 MQT: homeassistant/light/319044_LI_5/config = {"name":"Tasmota 5","cmd_t":"~cmnd/POWER5","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER5}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"319044_LI_5","device":{"identifiers":["319044"],"connections":[["mac","3C:71:BF:31:90:44"]]},"~":"sonoffled-4/","bri_cmd_t":"~cmnd/Dimmer","bri_stat_t":"~tele/STATE","bri_scl":100,"on_cmd_type":"brightness","bri_val_tpl":"{{value_json.Dimmer}}","rgb_cmd_t":"~cmnd/Color2","rgb_stat_t":"~tele/STATE","rgb_val_tpl":"{{value_json.Color.split(',')[0:3]|join(',')}}","fx_cmd_t":"~cmnd/Scheme","fx_stat_t":"~tele/STATE","fx_val_tpl":"{{val ... (retained)
00:09:35 MQT: homeassistant/light/319044_LI_6/config =  (retained)
00:09:35 MQT: homeassistant/switch/319044_RL_6/config =  (retained)
00:09:35 MQT: homeassistant/light/319044_LI_7/config =  (retained)
00:09:35 MQT: homeassistant/switch/319044_RL_7/config =  (retained)
00:09:35 MQT: homeassistant/light/319044_LI_8/config =  (retained)
00:09:35 MQT: homeassistant/switch/319044_RL_8/config =  (retained)
00:09:35 MQT: homeassistant/binary_sensor/319044_BTN_1/config =  (retained)
00:09:35 MQT: homeassistant/binary_sensor/319044_BTN_2/config =  (retained)
00:09:35 MQT: homeassistant/binary_sensor/319044_BTN_3/config =  (retained)
00:09:35 MQT: homeassistant/binary_sensor/319044_BTN_4/config =  (retained)
[... 1 lines cut ...]
00:09:35 MQT: homeassistant/binary_sensor/319044_SW_2/config =  (retained)
00:09:35 MQT: homeassistant/binary_sensor/319044_SW_3/config =  (retained)
[... cut ...]
00:09:35 MQT: homeassistant/binary_sensor/319044_SW_6/config =  (retained)
00:09:35 MQT: homeassistant/binary_sensor/319044_SW_7/config =  (retained)
00:09:35 MQT: homeassistant/binary_sensor/319044_SW_8/config =  (retained)
```

### TO REPRODUCE
Reset 6
SetOption68 1
SetOption19 1

### EXPECTED BEHAVIOUR
H801 configured in HomeAssistant as 5 lights

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

namadori, 2020-01-06:
Bump to keep this open.

ascillato2, 2020-02-04:
Closing this request as nobody is working on it.

If anyone is interested on working on this, please make a PR. If anyone needs help on that, please, just ask. Thanks.

closed 2020-02-04 by ascillato2