arendst/Tasmota #7150: Power Calibration Commands

daniam75, 2019-12-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Cannot issue commands to calibrate power sensor.  PowerSet, CurrentSet and VoltageSet all return "Command":"Unknown"

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____ Wemos D1 Mini 
- [x ] Tasmota binary firmware version number used: _____ 7.1.2
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: _____ OTA update
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

```
  Configuration output here:
02:47:07 RSL: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
02:47:08 RSL: stat/sonoff/RESULT = {"Module":{"18":"Generic"}}
02:47:08 RSL: stat/sonoff/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
02:48:33 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
02:48:33 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:19:18","StartupUTC":"2019-12-09T01:29:15","Sleep":50,"CfgHolder":4617,"BootCount":13,"SaveCount":24,"SaveAddress":"FA000"}}
02:48:33 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"7.1.2(sensors)","BuildDateTime":"2019-12-06T11:26:57","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
02:48:33 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TheInternet",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
02:48:33 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":582,"Free":420,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE387","0437A3A4","B7FFBFCD","01DA9BC0","0006FCC7"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,18,19,20,21,22,24,26,29,30","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,34,37,39,40,42,43,51,52,55,56"}}
02:48:33 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"INA_Mini","IPAddress":"192.168.1.239","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"2C:F4:32:78:18:4A","Webserver":2,"WifiConfig":4}}
02:48:33 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_78184A","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
02:48:33 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 09 01:48:33 2019","Local":"Mon Dec 09 02:48:33 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:31","Sunset":"16:53"}}
02:48:33 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-12-09T02:48:33","INA219":{"Id":40,"Voltage":1,"Current":2.533,"Power":3}}}
02:48:33 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-12-09T02:48:33","Uptime":"0T00:19:18","UptimeSec":1158,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"TheInternets","BSSId":"80:2A:A8:41:E2:21","Channel":6,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
"Command":"Unknown"

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Issue the commands PowerSet, VoltageSet, or CurrentSet

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Expected to set the current Power, Current and Voltage

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-09:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-09 by ascillato2

meingraham, 2019-12-09:
Since Tasmota only sees the you have an I<sup>2</sup>C sensor connected but not specifically a power monitoring peripheral, then it will not support energy related commands.