arendst/Tasmota #7154: Tasmota on PoW R2 - Energy consumption numbers don't add up correctly on 6.7.1 and 7.1.1

Hannu911, 2019-12-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have one Sonoff PoW R2 running with Tasmota and another POW R2 on the same line running on the factory FW. Then a third energy measuring devide (Zappi) on the same line. The reported energy consumption numbers from the Zappi and from the POW R2 with factory FW are always the same. Also the energy consumption numbers reported from the POW R2 with Tasmota FW were always correct when I had 6.5.0 version in it. But when I put the 6.7.1 version fw in, quite often the reported consumption is a lot less than what Zappi and POW R2 with factory FW is reporting, but not always.
The issues did not change with the upgrade from Tasmota fw 6.7.1 to 7.1.1.

The reported power seems to be always consistent between all the three devices, so the issue is probably somewhere how the calculates the consumption based on time.

POW R2 with Tasmota reported 3.296kWh for Yesterday
12:01:26 MQT: tele/sonoff-32A633/SENSOR = {"Time":"2019-12-09T12:01:26","ENERGY":{"TotalStartTime":"2019-05-06T14:20:29","Total":1368.188,"Yesterday":3.296,"Today":0.023,"Period":0.02,"Power":1.83,"ApparentPower":9.25,"ReactivePower":9.10,"Factor":0.20,"Voltage":236.0,"Current":0.039}}

Zappi and POW R2 with factory fw reported for yesterday 9.800kWh

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff POW R2_____
- [x] Tasmota binary firmware version number used: 7.1.1_____
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Upgrade by file upload on Crome browser interface. tasmota.bin_____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
12:30:16 MQT: stat/sonoff-32A633/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
12:30:16 MQT: stat/sonoff-32A633/RESULT = {"Module":{"43":"Sonoff Pow R2"}}
12:30:16 MQT: stat/sonoff-32A633/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"145":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"146":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
12:31:18 MQT: stat/sonoff-32A633/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
12:31:18 MQT: stat/sonoff-32A633/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
12:31:18 MQT: stat/sonoff-32A633/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
12:32:15 MQT: stat/sonoff-32A633/STATUS = {"Status":{"Module":43,"FriendlyName":["SonoffT8"],"Topic":"sonoff-32A633","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:32:15 MQT: stat/sonoff-32A633/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"3T23:54:21","StartupUTC":"2019-12-05T10:37:54","Sleep":50,"CfgHolder":4617,"BootCount":35,"SaveCount":441,"SaveAddress":"F6000"}}
12:32:16 MQT: stat/sonoff-32A633/STATUS2 = {"StatusFWR":{"Version":"7.1.1(tasmota)","BuildDateTime":"2019-12-01T13:00:09","Boot":7,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
12:32:16 MQT: stat/sonoff-32A633/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FAM-727",""],"TelePeriod":30,"Resolution":"559380C0","SetOption":["00008009","2805C8000100068000005AFF000000000000","00000000","00000000"]}}
12:32:16 MQT: stat/sonoff-32A633/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
12:32:16 MQT: stat/sonoff-32A633/STATUS5 = {"StatusNET":{"Hostname":"sonoff-32A633-1587","IPAddress":"192.168.18.7","Gateway":"192.168.18.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.18.1","Mac":"80:7D:3A:32:A6:33","Webserver":2,"WifiConfig":4}}
12:32:16 MQT: stat/sonoff-32A633/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.18.33","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_32A633","MqttUser":"DVES_USER","MqttCount":11,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
12:32:16 MQT: stat/sonoff-32A633/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 09 10:32:16 2019","Local":"Mon Dec 09 12:32:16 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+02:00","Sunrise":"09:31","Sunset":"17:53"}}
12:32:16 MQT: stat/sonoff-32A633/STATUS9 = {"StatusPTH":{"PowerDelta":25,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:32:16 MQT: stat/sonoff-32A633/STATUS10 = {"StatusSNS":{"Time":"2019-12-09T12:32:16","ENERGY":{"TotalStartTime":"2019-05-06T14:20:29","Total":1368.189,"Yesterday":3.296,"Today":0.024,"Power":1.80,"ApparentPower":9.21,"ReactivePower":9.00,"Factor":0.20,"Voltage":237.2,"Current":0.039}}}
12:32:16 MQT: stat/sonoff-32A633/STATUS11 = {"StatusSTS":{"Time":"2019-12-09T12:32:16","Uptime":"3T23:54:22","UptimeSec":345262,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":11,"POWER":"ON","Wifi":{"AP":1,"SSId":"FAM-727","BSSId":"20:78:52:CB:3F:3C","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-09:
https://github.com/arendst/Tasmota/wiki/Power-Monitoring-Calibration

ascillato2, 2019-12-09:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-09 by ascillato2

Hannu911, 2019-12-09:
> Lähetetty: maanantai 9. joulukuuta 2019 13.49
> Vastaanottaja: arendst/Tasmota <Tasmota@noreply.github.com>
> Kopio: Hannu911 <hannu911@live.fi>; Author <author@noreply.github.com>
>
> https://github.com/arendst/Tasmota/wiki/Power-Monitoring-Calibration

Hello

I have already done the calibration. Power, Voltage and Current numbers are correct but for some reason the energy consumption numbers are not.

[cid:d56b8b2f-0a50-4522-bff6-e4d2c7c024f4]

Regards,
Hannu

Hannu911, 2019-12-09:
Why did you close this issue? This is not a dublicate. The 1st one was a dubclicate