arendst/Tasmota #7155: PoW R2 - Energy consumption numbers don't add up correctly on 6.7.1 and 7.1.1

Hannu911, 2019-12-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have one Sonoff PoW R2 running with Tasmota and another POW R2 on the same line running on the factory FW. Then a third energy measuring devide (Zappi) on the same line. The reported energy consumption numbers from the Zappi and from the POW R2 with factory FW are always the same. 
The energy consumption numbers reported from the POW R2 with Tasmota FW were always correct when I had 6.5.0 version in it. 
**But when I put the 6.7.1 version fw in, quite often the reported consumption is a lot less than what Zappi and POW R2 with factory FW is reporting, but not always.
The issues did not change with the upgrade from Tasmota fw 6.7.1 to 7.1.1.**

 **I have calibrated the POW and it is reporting correct power, voltage and current values!!!**

The reported power seems to be always consistent between all the three devices, so the issue is probably somewhere how the calculates the consumption based on time.

**POW R2 with Tasmota reported 3.296kWh for Yesterday**
12:01:26 MQT: tele/sonoff-32A633/SENSOR = {"Time":"2019-12-09T12:01:26","ENERGY":{"TotalStartTime":"2019-05-06T14:20:29","Total":1368.188,"Yesterday":3.296,"Today":0.023,"Period":0.02,"Power":1.83,"ApparentPower":9.25,"ReactivePower":9.10,"Factor":0.20,"Voltage":236.0,"Current":0.039}}

**Zappi and POW R2 with factory fw reported for yesterday 9.800kWh**

So the difference is quite a big one.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): POW R2
- [x] Tasmota binary firmware version number used: 7.1.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: Upgrade by file upload on Crome browser interface. tasmota.bin
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
12:30:16 MQT: stat/sonoff-32A633/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
12:30:16 MQT: stat/sonoff-32A633/RESULT = {"Module":{"43":"Sonoff Pow R2"}}
12:30:16 MQT: stat/sonoff-32A633/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"145":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"146":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
12:31:18 MQT: stat/sonoff-32A633/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
12:31:18 MQT: stat/sonoff-32A633/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
12:31:18 MQT: stat/sonoff-32A633/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
12:32:15 MQT: stat/sonoff-32A633/STATUS = {"Status":{"Module":43,"FriendlyName":["SonoffT8"],"Topic":"sonoff-32A633","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:32:15 MQT: stat/sonoff-32A633/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"3T23:54:21","StartupUTC":"2019-12-05T10:37:54","Sleep":50,"CfgHolder":4617,"BootCount":35,"SaveCount":441,"SaveAddress":"F6000"}}
12:32:16 MQT: stat/sonoff-32A633/STATUS2 = {"StatusFWR":{"Version":"7.1.1(tasmota)","BuildDateTime":"2019-12-01T13:00:09","Boot":7,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
12:32:16 MQT: stat/sonoff-32A633/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FAM-727",""],"TelePeriod":30,"Resolution":"559380C0","SetOption":["00008009","2805C8000100068000005AFF000000000000","00000000","00000000"]}}
12:32:16 MQT: stat/sonoff-32A633/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
12:32:16 MQT: stat/sonoff-32A633/STATUS5 = {"StatusNET":{"Hostname":"sonoff-32A633-1587","IPAddress":"192.168.18.7","Gateway":"192.168.18.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.18.1","Mac":"80:7D:3A:32:A6:33","Webserver":2,"WifiConfig":4}}
12:32:16 MQT: stat/sonoff-32A633/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.18.33","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_32A633","MqttUser":"DVES_USER","MqttCount":11,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
12:32:16 MQT: stat/sonoff-32A633/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 09 10:32:16 2019","Local":"Mon Dec 09 12:32:16 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+02:00","Sunrise":"09:31","Sunset":"17:53"}}
12:32:16 MQT: stat/sonoff-32A633/STATUS9 = {"StatusPTH":{"PowerDelta":25,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:32:16 MQT: stat/sonoff-32A633/STATUS10 = {"StatusSNS":{"Time":"2019-12-09T12:32:16","ENERGY":{"TotalStartTime":"2019-05-06T14:20:29","Total":1368.189,"Yesterday":3.296,"Today":0.024,"Power":1.80,"ApparentPower":9.21,"ReactivePower":9.00,"Factor":0.20,"Voltage":237.2,"Current":0.039}}}
12:32:16 MQT: stat/sonoff-32A633/STATUS11 = {"StatusSTS":{"Time":"2019-12-09T12:32:16","Uptime":"3T23:54:22","UptimeSec":345262,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":11,"POWER":"ON","Wifi":{"AP":1,"SSId":"FAM-727","BSSId":"20:78:52:CB:3F:3C","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-09:
Let's make the calculation manually.

Which device is? 
How much current use? 
Which is the rated W of that device? 
How much time do you have it turned on yesterday?

ascillato, 2019-12-09:
@Hannu911 I'm asking that, because the type of load you are testing has a very low power factor (0.2) So, may be you are seeing Apparent Power (kVA) in those other devices while you are seeing Active Power (kW) in Tasmota (https://en.wikipedia.org/wiki/AC_power)

I have several power metering devices but controlling lights, and the yesterday and today measurements are ok doing the simple math ( Active Energy = V * I * PF * RunningHours ) In my case PF is close to 1.

Please share more information to troubleshoot your issue:

* Which device is your load?
* How much current it uses?
* Which is the rated W of that device?
* How much time do you have it turned on yesterday?

Hannu911, 2019-12-09:
> Lähetetty: maanantai 9. joulukuuta 2019 17.35
> Vastaanottaja: arendst/Tasmota <Tasmota@noreply.github.com>
> Kopio: Hannu911 <hannu911@live.fi>; Author <author@noreply.github.com>

Hello Adrian

I attach the file from the sonoff for 8th of December
The other one has all data that sonoff echoes and the other one only power and total.

The device is Sonoff POW R2

It measures Volvo T8 charging and the charging current is 16A/3.7kW at max on 230V. But sometimes the charging power is aroung 1.8kW
[cid:83a47ddb-b973-4dce-9ac0-6192ee335b28]
(x-axle is days in decimal number)

Regards,
Hannu

Hannu911, 2019-12-09:
Yes, the power factor is low when there is no load but when the 3.7kW load is on the power factor is 1.0

Lähetetty: maanantai 9. joulukuuta 2019 19.05

Kopio: Hannu911 <hannu911@live.fi>; Mention <mention@noreply.github.com>

ascillato, 2019-12-09:
How did you perform the calibration? with a resistive load?

Hannu911, 2019-12-09:
Yes, with resistive load. And I have to remind that the enegry consumption was reported correctly with 6.5.0 fw. and sometimes is reports energy correctly also with 6.7.1 and 7.1.1 fw. 
All these fw versions report power, voltage, current and power factor correctly all the time. Only the energy consumption is not reported correctly

ascillato, 2019-12-12:
I have continued testing my POW R2 (v7.1.2.4) and it stills shows the correct value on yesterday, compared to a manual calculation.

Have you find anything else in order to reach the root of this issue?

Can you share more information of your setup? photos, and how is connected?

Hannu911, 2019-12-12:
It look like the energy consumption is always correct when the power is less than 3.0kW. But when the power is 3.7kW the energy calculation is not correct

Hannu911, 2019-12-12:
Date | Time | Tasmota | Zappi | diff % | kW
-- | -- | -- | -- | -- | --
1.12.2019 | 16:30 | 1.505 | 1.513 | 0.5 % | 1.3
1.12.2019 | 19:24 | 5.651 | 7.465 | 24.3 % | 3.7**
2.12.2019 | 13:45 | 4.467 | 4.496 | 0.6 % | 1.7
2.12.2019 | 16:51 | 2.651 | 2.72 | 2.5 % | 1.9
3.12.2019 | 19:24 | 1.586 | 1.581 | -0.3 % | 1.3
4.12.2019 | 9:00 | 1.85 | 1.851 | 0.1 % | 1.3
4.12.2019 | 15:57 | 4.219 | 4.205 | -0.3 % | 1.6
4.12.2019 | 20:24 | 0.638 | 3.259 | 80.4 % | 3.7**
4.12.2019 | 22:01 | 1.753 | 7.651 | 77.1 % | 3.7**
5.12.2019 | 16:27 | 4.814 | 4.756 | -1.2 % | 3
5.12.2019 | 19:51 | 0.929 | 3.647 | 74.5 % | 3.7**
6.12.2019 | 23:11 | 6.672 | 6.643 | -0.4 % | 1.6
7.12.2019 | 9:55 | 3.226 | 3.206 | -0.6 % | 1.7
7.12.2019 | 16:15 | 3.172 | 3.147 | -0.8 % | 1.7
7.12.2019 | 18:40 | 3.19 | 3.166 | -0.8 % | 1.8
8.12.2019 | 16:37 | 1.389 | 1.389 | 0.0 % | 1.6
8.12.2019 | 19:21 | 1.853 | 8.434 | 78.0 % | 3.7**
9.12.2019 | 13:36 | 1.727 | 1.721 | -0.3 % | 1.6
10.12.2019 | 12:36 | 6.309 | 6.306 | 0.0 % | 1.5
10.12.2019 | 20:58 | 3.297 | 3.288 | -0.3 % | 1.4
11.12.2019 | 14:39 | 1.356 | 1.362 | 0.4 % | 1.2
11.12.2019 | 20:22 | 4.155 | 8.112 | 48.8 % | 3.7**

Hannu911, 2019-12-12:
[image]
This is the picture of the 11.12.19 20:22 charging. I assume that the 1st half of the changing at around 3.0kW the energy has been calculated correctly but the 2nd half at 3.7kW has not.

Hannu911, 2019-12-12:
[image]
This is the picture of the 8.12.19 19:21 charging. The whole charging happens at 3.7kW and the energy error is about 80%

Hannu911, 2019-12-12:
Earlier this year I used the same POW R2 with 6.5.0 fw and the energy calculation worked correctly also at 3.7kW load. So something must have changed after that version

localhost61, 2019-12-13:
I'm confident that your table will help to troubleshoot the issue ;-)

localhost61, 2019-12-13:
Isn't the POW R2 limited to 15A and 3600W?

Hannu911, 2019-12-13:
-POW R2 specification may state that 3600W/15A but it is not limited. power and current is reported correctly at 16A and 3.7kW
-yes
-kWh
-duration is in days as a decimal number. So for example in the latter picture the charging start at 8.8 and ends at 9.8, means that the charging starts at 19:21 and ends at 21:35 on the 8th of December

arendst, 2019-12-31:
Just looking around here and can only confirm that >3.68kW things should go hay-wire. Take a look ate the code:
```
      if (cf_frequency && Energy.active_power[0])  {
        unsigned long delta = (cf_frequency * Settings.energy_power_calibration) / 36;
        // prevent invalid load delta steps even checksum is valid (issue #5789):
        if (delta <= (3680*100/36) * 10 ) {  // max load for S31/Pow R2: 3.68kW
          Cse.cf_pulses_last_time = Cse.cf_pulses;
          Energy.kWhtoday_delta += delta;
        }
        else {
          AddLog_P(LOG_LEVEL_DEBUG, PSTR("CSE: Load overflow"));
          Cse.cf_pulses_last_time = CSE_PULSES_NOT_INITIALIZED;
        }
        EnergyUpdateToday();
      }
```
There you have it.

When you enable more logging you should see the overflow message.

arendst, 2019-12-31:
The 3.68kW is questionable of course. You might want to raise it to 4000 and see what happens.

NB. The code is in file xnrg_02_cse7766.ino lines 193 - 205

djashjones, 2020-01-01:
Just for your information my Pow R2 V2.0 has a CSE7759B chip

Jason2866, 2020-01-02:
4kw with a POW. Border line... Safe would be using a PZEM with a CT

ascillato2, 2020-01-03:
@Hannu911 and @boyyao

This issue has been fixed, can you confirm that is solved for you? Thanks

Hannu911, 2020-01-03:
From where do I find the tasmota.bin file for this fix?

meingraham, 2020-01-03:
http://thehackbox.org/tasmota

closed 2020-01-03 by ascillato2

Hannu911, 2020-01-04:
The same problem still exists in the version 8.1.0.2 that I took from http://thehackbox.org/tasmota

Hannu911, 2020-01-04:
Sonoff POW with 8.1.0.2 reported only 1.274kWh when I charged my car at 3.75kW for one hour