arendst/Tasmota #7157: Multi function Button

ltagliamonte-dd, 2019-12-09:
Dear community I recently bought an outdoor smart plug from Amazon [link](https://www.amazon.com/gp/product/B07YXK9717/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1)

The smart plug look a lot like this [one](https://blakadder.github.io/templates/king-link_C128.html) from the template repository but the part number and the mapping are different.

After some spending some time mapping the GPIOs I've come up with the following template:
``` 
{"NAME":"Amzdest OSP","GPIO":[0,0,0,0,0,0,0,0,21,54,22,23,0],"FLAG":0,"BASE":18}
```
With this template the power Led is on, and all the relays are working via the tasmota webConsole.

The only missing piece is the button. Originally the button worked as follow:
- pressed once will turn relay1 ON
- pressed the second time **(NOT twice)** turn relay2 ON
- pressed the third time turn relay3 ON
- pressed the forth time turn relay1 OFF
- pressed the fifth time turn relay2 OFF
- pressed the sixth time turn relay3 OFF

For what i've understood so far I can implement a behavior like that with "Rules", keeping the state in a var.

Is there a simpler way to do it?
Did anyone already have a rule for this use case willing to share?
Thanks in advance.

closed 2019-12-09 by close-issue-app[bot]

ascillato, 2019-12-09:
Sorry, That is not supported. only one tap, double tap and hold using rules (as the example in the rules cookbook of the docs)

ltagliamonte-dd, 2019-12-09:
@ascillato thank you for the reply, is it possible to do something easier like:
press once all ON
press the second time all OFF

ascillato, 2019-12-09:
Yes, that can be achieved easily. If you need assistance on that, please, find us in the Tasmota support chat (https://discord.gg/Ks2Kzd4) Thanks.

ltagliamonte-dd, 2019-12-09:
@ascillato thank you, I already wrote on #general but this get any reply there.