arendst/Tasmota #7162: Unable to connect to wifi

edwolfe807, 2019-12-09:
### PROBLEM DESCRIPTION
A WiFi connection is not made.  the following messages are repeated over and over in the SerialLog

WIF: Checking connection...
WIF: Connected
WIF: Checking connection...
WIF: Connected

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff 4CH R2 V1.0
- [X] Tasmota binary firmware version number used: 7.1.2
  - [X] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Atom
- [X] Flashing tools used: Tried both Atom and ESP8266Flasher.exe
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
22:16:39 CMD: backlog template; module; gpio 255
22:16:39 SRC: Serial
22:16:39 CMD: Group 0, Index 1, Command "BACKLOG", Data "template; module; gpio 255"
22:16:39 SRC: Backlog
22:16:39 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""

22:16:39 RSL: stat/crealtycr10spro/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
22:16:39 SRC: Backlog
22:16:39 CMD: Group 0, Index 1, Command "MODULE", Data ""
22:16:39 RSL: stat/crealtycr10spro/RESULT = {"Module":{"23":"Sonoff 4CH Pro"}}
22:16:39 SRC: Backlog
22:16:39 CMD: Group 0, Index 1, Command "GPIO", Data "255"
22:16:39 RSL: stat/crealtycr10spro/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"23":"Relay3"}
,"GPIO5":{"22":"Relay2"},"GPIO9":{"18":"Button2"},"GPIO10":{"19":"Button3"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"20":"Button4"},"GPIO
15":{"24":"Relay4"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
 22:17:28 CMD: Backlog rule1; rule2; rule3

22:17:28 SRC: Serial
22:17:28 CMD: Group 0, Index 1, Command "BACKLOG", Data "rule1; rule2; rule3"
22:17:28 SRC: Backlog
22:17:28 CMD: Group 0, Index 1, Command "RULE", Data ""
22:17:28 RSL: stat/crealtycr10spro/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
22:17:29 SRC: Backlog
22:17:29 CMD: Group 0, Index 2, Command "RULE", Data ""
22:17:29 RSL: stat/crealtycr10spro/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
22:17:29 SRC: Backlog
22:17:29 CMD: Group 0, Index 3, Command "RULE", Data ""
22:17:29 RSL: stat/crealtycr10spro/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [X] Provide the output of this command: `Status 0`:
```
  22:18:12 CMD: Group 0, Index 1, Command "STATUS", Data "0"
22:18:12 RSL: stat/crealtycr10spro/STATUS = {"Status":{"Module":23,"FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4"],"Topic":"crealtycr10spro","Bu
ttonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonR
etain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:18:12 RSL: stat/crealtycr10spro/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota
.bin","RestartReason":"Power on","Uptime":"0T00:19:14","StartupUTC":"2019-12-09T20:58:58","Sleep":50,"CfgHolder":6000,"BootCount":1,"SaveCount":7,"SaveAddre
ss":"F5000"}}
22:18:12 RSL: stat/crealtycr10spro/STATUS2 = {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019-12-09T15:56:46","Boot":31,"Core":"2_6_1","SDK":"
2.2.2-dev(38a443e)","Hardware":"ESP8285"}}
22:18:12 RSL: stat/crealtycr10spro/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WolfeDen-Auto

mation","WolfeDen-House2.4"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}
}
22:18:12 RSL: stat/crealtycr10spro/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":29,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14
4051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,2
6,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
22:18:12 RSL: stat/crealtycr10spro/STATUS5 = {"StatusNET":{"Hostname":"crealtycr10spro-1937","IPAddress":"192.168.10.106","Gateway":"192.168.10.1","Subnetma
sk":"255.255.255.0","DNSServer":"172.32.0.250","Mac":"C4:4F:33:B1:47:91","Webserver":2,"WifiConfig":2}}
22:18:12 RSL: stat/crealtycr10spro/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B14791","MqttUser":
"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:18:12 RSL: stat/crealtycr10spro/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 09 21:18:12 2019","Local":"Mon Dec 09 22:18:12 2019","StartDST":"Sun Mar 31 02:00:
00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:31","Sunset":"16:53"}}
22:18:12 RSL: stat/crealtycr10spro/STATUS10 = {"StatusSNS":{"Time":"2019-12-09T22:18:12"}}
22:18:12 RSL: stat/crealtycr10spro/STATUS11 = {"StatusSTS":{"Time":"2019-12-09T22:18:12","Uptime":"0T00:19:14","UptimeSec":1154,"Heap":29,"SleepMode":"Dynam
ic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"WolfeDen-Automation","BSSId":"8
0:3F:5D:97:06:49","Channel":1,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
The ip address should by a 172.31.10.x address
21:59:25 WIF: Checking connection...
21:59:25 WIF: Connected
21:59:32 CMD: ap
21:59:32 SRC: Serial
21:59:32 CMD: Group 0, Index 1, Command "AP", Data ""

21:59:32 RSL: stat/crealtycr10spro/RESULT = {"Ap":{"1":"WolfeDen-Automation"}}
21:59:37 CMD: ipaddress
21:59:37 SRC: Serial
21:59:37 CMD: Group 0, Index 1, Command "IPADDRESS", Data ""
21:59:37 RSL: stat/crealtycr10spro/RESULT = {"IPAddress1":"(IP unset) (192.168.10.106)"}
21:59:45 WIF: Checking connection...
21:59:45 WIF: Connected
22:00:05 WIF: Checking connection...
22:00:05 WIF: Connected
22:00:25 WIF: Checking connection...
22:00:25 WIF: Connected
22:00:45 WIF: Checking connection...
22:00:45 WIF: Connected
22:01:05 WIF: Checking connection...
22:01:05 WIF: Connected

```

### TO REPRODUCE
Just flashed it with pre-built and self built same results

### EXPECTED BEHAVIOUR
Never connects to wifi (confirmed by look at DCHP entries on router)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-09:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-09 by ascillato2

edwolfe807, 2019-12-09:
This looks like a bug to me as the precompiled bin file from github fails to work correctly.