arendst/Tasmota #7165: H801 controller startup issue

scargill, 2019-12-09:
### PROBLEM DESCRIPTION
In Tasmota the H801 controller flahes LEDS attached, briefly at full brilliance on powerup.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
No rules involved

```
- [ ] Provide the output of this command: `Status 0`:
```
 Status is fine.

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
 Output is fine.

```

### TO REPRODUCE
Flash Tasmota 7.1.2 onto an H801 controller

### EXPECTED BEHAVIOUR
Power up with levels as set last time the unit wsa powered up

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Hi guys

Simple one here, maybe I just need enlightening if this is not new. I just received an H801 controller from AliExpress and with a suitably powerful supply and 5m of 12v RGB LED STRIP, I went into the deep end and installed Tasmota (7.1.2 release). It works, no problem but opens 2 questions.

On powerup the LEDs flash brightly for an instant - is that within Tasmota's remit or an issue with the controller?

TDM (I know that's another subject but I'll mention it while I'm here)... RGB + brilliance is great - and easy (top left of TDM screen) - but annoying that when you change colour in TDM - the brilliance goes to hell and you have to set the latter again. Wouldn't it be nice to be able to mess with the hue and leave the brightness intact?

Just trying to get the best out of already brilliant software.

Regards

Pete

s-hadinger, 2019-12-10:
The flash is not normal. That's why we need `Status 0` to check how you configured it. 

What is your `PowerOnState` and make sure you are not using rules.

scargill, 2019-12-10:
Hi there s-hadinger

I just turned the unit on after leavng it on dim green last night and powering off (but leaving TDM on).

Here is the result of staus 0.

[08:07:36] cmnd/h801/status 0
[08:07:36] stat/h801/STATUS {"Status":{"Module":20,"FriendlyName":["h801"],"Topic":"h801","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
[08:07:36] stat/h801/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:00:37","StartupUTC":"2019-12-10T08:07:06","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":76,"SaveAddress":"F8000"}}
[08:07:36] stat/h801/STATUS2 {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019-12-06T11:22:22","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
[08:07:36] stat/h801/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["devices",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
[08:07:36] stat/h801/STATUS4 {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
[08:07:36] stat/h801/STATUS5 {"StatusNET":{"Hostname":"h801","IPAddress":"192.168.14.137","Gateway":"192.168.14.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.14.1","Mac":"84:F3:EB:E0:D2:40","Webserver":2,"WifiConfig":4}}
[08:07:36] stat/h801/STATUS6 {"StatusMQT":{"MqttHost":"192.168.14.71","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E0D240","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
[08:07:36] stat/h801/STATUS7 {"StatusTIM":{"UTC":"Tue Dec 10 08:07:43 2019","Local":"Tue Dec 10 09:07:43 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:32","Sunset":"16:53"}}
[08:07:36] stat/h801/STATUS10 {"StatusSNS":{"Time":"2019-12-10T09:07:43"}}
[08:07:36] stat/h801/STATUS11 {"StatusSTS":{"Time":"2019-12-10T09:07:43","Uptime":"0T00:00:37","UptimeSec":37,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":89,"MqttCount":1,"POWER":"ON","Dimmer":46,"Color":"0875000000","HSBColor":"116,100,46","Channel":[3,46,0,0,0],"CT":153,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"devices","BSSId":"00:A6:C6:2C:41:B2","Channel":11,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:11"}}}

No rules, everything at defaults - not messedwith any controls or rules yet. if I can provide any moer info let me know as I could see this as very useful. I'm using 5050 strip, no W or WW, jst RSB, on a 6 amp PSU set to 12v out.

Anything you want me to try, fire away. 

Pete

scargill, 2019-12-10:
To add to my woes... I upgraded to 7.1.2.4 just for the sake of it - and all the settings including EIFI went AWOL after powering off and on 4 times. The lights started flahing green. This is with a known good (large toroid) PSU able to output 6 amps  at up to 60v, with a 12v, 2 amp limit set in place. Max strip consumption is 1.6 amps.  I checked, the 801 unit was back to using it's own access point and had lost all the configuration. I put it back in and all is well. I wonder where these issues lie?

blakadder, 2019-12-10:
I thought that was the default behaviour for H801 and it has been in the H801 article for ages.

@scargill  Power cycling your device 4 times in quick succession activates the [Power cycle recovery](https://tasmota.github.io/docs/#/help/Device-Recovery?id=fast-power-cycle-device-recovery)

scargill, 2019-12-10:
It could well have been - I'm very new (as of yesterday) to the H801 and clearly not read everything there is to know about it.

s-hadinger, 2019-12-10:
Could you please set `weblog 4` and paste the boot log, until the flash. Is the flash just at power up or when connecting Mqtt?

scargill, 2019-12-10:
Sure... powered up I put the weblog 4 into the console

[16:00:16] cmnd/h801/weblog 4
[16:00:16] stat/h801/RESULT {"WebLog":4}

I turned off - then on,

[16:00:16] cmnd/h801/weblog 4
[16:00:16] stat/h801/RESULT {"WebLog":4}
[16:02:00] tele/h801/LWT Offline
[16:02:00] tele/h801/LWT Online
[16:02:00] cmnd/h801/POWER 
[16:02:00] tele/h801/INFO1 {"Module":"H801","Version":"7.1.2.4(3d2e77f-tasmota)","FallbackTopic":"cmnd/DVES_E0D240_fb/","GroupTopic":"cmnd/tasmotas/"}
[16:02:00] tele/h801/INFO2 {"WebServerMode":"Admin","Hostname":"h801","IPAddress":"192.168.14.137"}
[16:02:00] tele/h801/INFO3 {"RestartReason":"External System"}
[16:02:00] stat/h801/RESULT {"POWER":"ON"}
[16:02:00] stat/h801/POWER ON
[16:02:00] cmnd/h801/status 0
[16:02:00] cmnd/h801/modules 
[16:02:00] cmnd/h801/gpios 255
[16:02:00] cmnd/h801/switchdebounce 
[16:02:00] cmnd/h801/blinktime 
[16:02:00] cmnd/h801/mqttlog 
[16:02:00] cmnd/h801/pulsetime2 
[16:02:00] cmnd/h801/pulsetime4 
[16:02:00] cmnd/h801/pulsetime6 
[16:02:00] cmnd/h801/pulsetime8 
[16:02:01] stat/h801/STATUS {"Status":{"Module":20,"FriendlyName":["h801"],"Topic":"h801","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
[16:02:01] cmnd/h801/status 0
[16:02:01] stat/h801/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:00:21","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":16,"SaveCount":46,"SaveAddress":"F6000"}}
[16:02:01] stat/h801/STATUS2 {"StatusFWR":{"Version":"7.1.2.4(3d2e77f-tasmota)","BuildDateTime":"2019-12-09T19:00:09","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"239/1151"}}
[16:02:01] stat/h801/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["office",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
[16:02:01] cmnd/h801/modules 
[16:02:01] cmnd/h801/gpios 255
[16:02:01] cmnd/h801/switchdebounce 
[16:02:01] cmnd/h801/blinktime 
[16:02:01] cmnd/h801/mqttlog 
[16:02:01] cmnd/h801/pulsetime2 
[16:02:01] cmnd/h801/pulsetime4 
[16:02:01] cmnd/h801/pulsetime6 
[16:02:01] cmnd/h801/pulsetime8 
[16:02:01] cmnd/h801/template 
[16:02:01] cmnd/h801/buttondebounce 
[16:02:01] cmnd/h801/blinkcount 
[16:02:01] cmnd/h801/pulsetime3 
[16:02:01] cmnd/h801/pulsetime7 
[16:02:01] stat/h801/STATUS4 {"StatusMEM":{"ProgramSize":563,"Free":440,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
[16:02:01] stat/h801/STATUS5 {"StatusNET":{"Hostname":"h801","IPAddress":"192.168.14.137","Gateway":"192.168.14.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.14.1","Mac":"84:F3:EB:E0:D2:40","Webserver":2,"WifiConfig":4}}
[16:02:01] stat/h801/STATUS6 {"StatusMQT":{"MqttHost":"192.168.14.71","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E0D240","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
[16:02:01] stat/h801/STATUS7 {"StatusTIM":{"UTC":"Thu Jan 01 00:00:20 1970","Local":"Thu Jan 01 00:00:20 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
[16:02:01] stat/h801/STATUS10 {"StatusSNS":{"Time":"1970-01-01T00:00:20"}}
[16:02:01] cmnd/h801/template 
[16:02:01] cmnd/h801/buttondebounce 
[16:02:01] cmnd/h801/blinkcount 
[16:02:01] cmnd/h801/pulsetime3 
[16:02:01] cmnd/h801/pulsetime7 
[16:02:01] stat/h801/STATUS11 {"StatusSTS":{"Time":"1970-01-01T00:00:20","Uptime":"0T00:00:21","UptimeSec":21,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":79,"MqttCount":1,"POWER":"ON","Dimmer":28,"Color":"0018470000","HSBColor":"220,100,28","Channel":[0,9,28,0,0],"CT":153,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"office","BSSId":"00:1B:29:66:D1:B8","Channel":11,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:20"}}}
[16:02:01] stat/h801/RESULT {"Modules1":{"0":"Generic","1":"Sonoff Basic","2":"Sonoff RF","4":"Sonoff TH","5":"Sonoff Dual","39":"Sonoff Dual R2","6":"Sonoff Pow","43":"Sonoff Pow R2","7":"Sonoff 4CH","23":"Sonoff 4CH Pro","41":"Sonoff S31","8":"Sonoff S2X","10":"Sonoff Touch","28":"Sonoff T1 1CH","29":"Sonoff T1 2CH","30":"Sonoff T1 3CH","11":"Sonoff LED","22":"Sonoff BN-SZ","70":"Sonoff L1","26":"Sonoff B1","9":"Slampher","21":"Sonoff SC","44":"Sonoff iFan02","71":"Sonoff iFan03","25":"Sonoff Bridge","3":"Sonoff SV","19":"Sonoff Dev","12":"1 Channel","13":"4 Channel","14":"Motor C/AC","15":"ElectroDragon","16":"EXS Relay(s)"}}
[16:02:01] stat/h801/RESULT {"Modules2":{"31":"Supla Espablo","35":"Luani HVIO","33":"Yunshan Relay","17":"WiOn","46":"Shelly 1","47":"Shelly 2","45":"BlitzWolf SHP","52":"Teckin","59":"Teckin US","53":"AplicWDP303075","55":"Gosund SP1 v23","65":"Luminea ZX2820","57":"SK03 Outdoor","63":"Digoo DG-SP202","64":"KA10","67":"SP10","68":"WAGA CHCZ02MB","49":"Neo Coolcam","51":"OBI Socket","61":"OBI Socket 2","60":"Manzoku strip","50":"ESP Switch","54":"Tuya MCU","56":"ARMTR Dimmer","58":"PS-16-DZ","20" [... cut ...]
[16:02:05] tele/h801/STATE {"Time":"2019-12-10T17:02:06","Uptime":"0T00:00:26","UptimeSec":26,"Heap":28,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":28,"Color":"0018470000","HSBColor":"220,100,28","Channel":[0,9,28,0,0],"CT":153,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"office","BSSId":"00:1B:29:66:D1:B8","Channel":11,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:20"}}

And I'm not usin MQTT - just the console - the flash is immediately the unt comes on.

Hope that helps. And yes, the power supply is WELL able to handle the LEDs...

Pete

florian-asche, 2019-12-10:
Hi, 
i just tested this, and i can confirm your issue. If i power on the H801 its lightning up shortly. I am also using an RGBWW self made board from the FHEM Forum. That doesnt do that.
Florian

s-hadinger, 2019-12-10:
@scargill I have no clue what you are doing with your device, but it received 41 commands in less than 2 seconds.

Your claim you are not using MQTT but it seems it is configured and connected: `{"MqttHost":"192.168.14.71","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E0D240","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}`

blakadder, 2019-12-10:
As previously stated, H801 flashing on power-up is due to the circuit design and cannot be corrected in Tasmota. 
A cursory search of the issues would reveal this question has been already asked a few times and the answer was always the same.

This issue, on the virtue of being a non-issue, is now closed!

closed 2019-12-10 by blakadder

florian-asche, 2019-12-10:
Hey Peter, maybe you can find the solution in the technical details here: https://github.com/patrickjahns/esp_rgbww_controller

Thats the controller that is working for me.

florian-asche, 2019-12-10:
Also feel free to contact me if you like. ;)

s-hadinger, 2019-12-10:
Actually I think I found the issue. It looks like when GPIOSs are in PWM mode, the PWM GPIO are not initialized until `LightAnimate()` is called 50 ms after start. They remain in undefined state until they are pulled down by `LightAnimate()`.

Let me try a PR to force the PMW pins are low or high (for PMWi) at start.

blakadder, 2019-12-10:
Apparently a wizard can fix it despite this being deemed impossible in previous issues

joba-1, 2019-12-10:
Hope you are right. It just reminded me of my garage door magically opening all by itself due to not so clever gpio init during startup after power loss...

scargill, 2019-12-11:
> @scargill I have no clue what you are doing with your device, but it received 41 commands in less than 2 seconds.
>
> Your claim you are not using MQTT but it seems it is configured and connected: [...]

I'n not doing anything with MQTT on it. I do set up MQTT on all my devices then control them in Node-Red via the MQTT nodes but on this device so far all it is doing is setting up the MQTT user, IP, port and passwor details in the config. That's it.

scargill, 2019-12-11:
> Hey Peter, maybe you can find the solution in the technical details here: https://github.com/patrickjahns/esp_rgbww_controller
> 
> Thats the controller that is working for me.

Thanks for that. As it happens I get PCB companies harrasssing me all the time for sponsorship andI keep telling them I only write about things I've tried personally - so getting some of those PCBs made would be no problem , but at end user prices by the time you get the ESP, the Mosfets, the large block whatever it is,, the connectors, any passives and put them in a suitably slotted box, interesting though it may be..... That H801 complete inc post was only £6 (7.12 Euros) from China. A shame if that flashing is unavoidable but thinkin about it, it is only when power is applied... if the power is there permanently and Tasmota turns the units on and off - I guess it is worth puttng up with.

PR #7177 (2019-12-10): Fix flashing H801 led at boot (#7165, #649)
## Description:

In light mode, PWM GPIO were not initialized until `LightAnimate()` is called 50ms after the boot. It seems that leaving GPIOs in undefined state sets light on H801, but has no effect on other led controllers. Maybe a MOSFET effect (I don't know enough about MOSFETs).

Now PWM GPIOs are set to low value (Off) right at start and may change to the desired value 50ms later when `LightAnimate()` is called.

**Related issue (if applicable):** fixes #7165, #649

## Checklist:
  - [x] The pull request is done against the latest dev branch
  [... cut]