arendst/Tasmota #7169: PulseTime not usable by MQTT at 7.1.x

sunfingerde, 2019-12-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
on 7.1.x PulseTime seems to be changed and not implemented right.
On command line PulseTime still has the same syntax (e.g. PulseTime1 145)
On MQTT new objects are created
before: PulseTime1
after: PulseTime1_set & PulseTime1_remain

But I can't use PulseTime1_set by MQTT - Tasmota reports an unknown command.
When sending MQTT message with PulseTime1 tasmota accept it but shadow will be updated for PulseTime1_set, not for PulseTime1

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): doesn't matter (ESP01, WemosD1, Sonoff Dual R2, ...)
- [ x] Tasmota binary firmware version number used: 7.1.2
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: firmware upgrade web frontend
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
using PulseTime by MQTT

### EXPECTED BEHAVIOUR
When PulseTimeX has changed to PulseTimeX_set I should be able to use PulseTimeX_set

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

localhost61, 2019-12-10:
For me your issue is not clear, please add some minimal context info to help the understanding. Like a quoted weblog extracted from the console.

ascillato, 2019-12-10:
Hi, 

I also don't understand your issue. Just tested and pulsetime command is working fine. Besides PulseTime1_set & PulseTime1_remain are not Tasmota commands. I don't know where you see those.

ascillato, 2019-12-10:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-10 by ascillato2

McVareck, 2020-04-20:
I had the same problem and I solved adding a command line in the configuration.yaml of Home Assistant like this:

[image]

ascillato, 2020-04-20:
Pulsetime is a configuration, not a command. Pulsetime is needed to be executed in the Tasmota console just once to set the time where it is going to turn off the relay

McVareck, 2020-04-20:
> Pulsetime is a configuration, not a command. Pulsetime is needed to be executed in the Tasmota console just once to set the time where it is going to turn off the relay

ok, and how can I execute only the first time in Tasmota?

ascillato, 2020-04-20:
By the Tasmota console as all other configs.

Or by http.

Or by serial. 

Or by mqtt if you prefer. As you are using home assistant, you can go to the developer tab, then mqtt and put the topic and command. 
Topic: `cmnd/Portonne_cancelletto/pulsetime`
Message:
`10`

If you want to see the response there, you can add # in the listen topic and click start listen before sending any command. The # means everything.