arendst/Tasmota #7170: Make weight change of 4 grams for Sensor34 configurable

tobox, 2019-12-10:
**Have you looked for this feature in other issues and in the wiki?**  

Yes.

**Is your feature request related to a problem? Please describe.**  

My HX711 weighs a running motor that creates vibration, and a tank. I need an estimate of the tank level. The vibrations of the motor cause the readings to jump around 10g, so raising the fixed delta from 4 grams to 15 grams would be helpful.

**Describe the solution you'd like**  

Have a new setting to make the 4 grams hysteresis configurable.

**Describe alternatives you've considered**  

I changed the value in the code and compiled myself, but I'd rather run unmodified Tasmota. 

**Additional context**  

The whole Sensor34 documentation seems to be cluttered... The new search in the new Wiki at https://tasmota.github.io/docs/#/Home does not find any results for HX711. The only info about calbration etc. I found was in (already closed) github issues. What would be the correct location to document Sensor34 / HX711 in more detail than the "Commands" page?

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-12-10:
A page for sensor34/HX711 should be added to the [peripherals](https://tasmota.github.io/docs/#/peripherals/)

tobox, 2019-12-11:
If you assign a settings location, I would volunteer to do the coding. One byte should suffice. 
Either 
* 0-255 -> 0-255 grams 
* 0-99 -> 0-99g (1g increments) and 100-255 -> 100g-2550g (10g increments)
* 0: 4grams (backwards compatibility), 1-255: 0-254 grams
* 0: 4grams (backwards compatibility), 1-100 -> 1-99g (1g increments) and 101-255 -> 110g-2550g (10g increments)

arendst, 2019-12-11:
You've got it.

Use the last option with backward compatibility pls.

tobox, 2019-12-11:
It seems to be working correctly. But one question: What exactly is the code in Line 365 supposed to do? Line 361 checks for the delta and sets a flag. The next call to HxEvery100mSecond could then send the JSON message - is this some form of rate limiting? Suppose the weight value changes every 100ms by more than the delta, the else part would never be executed!

https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_34_hx711.ino#L365

arendst, 2019-12-11:
If the weight changes back below the delta it won't send a message. 

It makes sure that the weight changed over the delta at least 200mSec

tobox, 2019-12-11:
But if the weight increases by 5 grams each 100ms, the message would never be sent. Correct?

arendst, 2019-12-11:
If delta is 4 and the weight changed over 4 during 200mSec it should send a message.

tobox, 2019-12-11:
But it doesn't, because sending of the message is done in the else part, which is only executed if the change was below the delta. (or maybe I am just confused)

arendst, 2019-12-11:
- Well the first time over 4 it sets ``Hx.weight_changed = true``. 
- The second time when weight has not changed it will execute the else part as the if part now is less than 4

tobox, 2019-12-11:
The more I look at it (and test my code) the more I think there is something wrong. Almost the whole body of HxEvery100mSecond() runs only once per second, because of the if command in 

https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_34_hx711.ino#L283

So if the measured value is over the delta, the flag Hx.weight_changed is set. Then after 10 measurements a new average is calculated, and the JSON message is only sent, if the new average is exactly the same as the value that triggered the weight_changed flg. 

Anyway: I checked in my code (and left this logic the way it was) and created a PR. Please have someone look at the code first before merging it.

PR #7188 (2019-12-12): Make WeightDelta for JSON message configurable.
## Description:

Make weight delta for JSON message configurable

**Related issue (if applicable):** fixes #7170 

## Checklist:
  - [X] The pull request is done against the latest dev branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR.
  - [X] The code change is tested and works on core 2.6.1
  - [X] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [X] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement [... cut]
closed 2019-12-12 by arendst