arendst/Tasmota #7173: DS18B20 stopped working with 7.2.2

h4nc, 2019-12-10:
### PROBLEM DESCRIPTION
When I went to 7.1.0 my DS18B20 Sensor stopped working. After updating to 7.1.1 it worked again.
Now I updated to 7.1.2 and doesn't show up again. Also with the current dev version 7.1.2.4 it doesn't work.

But the sensor value doesn't show up any more and is also not available via mqtt.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Esp8266 USB module
- [ ] Tasmota binary firmware version number used: 7.2.1
  - [ ] Pre-compiled; yes
  - [ ] Self-compiled; tried both
    - [ ] IDE / Compiler used: Platformio
- [ ] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
16:27:11 MQT: stat/sonoff_one/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
16:27:11 MQT: stat/sonoff_one/RESULT = {"Module":{"18":"Generic"}}
16:27:12 MQT: stat/sonoff_one/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"37":"PWM1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"4":"DS18x20"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

`16:28:18 MQT: stat/sonoff_one/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":428,"Rules":"on POWER1#STATE=1 do backlog dimmer 10; delay 5; dimmer 18; delay 5; dimmer 0 endon"}`

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
-

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
install 7.1.2

### EXPECTED BEHAVIOUR
show the temperature value, as before

### SCREENSHOTS
-

### ADDITIONAL CONTEXT
-

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-10:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-10 by ascillato2

h4nc, 2019-12-10:
Ok, already tried to get help in the support chat with this, but will try again.