arendst/Tasmota #7176: Ideas for new functions

Aint-Nobody, 2019-12-10:
1. it will be great to have a command/rule like "if ping [1.2.3.4 and/or 4.5.6.7] true/false do ...".
Example: So it will be possible, when you leave your home with your smartphone, to detect if you at home or not. So you can automate turn off all you have to turn off. Or turn on/off your homealarm or some other stuff.

2.expand the command websend in this way, that i can define what will be send.
(Like as unix wget)
So it will be possible for example: to control my ipcam turn on or off ftp upload ect. Or turn around PTZ to different positions. Or some other stuff.

**Have you looked for this feature in other issues and in the wiki?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-10:
Ping was already requested (see old issues) but noone got enough interested on spending time programming a driver for that due with a home automation software, all those functions are already implemented and working fine with much more features that can be added to Tasmota. That is why, those types of automations are recommended to be configured in your Home Automation Software.

meingraham, 2019-12-10:
Perhaps the Ping capability can be implemented with a dedicated slave (via TasmotaSlave) but Tasmota isn't built to do this itself in its normal processing loop. You would have to think of this as a "sensor" that is sending telemetry to Tasmota and what the sensor is doing is pinging a set of resources to determine presence. Then you can use Tasmota with a Rule that triggers when the "sensor" meets the rule trigger condition.

However, by the time you do this, you are in essence offloading this presence detection to another entity... and that might as well be a script running on a server, your PC, etc. For example, in a HAB (openHAB, Home Assistant, etc.). Then the HAB can send this presence status to Tasmota (any and all devices you want included) without taxing each Tasmota device with this extra task... and load.

As for your second feature, what do you mean "in this way"? Send to multiple IPs? You can do this with a Backlog and multiple WebSend. But perhaps I don;t understand what you are suggesting.

Net-net, I think this really ends up being a Level 2 HAB function.

Aint-Nobody, 2019-12-10:
@meingraham

the second feature means, websend has it own structure. i can not change what will be send.
what behind the slash is "http://<ip>/cm?cmnd=POWER1 ON" i cant define.
so i would send a string like this for example:
http://10.1.1.11/decoder_control.cgi?command=94&user=admin&pwd=1234

Sure, in a homeautomation software you can do all u want.
But if i have only 3-8 tasmota devices, only for this i dont want to run a server all time to do this.

meingraham, 2019-12-10:
So, you are proposing that `WebSend` become more of a curl command? I would suggest WebSend is retained to send Tasmota commands. Then you can add a new curl command that allows sending "custom" commands via HTTP.

As for the "presence" request and, the other part of this is what methods of detecting presence would users want supported: detecting an IP address for a Wi-Fi connected mobile device? Or Bluetooth? Or ...? What share of the user community would use this weighed against the "demand" this would place on Tasmota execution.

For a small deployment 3-8 devices, you can run MQTT and NodeRed on a Pi0 ($10). Or openHAB on 1 Pi4 ($35).

ascillato2, 2019-12-10:
Added the label Feature Request to see if there is someone interested on writing a PR for a new PING command for Tasmota with a rule trigger.

ascillato2, 2019-12-10:
About websend, please, re read the command description (the first sentence). That link is already supported.

```
00:23:57 CMD: websend [10.1.1.11:80] /decoder_control.cgi?command=94&user=admin&pwd=1234
00:23:57 SRC: WebConsole from 192.168.88.18
00:23:57 CMD: Grupo 0, Índice 1, Comando "WEBSEND", Datos "[10.1.1.11:80] /decoder_control.cgi?command=94&user=admin&pwd=1234"
00:24:02 MQT: stat/balcon/RESULT = {"WebSend":"Connect failed"}
```

meingraham, 2019-12-10:
Yeah, didn't bother to re-read assuming the OP had already tested. Yes, I assumed - my bad! :wink:

Aint-Nobody, 2019-12-11:
@ascillato2
websend [10.1.1.10:80] /decoder_control.cgi?command=33&user=adminclaus&pwd=Ksjw4gbp
that is exactly what i searched for, and it works perfect.
Thank you for this advice.

Aint-Nobody, 2019-12-11:
@meingraham
Yes thats it, to detect if a mobiledevice is connected to local wifi.
May you can also detect if your smart TV is on or off.
Or many more other situations.

But i think, the consumtion will ingrease inmens if the device need to send a ping every second?

I solved this with my old gatewayrouter (AVM FRITZBOX!) this old one you can connect by telnet.
There is running a small bashscript in loop that checked by ping some ip addresses and react if need. That works fine and the router run 24/7 anyway. With this method i can also swich on/off /activate rules/....   tasmota devices by http command.

meingraham, 2019-12-11:
Bingo... Delegate presence detection to a higher level device in your automation architecture.

ascillato, 2019-12-11:
> I solved this with my old gatewayrouter

great to know that you have found a solution for your use-case. So, that means that this feature request is no longer required? In that case, please close it. Thanks.

closed 2019-12-11 by Aint-Nobody

MarkCiliaVincenti, 2020-03-21:
I'd like this feature to be honest. I plan on using MQTT to publish sensor data to the cloud, but I'd also like to ping specific IP addresses (configurable). That's something that needs to run on-premise (on the network Tasmota is on).

ascillato, 2020-03-21:
@MarkCiliaVincenti Please see https://github.com/arendst/Tasmota/issues/7176#issuecomment-564187650

MarkCiliaVincenti, 2020-03-21:
> @MarkCiliaVincenti Please see [#7176 (comment)](https://github.com/arendst/Tasmota/issues/7176#issuecomment-564187650)

I read the whole thread. The OP found a workaround. I'm still looking for this because I will only have a Sonoff on site and need to ping clients.

MarkCiliaVincenti, 2020-04-10:
> Ping was already requested (see old issues) but noone got enough interested on spending time programming a driver for that due with a home automation software, all those functions are already [...]

That works if the home automation software is on premise. The issue is that I'm building a cloud service (SaaS) and therefore need the ping to work from the IoT device.

s-hadinger, 2020-04-10:
The use case makes sense, I will look at it.

s-hadinger, 2020-04-10:
It was not very difficult. As mentioned it adds 3KB of Flash so it probably won't be included in the standard distribution. Self-compile and activate `#define USE_PING` (after Theo merges in development).

PR #8119 (2020-04-10): Add command ``Ping`` (#7176)
## Description:

Add `Ping<x> <addr>` where `<x>` is the number of packets to send (at 1 second interval) and `<addr>` the address either as domain name or IPv4 address.

The answer is asynchronous when all packets have been sent and responses handled (success or time-outs).

As ICMP ping code from Arduino takes more than 2KB of Flash, `Ping` will most certainly not be included by default. You will need to self compile and set `#define USE_PING`

Ex:
```
Ping4 192.168.1.203
20:43:59 CMD: Ping4 192.168.1.203
20:43:59 RSL: stat/tasmota_xxx/RESULT = {"Ping":"Done"}
20 [... cut]