arendst/Tasmota #7179: Fade update rate is fixed at 50ms

Taudris, 2019-12-11:
### PROBLEM DESCRIPTION
Fades update at a low rate which is visually distracting and unappealing. The `Speed` and `Sleep` settings have no impact on the update rate.

I compiled my own version of the code where I changed `xdrv_04_light.ino` to use `FUNC_LOOP` instead of `FUNC_EVERY_50_MSECOND` when calling `LightAnimate()`. `Speed` and `Sleep` together now have the effect I originally expected.

When the light is on, `Sleep` is currently forced to 10ms via `PWM_MAX_SLEEP`. It should probably only have an effect while a fade is active, and 10ms is probably too small since the `LoadAvg` hits 99 a lot, and there are frequent stutters during fades. I set `PWM_MAX_SLEEP` to 20ms and still got a decent-looking fade with a much lower `LoadAvg` around 55. Unfortunately, there is still often a stutter during fades.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): MagicHome 1 channel somethingrather
- [ ] Tasmota binary firmware version number used: 7.1.2.4
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: VSCode / PlatformIO
- [X] Flashing tools used: CP2102 for initial flash, then web upgrade tool
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
07:12:49 MQT: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
07:12:49 MQT: stat/tasmota/RESULT = {"Module":{"0":"Generic"}}
07:12:50 MQT: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"37":"PWM1"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```
07:46:47 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Office Robert"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:46:47 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:58","StartupUTC":"2019-12-11T06:43:49","Sleep":20,"CfgHolder":4617,"BootCount":33,"SaveCount":365,"SaveAddress":"FB000"}}
07:46:47 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"7.1.2.4(basic)","BuildDateTime":"2019-12-10T22:07:04","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"286/1151"}}
07:46:47 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["PatchyLAN Guest 2.4",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
07:46:47 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":460,"Free":540,"Heap":30,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8F082383","041083A0","20021706","010001C0","00000000"],"Drivers":"1,2,3,4,9,10,16,20,21","Sensors":"3"}}
07:46:47 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"office-robert.led","IPAddress":"192.168.12.45","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.12.1","Mac":"D8:F1:5B:A2:AA:E4","Webserver":2,"WifiConfig":4}}
07:46:47 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A2AAE4","MqttUser":"mosquitto","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
07:46:47 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Wed Dec 11 06:46:47 2019","Local":"Wed Dec 11 07:46:47 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:33","Sunset":"16:53"}}
07:46:47 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2019-12-11T07:46:47"}}
07:46:47 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2019-12-11T07:46:47","Uptime":"0T00:02:58","UptimeSec":178,"Vcc":3.517,"Heap":30,"SleepMode":"Dynamic","Sleep":20,"LoadAvg":49,"MqttCount":0,"POWER":"ON","Dimmer":100,"Fade":"ON","Speed":3,"LedTable":"ON","Wifi":{"AP":1,"SSId":"PatchyLAN Guest 2.4","BSSId":"E6:F4:C6:0E:A4:88","Channel":1,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Enable fading on a compatible device and toggle the output.

### EXPECTED BEHAVIOUR
Smooth fading with a high update rate (ideally >50Hz).

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2019-12-11:
The current implementation of fade runs at 20Hz (50ms), because almost everything runs at 20Hz in Tasmota. It is possible to make it faster, ideally 40Hz (25ms so it's half of 50ms) so `Sleep 25`. But it will require to adapt `LightApplyFade()` to use `millis()` instead of tick counts, otherwise timings are wrong.

I hope that 40Hz will not be visible anymore.

`LoadAvg` always goes up to 99 when using PWM, that's normal and not an issue to worry about.

I'm just surprised there is still stutter in the anymation. The main reason was because of writing settings to flash, and was handled by postponing flash writes to after the animation. Maybe the `save_data_counter` is wrong with the changes you've made.

Stay tuned

stefanbode, 2019-12-20:
For the shutter I have overcome the problem. I have anyway a RTC at 50ms. Now using this to update the PWM Speed. `TickerShutter.attach_ms(50, ShutterRtc50mS );`. In this case at least I do not miss an in-/decrease. Outside the ticker, I can adapt the acceleration amount to hit the end stops at a point. Maybe the RTC Ticker is something that could make sense. Anyhow you must be very sensitive not to load to much code into this. Best just simple and a few commands.

closed 2019-12-20 by ascillato2

Taudris, 2019-12-21:
#7264 changed `xdrv_27_shutter.ino`. This issue is about `xdrv_04_light.ino`. Please reopen.

I believe #7264 was linked because it features a solution which could be usable to solve this problem. There's a lot of changed lines there, though, so I haven't taken the time to understand what's going on there.

stefanbode, 2019-12-21:
Correct, this is just a hint that we might get the fade problem solved toonwith a rtc ticker like the one I use. I can have a look later Christmas to the light driver an make a proposal

stefanbode, 2019-12-21:
Anyhow we should reopen this case until it is fixed.

stefanbode, 2019-12-21:
The rtc ticker is quit neutral to the load. I Would start it at the fade process and end it afterwards. The benefit is you can run it on 10ms without blocking the code. This is very important. Please no output in the rtc. Also try to control the action inside with variables you change outside e.g in the every 50ms loop. I do this for the accelerate. (Very similar to fade). The acceleration is done in the rtc but the amount of acceleration is done outside. I”m pretty sure the light can work very similar.

s-hadinger, 2019-12-21:
Thanks @stefanbode, I didn't know about RTC ticker. However, each Fade change needs to execute a lot of code to change either PWM values or My92x2 values.

Let me try another approach: set Sleep to 10, and use `FUNC_LOOP` event to be called every 10 milliseconds. I will also be using `millis()` as a time reference to know what value should be used to the fade, and adjust for missing ticks.

stefanbode, 2019-12-21:
If you execute this in the loop you can also use the rtc s as a more convince way

PR #7283 (2019-12-21): Change Smoother Fade using 100Hz instead of 20Hz animation (#7179)
## Description:

Smoother `Fade` animation using 100Hz updates instead of 20Hz updates.

Internally, `Sleep` is lowered to a maximum of `10` during Fade to give a 100Hz base event loop. Fade is now recalculated at each event loop, but not updated more often than 200Hz to lower pressure on the CPU.

**Related issue (if applicable):** fixes #7179

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code cha [... cut]