arendst/Tasmota #7180: Gosund SP111 Voltage is 0

jossie67, 2019-12-11:
### PROBLEM DESCRIPTION
After flashing my Gosund SP111 V1.1 to tasmota 7.1.2 my voltage is still 0
I configed my gosund as a BlitzWolf SHP (45) and used the template {"NAME":"Gosund SP111","GPIO":[56,0,57,0,132,134,0,0,131,17,0,21,0],"FLAG":0,"BASE":18}
I have 4 new SP111 all 4 have the same issue, on I opend so I could see with version it is

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): _____
- [X ] Tasmota binary firmware version number used: _____
  - [X ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
13:19:24 CMD: Backlog
13:19:24 MQT: stat/gosund48/RESULT = {"Backlog":"Empty"}

13:19:53 CMD: Template; 
13:19:53 MQT: stat/gosund48/RESULT = {"NAME":"Gosund SP111","GPIO":[56,0,57,0,132,134,0,0,131,17,0,21,0],"FLAG":0,"BASE":18}

13:20:50 CMD: Module;
13:20:50 MQT: stat/gosund48/RESULT = {"Module":{"45":"BlitzWolf SHP"}}

13:21:19 CMD: GPIO 255
13:21:19 MQT: stat/gosund48/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"None"},"GPIO2":{"158":"LedLinki"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"134":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"17":"Button1"},"GPIO14":{"132":"HLWBL CF1"},"GPIO15":{"21":"Relay1"},"GPIO16":{"0":"None"}}
``
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
13:22:23 CMD: Status 0
13:22:23 MQT: stat/gosund48/STATUS = {"Status":{"Module":45,"FriendlyName":["gosund48"],"Topic":"gosund48","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:22:23 MQT: stat/gosund48/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:23:37","StartupUTC":"2019-12-11T11:58:46","Sleep":50,"CfgHolder":4617,"BootCount":8,"SaveCount":80,"SaveAddress":"F4000"}}
13:22:23 MQT: stat/gosund48/STATUS2 = {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019-12-06T11:22:22","Boot":7,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285"}}
13:22:23 MQT: stat/gosund48/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Home","zebejo"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
13:22:23 MQT: stat/gosund48/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
13:22:23 MQT: stat/gosund48/STATUS5 = {"StatusNET":{"Hostname":"gosund48","IPAddress":"192.168.10.248","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.30","Mac":"2C:F4:32:DB:E4:71","Webserver":2,"WifiConfig":4}}
13:22:23 MQT: stat/gosund48/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.lan","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DBE471","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
13:22:23 MQT: stat/gosund48/STATUS7 = {"StatusTIM":{"UTC":"Wed Dec 11 12:22:23 2019","Local":"Wed Dec 11 13:22:23 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:33","Sunset":"16:53"}}
13:22:23 MQT: stat/gosund48/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:22:23 MQT: stat/gosund48/STATUS10 = {"StatusSNS":{"Time":"2019-12-11T13:22:23","ENERGY":{"TotalStartTime":"2019-12-11T12:54:48","Total":0.041,"Yesterday":0.000,"Today":0.041,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
13:22:23 MQT: stat/gosund48/STATUS11 = {"StatusSTS":{"Time":"2019-12-11T13:22:23","Uptime":"0T00:23:37","UptimeSec":1417,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Home","BSSId":"C6:FB:E4:91:46:8B","Channel":10,"RSSI":40,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

0xFelix, 2019-12-11:
Have you tried all the different templates available?

https://blakadder.github.io/templates/gosund_SP111.html
https://blakadder.github.io/templates/gosund_SP111_v1_1.html
https://blakadder.github.io/templates/gosund_SP111_v2.html

ascillato, 2019-12-11:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-11 by ascillato2

jossie67, 2019-12-11:
> Have you tried all the different templates available?
> 
> https://blakadder.github.io/templates/gosund_SP111.html
> https://blakadder.github.io/templates/gosund_SP111_v1_1.html
> https://blakadder.github.io/templates/gosund_SP111_v2.html

Yes I did, I first used v1 then I found v2, when I openend one, I could see that it is V1.1

jossie67, 2019-12-11:
> Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug [...]

think it is a bug, thats why I opend an issue

meingraham, 2019-12-11:
As evidenced by there being three templates for this device, it's likely this is yet another hardware revision of the same device.

TWuerdemann, 2020-04-03:
I tried this template with my 15A Gosund SP111
https://blakadder.github.io/templates/gosund_SP111_v2.html

Now I have at least some values for voltage, current and power.
Although voltage shows 290V should be around 220V.

How to fix this?

ascillato, 2020-04-03:
You need to do the calibrations. Please, see the docs (tasmota.com). Thanks.

TWuerdemann, 2020-04-03:
Yes, I saw that. I just was on my way to delete my comment.
Thanks ;)