arendst/Tasmota #7182: Set range for analog value with AdcParam

robin13, 2019-12-11:
**Have you looked for this feature in other issues and in the wiki?**  

Yes

**Is your feature request related to a problem? Please describe.**  
Ability to have a meaningful value returned from an analogue input.

**Describe the solution you'd like**  

I am using an analogue capacitive soil moisture sensor which outputs a voltage.  Reading using the Analog input on A0 I get a value between 720 (dry) and 200 (wet).  That's small subsection of the entire range 0 -> 1023, but also it is very difficult for me to represent in my head (or in a graph) how "wet" the soil is when I e.g. get the value 250 (quite wet).  What I would like is to be able to output the equivalent 0->100 reading.
I imagine this being configured with `AdcParam` something like:
```
AdcParam 4 200 720 100 0
```
Where 4 = Remap analogue values.
`<param1>` = lower limit
`<param2>` = upper limit
`<param3>` = value for lower limit
`<param4>` = value for upper limit

In the example I have deliberately set the value for upper limit to a number lower than the value for upper limit because in my example I want 0 to be "dry" and 100 to be "wet", so while `<param1>` should always be less than `<param2>`, `<param3>` and `<param4>` should just be a range where it is not important which is greater.
So going back to the example above, the value `250` would be:
```
int( ( ( (<param2> - <analogue-value>) / ( <param2> - <param1> ) ) * ( <param3> - <param4> ) ) + <param4> )
or
int( ( ( 720 − 250 ) ÷ ( 720 − 200 ) ) × (  100 − 0 ) ) + 0
= 90
```
So the result `90` (of 100) is much more meaningful than 250 as a measure of "wetness".

**Describe alternatives you've considered**  

It is possible to do the manipulation later on in the stack - e.g. using templating in home assistant, but this still makes looking at the tasmota web interface and seeing a number there quite meaningless.

**Additional context**  
_none_

ascillato, 2019-12-11:
Please, check in the wiki at rules cookbook. There is an example on catching the value of ADC, converting and publishing as you need.

If you need help on adapting that, please address this to the Tasmota Support Chat.

Thanks.

ascillato, 2019-12-12:
> It is possible to do the manipulation later on in the stack - e.g. using templating in home assistant, but this still makes looking at the tasmota web interface and seeing a number there quite meaningless.

good point.

If you want, please do a PR to add that. Thanks.

effelle, 2019-12-13:
@robin13  `AdcParam` has not a setting for pure analog, is just for temperature and light. What you want is to scale the result using `scale` and a var.

`Syntax: SCALEx value, fromLow, fromHigh, toLow, toHigh` 

Please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for **Tasmota software feature requests and bug reporting**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

effelle, 2019-12-13:
@ascillato2 could be interesting add a conversion 0-100 to `AdcParam` and avoid the use of variables and rules to publish the edited value but we need to add one more argument to the command because
```json
Response_P(PSTR("{\"" D_CMND_ADCPARAM "\":[%d,%d,%d,%s]}"), 
Settings.adc_param_type, Settings.adc_param1, Settings.adc_param2, param3);
```
Still, since we already have `scale` command I don't know if is necessary. 
What do you think?

ascillato, 2019-12-13:
Yes, that sounds good in order to simplify rules. In the case the ADC value is used in rules and need a calibration, the SCALE can be avoided using the ADCPARAM command.

You want to do a PR ?

effelle, 2019-12-13:
Sure, but not this week, perhaps on the next. 
Free values as for `scale` command or just a simple 0-100% ?

ascillato, 2019-12-13:
The same as scale I think should be the best

closed 2019-12-24 by arendst

robin13, 2019-12-25:
Woohoo!  Thank you for the :christmas_tree: gift :santa: ! :)

robin13, 2019-12-25:
For completeness, to implement the range I intended above, the new command is:
```
AdcParam 6,720,200

effelle, 2019-12-25:
Hi @robin13 , 

the correct syntax is `AdcParam 6,200,720`.

Issue `adcparam 6` to reset to default (all the analog range).

`11:03:35 MQT: tasmota/stat/RESULT = {"AdcParam":[6,0,1023]}`

For completeness on my side since there is no documentation yet, here is how I calibrate my analogs.
I use just capacitive sensors and without calibration I get a narrow range of about 240-630 :
- Configure the moisture sensor as normal `ADC0 analog (0)`
- Write down the minimum value registered on a completely dry ambient (in my case was 243)
- Write down the maximum value registered with the sensor in a glass of water (in my case was 632)
- Finally, change the configuration to `ADC0 Moisture (6)` and issue the command `AdcParam 6, Low, High` (`AdcParam 6,243,632` in my case).

robin13, 2019-12-26:
Actually, in my situation wet == 200 and dry == 720, so to have the correct transition from 0% (dry) to 100% (wet) I think that `AdcParam 6,200,720` is correct no?
Once this is set, is it still possible to query the raw analogue value for control?  I have an air quality sensor (MQ-135) where the zero value seems to be wandering upwards over time (I guess it's a cheap sensor and/or maybe the burn-in period is longer than the documented 1 day).  So I'm wondering what the easiest way is to re-calibrate.  Do I have to set the pin back to plain "Analog" and then read the max/min values of the range, then set back to "Moisture" to be able to set `AdcParam 6`?

effelle, 2019-12-26:
That's interesting. What moisture sensor are you using? I use just capacitive sensors like this one and act in the opposite direction. 
[image]

effelle, 2019-12-26:
> Once this is set, is it still possible to query the raw analogue value for control?

The only way to have Tasmota reporting the actual analog value instead on Moisture is to revert the config to Analog(0).
> Do I have to set the pin back to plain "Analog" and then read the max/min values of the range, then set back to "Moisture" to be able to set AdcParam 6?

Yes, because every AdcParam is bound to the respective analog, so we have `AdcParam 2` for `ADC0 Temperature (2)`, `AdcParam 3` for `ADC0 Light (3)` and `AdcParam 6` for `ADC0 Moisture (6)`
if you try to use Adcparam when using Analog (0) you will get 
```07:54:53 CMD: adcparam
07:54:53 MQT: tasmota/stat/RESULT = {"Command":"Error"}```
because 0 is not on command index.

robin13, 2019-12-26:
@effelle That's interesting!  I have basically the same sensor, but the V2.0 of it
[image]

I notice you have coated the electrics of yours in what looks like epoxy - does that work for using outdoors / occasional submersion?

effelle, 2019-12-26:
Use epoxy or a  good resin (both sides) and you will have a waterproof sensor. I never tried a long time submersion, but I use them outdoors, no issue at all. 
The only difference I see is the transistor you have between C5 and C6. Mine have a direct connection using a zero-ohm link. Would be interesting check how the 555 is connected to understand why there is an inversion. 
Anyway, the most important thing is the formula works in both ways!

PR #7309 (2019-12-24): Analog Moisture Sensor
## Description:
This PR adds ADC0 Moisture sensor on analog selection. 
Since the range of this sensors have a quite limited reading range when configured as pure analog  (220-550 instead of 0-1023) I've also added a calibration using `AdcParam 6, Low, High`.
Example of calibration using a glass of water: 
```
22:34:47 MQT: stat/tasmota/RESULT = {"AdcParam":[6,0,1023]}
22:34:47 MQT: stat/tasmota/STATUS8 = {"StatusSNS":{"Time":"2019-12-23T22:34:47","ANALOG":{"Moisture":76}}}
22:34:59 CMD: backlog adcparam 6,243,632; status 8 
22:34:59 MQT: stat/tasmota/RESULT = {"AdcParam [... cut]