arendst/Tasmota #7189: MQTT: Send Offline message to LWT on clean disconnect

0xFelix, 2019-12-11:
**Have you looked for this feature in other issues and in the wiki?**  
Yes

**Is your feature request related to a problem? Please describe.**  
On clean disconnects of Tasmota the LWT topic of the devices retains its message "Online". When for example reconfiguring the device with wrong parameters the device disconnects and can't come up again but the LWT still says "Online".

**Describe the solution you'd like**  
Send "Offline" message to LWT on clean disconnect. This should probably be configurable with SetOption as for example devices with DeepSleepTime set disconnect regularly but their last sent sensor values are still valid.

**Describe alternatives you've considered**  
Ignore it as it's not such a common use case.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-11:
> When for example reconfiguring the device with wrong parameters the device disconnects and can't come up again but the LWT still says "Online".

I think I'm not following you. If you have changed the topic, Tasmota won't know how to erase the previous config. That is a job for you to clean the retained messages of your mqtt broker. 

> devices with DeepSleepTime set disconnect regularly but their last sent sensor values are still valid.

In those cases, it is not Tasmota's job to fake a online status. That is a misconfiguration in your home automation software. That sensor shouldn't have an online status topic to check. It must be a simple sensor.

0xFelix, 2019-12-12:
> I think I'm not following you. If you have changed the topic, Tasmota won't know how to erase the previous config. That is a job for you to clean the retained messages of your mqtt broker.

Think of a wrong host address, port number or changed user credentials for the MQTT broker. You save the settings and restart the device, it disconnects cleanly and cannot reconnect after. LWT still says "Online", because "Offline" is not sent on a clean disconnect, but in fact the device is offline.

> In those cases, it is not Tasmota's job to fake an online status. That is a misconfiguration in your home automation software. That sensor shouldn't have an online status topic to check. It needs to be configured as a simple sensor without LWT.

It's not its job but that's the current behavior of Tasmota. When the device enters deep sleep Tasmota disconnects cleanly and so LWT still says "Online" although technically the device is offline until the next wake-up. I just thought that might be a use-case which breaks when always sending "Offline" to LWT on clean disconnects.

All in all those are both rather unusual use-cases, but they are edge-cases which could be considered.

arendst, 2019-12-12:
As far as I know the LWT message is NOT used for telling the broker the device goes offline. The goal of the LWT is that after THE BROKER signals the device is offline it then sends the LWT message to all clients that need it.

LWT means Last Will and Testament and doesn't mean Hey I'm Going Offline.

Tasmota does a clean disconnect and the broker should act upon this. No need to send another LWT message as the initial message is already send at first connection.

For more info see this http://www.steves-internet-guide.com/mqtt-last-will-example/

0xFelix, 2019-12-12:
> As far as I know the LWT message is NOT used for telling the broker the device goes offline. The goal of the LWT is that after THE BROKER signals the device is offline it then sends the LWT message to all clients that need it.

Yes, so on unclean (in your article "abnormal") disconnects other clients can react on the unclean shutdown.

> LWT means Last Will and Testament and doesn't mean Hey I'm Going Offline.

If you want to use LWT as availability_topic in Home Assistant for example it would be more consistent to also mark devices as unavailable by sending "Offline" to LWT during user-initiated restarts ("clean disconnects").

> Tasmota does a clean disconnect and the broker should act upon this. No need to send another LWT message as the initial message is already send at first connection.

On a clean disconnect the last will message is discarded and LWT stays on "Online". Only on abnormal disconnects LWT switches to "Offline".

ascillato2, 2019-12-12:
As you, I also use Tasmota on Home Assistant, and If any of my devices restarts, is left without power, or hungs, I see them as OFFLINE. As soon as they reconnect to the MQTT broker, I see them as ONLINE.

I don't understand what else is that you want.

In the case you mention before, if you change any configuration (topic, etc), the old configuration will show as an OFFLINE device in the broker. And as Tasmota don't know the old config, because you reflash, you did a reset 5 or whatever, it is up to you to delete those retained messages (if you care about those old messages because most of the time, if you have changed the topic, the old topic is no longer useful for you and your HA config)

If you are using HA autodiscovery (setoption19), is another subject not related to LWT.

0xFelix, 2019-12-12:
> If any of my devices restarts, is left without power, or hungs, I see them as OFFLINE.

My Home Assisant also shows devices as OFFLINE when they are left without power or they hang. But a user initiated restart is not shown as OFFLINE (because no message "Offline" is sent to tele/device/LWT, neither by the device itself or the broker).

> In the case you mention before, if you change any configuration (topic, etc),

I am not speaking about changing the topic but other configuration variables you can change (user/password, host address or port).

> If you are using HA autodiscovery (setoption19), is another subject not related to LWT.

I am not using autodiscovery.

Again, these are the steps to trigger to the edge-case:
1. Change configuration to something wrong (user/password, host address or port)
2. Restart device (clean disconnect, no message "Offline" on LWT topic)
3. Device cannot reconnect, Home Assistant still thinks device is online (Last retained message on LWT topic is "Online")

Yes, it is weird enough to not get there during normal operation. But think of a mistake during configuration you did not notice, your home automation software could give you a hint now...

ascillato, 2019-12-12:
If the user have changed password or host, Tasmota can't send this message, but the device will be logged as OFFLINE as soon as it is not connected to the broker. So, this message is not feasible to be sent. 

Besides, the way Tasmota manages the LWT is the standard. Modifying it to a different behavior might broke some setups and some other home automation softwares. 

I have performed some tests in order to understand where is the issue you post because I'm not seeing any problem. If I restart Tasmota, it is not shown OFFLINE because that reconnect is very fast. If I modify that (simulating a hot restart and hung) it is being shown as OFFLINE. So, I can't make any scenario where a modification to the standard is needed.

Can you share a test where the actual LWT don't perform correctly?

ascillato, 2019-12-13:
Hi,

Can you share a test where the actual LWT don't perform correctly?

0xFelix, 2019-12-14:
> If the user have changed password or host, Tasmota can't send this message, but the device will be logged as OFFLINE as soon as it is not connected to the broker. So, this message is not feasible to be sent.

This is my concern. It is not shown OFFLINE when the user changes password or host. The broker discards the last will message because the device disconnects gracefully and the device's LWT topic stays on "Online".
 
> If I restart Tasmota, it is not shown OFFLINE because that reconnect is very fast. 

It is not shown OFFLINE because it it's too fast but because no "Offline" message is sent to the LWT topic of the device. (At least in my case) If you follow my steps in https://github.com/arendst/Tasmota/issues/7189#issuecomment-565186381 the behavior should be reproducible. When I follow these steps and look at the logs of my broker (mosquitto) a graceful disconnect is logged when the device restarts and no "Offline" message appears on the LWT topic.

0xFelix, 2019-12-14:
See also [here](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/errata01/os/mqtt-v3.1.1-errata01-os-complete.html#_Toc442180843) in the MQTT Standard for version 3.1.1 in section 3.1.2.5: Last will is deleted by the receipt of a DISCONNECT packet.

ascillato, 2019-12-15:
I did several tests in order to try to reproduce this.

If I have a device connected to MQTT and I change:

* BROKER IP
* BROKER USERNAME
* BROKER PASSWORD
* TOPIC
* MQTT CLIENT NAME
* SSID 1 OR 2, OR PASSWORDS

After Tasmota **restarts** (and is not able to reconnect to the MQTT broker) the device is shown as ONLINE (no matter how much time you wait) while in reality, the device is OFFLINE.

So, a solution for this could be, if tasmota is going to RESTART, AND the MQTT connection is ALIVE, to send an OFFLINE MQTT message (`tele/living/LWT Offline`), before the restart and then perform the restart.

After some tests with this new approach, on clean disconnects, the device status is correctly informed in the broker and in the home automation software.

I'm doing a PR with this.

@0xFelix Thanks for reporting this and for the patience keeping with the explanations.

ascillato, 2019-12-15:
There are 2 approaches for solving this.

* One is to send a MQTT message `tele/living/LWT Offline` Before restarting.
* The other is to close the connection without the DISCONNECT package.

Both work fine, but the second is smaller in code (in fact it reduces some bytes the actual codebase)

I have tested both and seems not to be any inconvenience with any of those. The first approach is faster to be reflected in the broker (obviously), but uses a little more of code and the restart takes a little longer to be perfomed.

Do you have any thoughts about these approaches? I'm inclined to use the second one.

ascillato, 2019-12-15:
The PR https://github.com/arendst/Tasmota/pull/7231 has been proposed for review.

ascillato, 2019-12-15:
> This should probably be configurable with SetOption as for example devices with DeepSleepTime set disconnect regularly but their last sent sensor values are still valid.

For deepsleep devices, LWT should not be used in the the Home Automation software configuration due to LWT is aimed for continuos connected devices and leaving an ONLINE in the broker for deepsleep devices, won't be a real status.

closed 2019-12-15 by ascillato2

0xFelix, 2019-12-16:
> For deepsleep devices, LWT should not be used in the the Home Automation software configuration due to LWT is aimed for continuos connected devices and leaving an ONLINE in the broker for deepsleep devices, won't be a real status.

I suppose so. It"s just a change in behavior users of DeepSleepTime should be aware of.

PR #7231 (2019-12-28): PubSubClient: Allow clean disconnection as optional.
## Description:

**PubSubClient Library**: Allow clean disconnection as optional.

This avoid the automatic deletion of the LWT message in the MQTT broker for clean disconnections in order to let LWT to work on all disconnections cases (Wifi disconnection, wifi bad comms, Normal Restart, Config Changes, unintended or intentional crash, OTA updates, ArduinoOTA, etc) and showing the Tasmota device as OFF-LINE in the home automation software until it reconnects and the LWT with ONLINE is sent again.

**Related issue (if applicable):** fixes #71 [... cut]