arendst/Tasmota #7198: Issue Tassmota 7.1.2.4 on Wemos D1 mini

cliffscott, 2019-12-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Module is going to reset by itself after creating buttons

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Hello all,

I got an issue with my Wemos D1 mini.
I want to drive a shutter without setting SetOption80.

I created a generic 18 module and set two relais. One is on GPIO 4 (D2) the other one on GPIO5 (D1).
I also set InterLock 1.
Everything is working well.

Here my output at the console:

00:00:00 CFG: aus Flash geladen am F5, zählen 7
00:00:00 Projekt tasmota Tasmota Version 7.1.2.4(tasmota)-2_6_1
00:00:00 WIF: verbinden mit AP1 claushome in Modus 11N als tasmota-3783...
00:00:06 WIF: verbunden
00:00:06 HTP: Web-Server aktiv bei tasmota-3783 mit IP-Adresse 192.168.1.203
00:00:07 MQT: Verbindungsversuch...
00:00:07 MQT: verbunden
00:00:07 MQT: tele/tasmota/LWT = Online (beibehalten)
00:00:07 MQT: cmnd/tasmota/POWER =
00:00:07 MQT: tele/tasmota/INFO1 = {"Module":"Generic","Version":"7.1.2.4(tasmota)","FallbackTopic":"cmnd/DVES_ED0EC7_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:07 MQT: tele/tasmota/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-3783","IPAddress":"192.168.1.203"}
00:00:07 MQT: tele/tasmota/INFO3 = {"RestartReason":"Software/System restart"}
00:00:07 MQT: stat/tasmota/RESULT = {"POWER1":"OFF"}
00:00:07 MQT: stat/tasmota/POWER1 = OFF
00:00:07 MQT: stat/tasmota/RESULT = {"POWER2":"OFF"}
00:00:07 MQT: stat/tasmota/POWER2 = OFF
00:00:11 MQT: tele/tasmota/STATE = {"Time":"1970-01-01T00:00:11","Uptime":"0T00:00:13","UptimeSec":13,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"claushome","BSSId":"90:5C:44:F6:B0:A2","Channel":10,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:07"}}

German, sorry ;-)

Now I create two buttons on D1 and D2. But don't care about the pins. I tested all of the free GPIO's.
The buttons / GPIO's are connected with a pulldown 3.3K to GND.
After this the Wemos restarts after app. 40 seconds with a reset and goes back to the Sonoff basic settings. As you can see the flash is empty :-(. MQTT name is gone, all settings are gone.

Here the output from console:

00:00:00 CFG: aus Flash geladen am FA, zählen 2
00:00:00 Projekt tasmota Tasmota Version 7.1.2.4(tasmota)-2_6_1
00:00:00 WIF: verbinden mit AP1 claushome in Modus 11N als tasmota-3783...
00:00:05 WIF: verbunden
00:00:05 HTP: Web-Server aktiv bei tasmota-3783 mit IP-Adresse 192.168.1.203
16:10:30 MQT: Verbindungsversuch...
16:10:30 MQT: verbunden
16:10:30 MQT: tele/tasmota/LWT = Online (beibehalten)
16:10:30 MQT: cmnd/tasmota/POWER =
16:10:30 MQT: tele/tasmota/INFO1 = {"Module":"Sonoff Basic", "Version":"7.1.2.4(tasmota)","FallbackTopic":"cmnd/DVES_ED0EC7_fb/","GroupTopic":"cmnd/tasmotas/"}
16:10:30 MQT: tele/tasmota/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-3783","IPAddress":"192.168.1.203"}
16:10:30 MQT: tele/tasmota/INFO3 = {"RestartReason":"Software/System restart"}
16:10:30 MQT: stat/tasmota/RESULT = {"POWER":"OFF"}
16:10:30 MQT: stat/tasmota/POWER = OFF
16:10:34 MQT: tele/tasmota/STATE = {"Time":"2019-12-12T16:10:34","Uptime":"0T00:00:12","UptimeSec":12,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"claushome","BSSId":"90:5C:44:F6:B0:A2","Channel":10,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:06"}}

If tried with several settings for BOOT_LOOP_OFFSET like 0, 1 ,100 but ervrytime the same issue.
CFG_HOLDER is set to 0x20161211.

Any idea?
Thanks for your help.

Claus

**(Please, remember to close the issue when the problem has been addressed)**

localhost61, 2019-12-12:
Can you try to reverse your buttons and connect them to ground with a pullup when inactive.
and set `backlog switchmode1 4;switchmode2 4` ([wiki](https://tasmota.github.io/docs/#/Commands?id=control)

cliffscott, 2019-12-12:
Hi localhost61,

it looks like it's working with this configuration.
Question is: Why does the module reset?

localhost61, 2019-12-12:
Can't tell, I never use pulldowns for buttons/switches ;-).
Usually in electronic schemes when pins are not used, they are tight to VCC, not to ground.
You didn't say which pins you tried to connect your buttons to.
- For instance on a Wemos I connected 4 AM2301 to GPIO 2, 4 12 & 13... On GPIO15 it couldn't boot. 
- An other issue is the 3.3V power available, if like me you seek for the cheapest devices, you get clones with LDO regulators labeled 4B2X. They can't supply more than 150mA, it's even not enough to power the device with it's highest WiFi output level .

cliffscott, 2019-12-12:
I did an arduino sketch for my shutters as a test and I used the pulldowns with the wemos D1. It is working.
So I thought I can do with the tasmota, too.
But looks like it's not working with the pulldown resistors.

Thanks for all the Info.

My main issue is solved, thanks.
Will close the thread.

closed 2019-12-12 by cliffscott