arendst/Tasmota #7200: Sonoff Basic constant reboot on 7.1.2

imdos, 2019-12-12:
### PROBLEM DESCRIPTION
_The device a Sonoff basic reboots within a few minutes after power on. Furthermore I have updated  a second basic and experience similar behaviour altough a less frequent reboot._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic
- [x] Tasmota binary firmware version number used: _7.1.2_
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
00:09:07 CMD: Backlog Template; Module; GPIO 255
00:09:07 MQT: stat/Wii/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
00:09:08 MQT: stat/Wii/RESULT = {"Module":{"1":"Sonoff Basic"}}
00:09:08 MQT: stat/Wii/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
stat/Wii/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
00:00:00 CFG: Loaded from flash at F8, Count 753
00:00:00 Project tasmota Sonoff-wii Version 7.1.2(tasmota)-2_6_1
00:00:00 WIF: Connecting to AP1 IoT_limited in mode 11N as Wii-1267...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on Wii-1267 with IP address 192.168.2.25
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/Wii/LWT = Online (retained)
00:00:05 MQT: cmnd/Wii/POWER = 
00:00:05 MQT: tele/Wii/INFO1 = {"Module":"Sonoff Basic","Version":"7.1.2(tasmota)","FallbackTopic":"cmnd/Sonoff_wii_2EC4F3_fb/","GroupTopic":"cmnd/nintendo/"}
00:00:05 MQT: tele/Wii/INFO2 = {"WebServerMode":"Admin","Hostname":"Wii-1267","IPAddress":"192.168.2.25"}
00:00:05 MQT: tele/Wii/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x48402420 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
00:00:05 MQT: stat/Wii/RESULT = {"POWER":"ON"}
00:00:05 MQT: stat/Wii/POWER = ON
00:00:06 MQT: domoticz/in = {"idx":62,"nvalue":1,"svalue":"","Battery":100,"RSSI":9}
00:00:09 MQT: tele/Wii/STATE = {"Time":"1970-01-01T00:00:09","Uptime":"0T00:00:11","UptimeSec":11,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":33,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"IoT_limited","BSSId":"54:A0:50:E4:87:DA","Channel":7,"RSSI":90,"LinkCount":1,"Downtime":"0T00:00:05"}}
00:00:25 MQT: stat/Wii/STATUS5 = {"StatusNET":{"Hostname":"Wii-1267","IPAddress":"192.168.2.25","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"2C:3A:E8:2E:C4:F3","Webserver":2,"WifiConfig":5}}
00:00:25 MQT: stat/Wii/STATUS6 = {"StatusMQT":{"MqttHost":"pi","MqttPort":1883,"MqttClientMask":"Sonoff_wii_%06X","MqttClient":"Sonoff_wii_2EC4F3","MqttUser":"MQTT","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:00:25 MQT: stat/Wii/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:25 1970","Local":"Thu Jan 01 00:00:25 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:47","Sunset":"15:37"}}
00:00:25 MQT: stat/Wii/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:25"}}
00:00:26 MQT: stat/Wii/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:25","Uptime":"0T00:00:24","UptimeSec":24,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":170,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"IoT_limited","BSSId":"54:A0:50:E4:87:DA","Channel":7,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:05"}}}
00:00:33 MQT: stat/Wii/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff-wii"],"Topic":"Wii","ButtonTopic":"0","Power":1,"PowerOnState":2,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:00:33 MQT: stat/Wii/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"nintendo","OtaUrl":"https://github.com/arendst/Tasmota/releases/download/v6.7.1/sonoff-minimal.bin","RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x48402420 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000","Uptime":"0T00:00:28","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":591,"SaveCount":754,"SaveAddress":"F7000"}}
00:00:33 MQT: stat/Wii/STATUS2 = {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019-12-06T11:22:22","Boot":6,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
00:00:33 MQT: stat/Wii/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":1,"LogHost":"syslog","LogPort":514,"SSId":["IoT_limited",""],"TelePeriod":29,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","00000000","00000000"]}}
00:00:33 MQT: stat/Wii/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
00:00:33 MQT: stat/Wii/STATUS5 = {"StatusNET":{"Hostname":"Wii-1267","IPAddress":"192.168.2.25","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"2C:3A:E8:2E:C4:F3","Webserver":2,"WifiConfig":5}}
00:00:33 MQT: stat/Wii/STATUS6 = {"StatusMQT":{"MqttHost":"pi","MqttPort":1883,"MqttClientMask":"Sonoff_wii_%06X","MqttClient":"Sonoff_wii_2EC4F3","MqttUser":"MQTT","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:00:33 MQT: stat/Wii/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:33 1970","Local":"Thu Jan 01 00:00:33 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:47","Sunset":"15:37"}}
00:00:33 MQT: stat/Wii/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:33"}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
[... 26 lines cut ...]
2019-12-12 15:51:54.149  Status: dzVents: Info: Handling events for: "Wii", value: "On"
2019-12-12 15:55:58.006  Status: dzVents: Info: Handling events for: "Wii", value: "On"
2019-12-12 15:58:09.685  Status: dzVents: Info: Handling events for: "Wii", value: "On"
2019-12-12 16:00:56.045  Status: dzVents: Info: Handling events for: "Wii", value: "On"
2019-12-12 16:02:28.474  Status: dzVents: Info: Handling events for: "Wii", value: "On"
2019-12-12 16:04:32.324  Status: dzVents: Info: Handling events for: "Wii", value: "On"
2019-12-12 16:05:35.057  Status: dzVents: Info: Handling events for: "Wii", value: "On"
2019-12-12 16:09:18.195  Status: dzVents: Info: Handling events for: "Wii", value: "On"
2019-12-12 16:10:16.335  Status: dzVents: Info: Handling events for: "Wii", value: "On"
2019-12-12 16:13:05.980  Status: dzVents: Info: Handling events for: "Wii", value: "On"
2019-12-12 16:14:25.470  Status: dzVents: Info: Handling events for: "Wii", value: "On"
2019-12-12 16:15:55.707  Status: dzVents: Info: Handling events for: "Wii", value: "On"
2019-12-12 16:18:09.283  Status: dzVents: Info: Handling events for: "Wii", value: "On"
_

### EXPECTED BEHAVIOUR
_The device should stay online. The load is almost nothing. Instead it reboots within 4 minutes. _

### SCREENSHOTS
_Not applicable._

### ADDITIONAL CONTEXT
_Haven't experienced any reboots before while on other firmware versions. Like 6.4.0 or 6.6.0 and all on the same device. Had to reflash it to 6.7.1 to let the kids play on the Wii!._

**(Please, remember to close the issue when the problem has been addressed)**

localhost61, 2019-12-12:
What lot of people would tell you is backup your config if needed and then issue a Reset 5 command in the console of the WebUI. It will retain MQTT and WiFi settings but erase all other settings. Then restore your settings. If it's not better, the do it again but reenter your configuration manually, one command after the other or with backlog to group them.

localhost61, 2019-12-12:
You get an "Fatal exception", I guess someone can help you to decode it.

ascillato, 2019-12-12:
Exception (0) indicates an illegal instruction.

So, that means bad flashing. Please, do a reset 5 as requested before. If that don't solve your issue, you will need to erase all flash and flash again by serial using a good FTDI and providing enough power on 3.3v.

If you need further assistant on this, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. 

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-12 by ascillato2

imdos, 2019-12-13:
Great.

Reset 5 did the trick as it seems up until now. 
For clarification: Reset 6 is for retaining MQTT settings as well.