arendst/Tasmota #7205: DS18B20 only does not work with tasmota.bin or tasmota-DE.bin

apostrophe55, 2019-12-13:
Hello,

I'm using a ESP-12F module and connected a DS18B20 sensor to D2. Temperature is only shown when firmware tasmota-sensors.bin is loaded. No success with tasmota.bin or tasmota-DE.bin (all 7.1.2).

What am I doing wrong?

closed 2019-12-13 by close-issue-app[bot]

ascillato, 2019-12-13:
You are not doing anything wrong. That sensor is only supported in the Tasmota-sensors.bin

In the rest of the files, this sensors is not included. See release notes. There is a table explaining what is included on each firmware type.

If you want a DE versions of tasmota-sensors.bin you will need to compile by yourself.

if you need further assistance, please address this to the Tasmota Support Chat. (link in the message above)

Aint-Nobody, 2019-12-13:
@ascillato 
Hm...? I use a DS18B20 at RX Port of a Sonoff S20 with Tasmota 7.0.0.6 -DE and it works.

Aint-Nobody, 2019-12-13:
I use the precompiled version.

apostrophe55, 2019-12-13:
I've already looked at the [table](https://github.com/arendst/Tasmota/blob/v7.1.2/BUILDS.md). USE_DS18x20 is set for tasmota, knx, sensors and display bins. So it should also work with the DE version.

Aint-Nobody, 2019-12-13:
But it works only with a pullup resistor ! 4k7 i/o to vcc.