arendst/Tasmota #7206: IBeacon

lacrimosa2k2, 2019-12-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
i have put a HM-10 bluetooth modul on an wemos d1 mini, anconfigure it on the hardware tx and rx pins, i can see the communikation when i open the serial console, the tasmota formware sends at and at+disc and the hm-10 answers with the scan results, but nothing i cant find in the web console or in the mqtt data.

the serial data like this
OK+DISISOK+DISC:00000000:00000000000000000000000000000000:0000000000:FFFF365C3396:-063OK+DISC:00000000:00000000000000000000000000000000:0000000000:FEF4FD96F07C:-084OK+DISCEOK+DISISOK+DISC:00000000:00000000000000000000000000000000:0000000000:FEF4FD96F07C:-075OK+DISCEOK+DISISOK+DISC:00000000:00000000000000000000000000000000:0000000000:FEF4FD96F07C:-082OK+DISC:00000000:00000000000000000000000000000000:0000000000:FFFF365C3396:-062
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): WEMOS D1 Mini & hm-10
- [ ] Tasmota binary firmware version number used: 2_6_1/2.2.2-dev(5ab15d1)
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [x ] IDE / Compiler used: Arduino IDE
- [x ] Flashing tools used: Arduino IDE
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
{"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
17:42:27 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:42:27 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:53","StartupUTC":"2019-12-13T16:35:34","Sleep":50,"CfgHolder":4617,"BootCount":5,"SaveCount":14,"SaveAddress":"3F5000"}}
17:42:27 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"7.1.2.5(tasmota)","BuildDateTime":"2019.12.13 17:29:52","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(5ab15d1)","Hardware":"ESP8266EX","CR":"258/1151"}}
17:42:27 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["0011001000110011",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
17:42:27 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":597,"Free":1448,"Heap":21,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":2,"Features":["00000407","8FDAE397","003683A0","22B617CD","01001BC0","00007885"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34,52"}}
17:42:27 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2261","IPAddress":"192.168.178.23","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"BC:DD:C2:B6:68:D5","Webserver":2,"WifiConfig":4}}
17:42:27 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"iobroker.local","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B668D5","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
17:42:27 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 13 16:42:27 2019","Local":"Fri Dec 13 17:42:27 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:35","Sunset":"16:53"}}
17:42:27 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-12-13T17:42:27"}}
17:42:27 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-12-13T17:42:27","Uptime":"0T00:06:53","UptimeSec":413,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxxxxxxx","BSSId":"00:17:13:15:65:98","Channel":1,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Put a HM-10 module on tx and rx, the tasmota firmware sends over serial the at and disc commants and the hm-10 answers correct. but nothing in web console or mqtt output

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
trying enable debug in firmware, but nothing see what happens with the hm-10 module

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-12-13:
HM10 module is NOT support. See wiki which module can be used and how to configure
https://tasmota.github.io/docs/#/peripherals/iBeacon-driver

lacrimosa2k2, 2019-12-13:
> 
> 
> HM10 module is NOT supported. See wiki which module can be used and how to configure
> https://tasmota.github.io/docs/#/peripherals/iBeacon-driver

yes but "it is driven by AT cmds so there may be other compatible modules too." and when i watch the serial communication between the hm-10 and tasmota it send and recive the correct data

Jason2866, 2019-12-13:
The AT command set from HM10 is different. The driver does not support this module

lacrimosa2k2, 2019-12-13:
> 
> 
> The AT command set from HM10 is different. The driver does not support this module

damn, thanks.

lacrimosa2k2, 2019-12-20:
Now i have a HM-17 module here.
the tasmota firmware sends the AT command, my key beacon beeps by every at command, but the hm-17 module not answer in any way. what i doing wrong?

ascillato, 2019-12-20:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-20 by ascillato2

lacrimosa2k2, 2019-12-26:
i think it has a little issue, i have the beacon was can beep, in the webstie stand is was supportet, but in der console with debug info it tells > 00000000:0000 and in mqtt show no data