arendst/Tasmota #7213: Tasmota does not enter DeepSleep mode in case there is no wifi. Forcing it with a rule does not work either

peret2000, 2019-12-14:
### PROBLEM DESCRIPTION
I am trying to deploy a standalone ESP82166 (powered by a 9v battery), as a temperature sensor. So, the concern is to save as much energy as possible. As the temperature can be observed every several minutes, I want to use deepsleep feature. So, it should be in deepsleep state most of the time. After a period of time (in the test, 30 seconds), it would wakeup, send mqtt data and again, deepsleep state.
It works very well if there is wifi, mqtt, etc. But, in case  there is no wifi, for example, deepsleep is not activated: code enters in an infinite loop waiting for wifi. I have tried with different WifiConfig values, with no success. My concern is: in case there is no wifi, ti should anyway enter into deepsleep state, to avoid power leakage if there is a problem with the router, for example.
I have solved it with a rule, to force deepsleep state in 30 seconds, whatever the situation is.
So, if every thing is ok (wifi, mqtt,ntp), it will enter deepsleep state; and if there are problems, it will also enter deep sleep state after 30 seconds:

```rule1 on Dimmer#Boot do ruletimer1 30 endon on rules#timer=1 do deepsleeptime 30 endon```

The problem is, after 30 seconds:

- If SetOption60 = 0 (default), ESP8266 restarts with a crash
- If SetOption60 = 1, it enters correctly in Sleep state, but it does not wake up! it needs a manual RST, or power off and on

So, it seems there is a problem executing deepsleeptime through a rule

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU v3
- [x] Tasmota binary firmware version number used: 7.1.2.5 (21fa1fa0)
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PlatformIO (On Ubuntu 19.10)
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:

```
11:57:14 MQT: tasmota/stat/tempterrace/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:57:14 MQT: tasmota/stat/tempterrace/RESULT = {"Module":{"18":"Generic"}}
11:57:14 MQT: tasmota/stat/tempterrace/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"4":"DS18x20"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```

- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  12:03:38 MQT: tasmota/stat/tempterrace/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":431,"Rules":"on Dimmer#Boot do ruletimer1 30 endon on rules#timer=1 do deepsleeptime 30 endon"}
12:03:38 MQT: tasmota/stat/tempterrace/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
12:03:38 MQT: tasmota/stat/tempterrace/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":509,"Rules":"[F"}
```
- [x] Provide the output of this command: `Status 0`:
```
12:04:52 MQT: tasmota/stat/tempterrace/STATUS = {"Status":{"Module":18,"FriendlyName":["tempterrace"],"Topic":"tempterrace","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:04:52 MQT: tasmota/stat/tempterrace/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://192.168.1.120:8000/tasmota.bin","RestartReason":"Deep-Sleep Wake","Uptime":"0T00:00:51","StartupUTC":"2019-12-14T11:04:01","Sleep":50,"CfgHolder":4617,"BootCount":62,"SaveCount":344,"SaveAddress":"F4000"}}
12:04:52 MQT: tasmota/stat/tempterrace/STATUS2 = {"StatusFWR":{"Version":"7.1.2.5(21fa1fa0)","BuildDateTime":"2019-12-14T11:14:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"316/1151"}}
12:04:52 MQT: tasmota/stat/tempterrace/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"192.168.1.120","LogPort":514,"SSId":["otherwifi","Vodafone_160919"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
12:04:52 MQT: tasmota/stat/tempterrace/STATUS4 = {"StatusMEM":{"ProgramSize":563,"Free":440,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
12:04:52 MQT: tasmota/stat/tempterrace/STATUS5 = {"StatusNET":{"Hostname":"tempterrace-2048","IPAddress":"192.168.1.107","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.105","Mac":"84:0D:8E:8C:88:00","Webserver":2,"WifiConfig":4}}
12:04:52 MQT: tasmota/stat/tempterrace/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.105","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8C8800","MqttUser":"write_burgos","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
12:04:53 MQT: tasmota/stat/tempterrace/STATUS7 = {"StatusTIM":{"UTC":"Sat Dec 14 11:04:52 2019","Local":"Sat Dec 14 12:04:52 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:36","Sunset":"16:53"}}
12:04:53 MQT: tasmota/stat/tempterrace/STATUS10 = {"StatusSNS":{"Time":"2019-12-14T12:04:53","DS18B20":{"Id":"051781417DFF","Temperature":21.0},"TempUnit":"C"}}
12:04:53 MQT: tasmota/stat/tempterrace/STATUS11 = {"StatusSTS":{"Time":"2019-12-14T12:04:53","Uptime":"0T00:00:52","UptimeSec":52,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"Vodafone_160919","BSSId":"7A:B5:7E:F2:AB:5D","Channel":9,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  With SetOption60 = 0:

	00:00:00 CFG: Loaded from flash at F5, Count 391
	00:00:00 APP: Some settings have been reset (2)
	00:00:00 Project tasmota mytasmota Version 7.1.2.5-2_6_1
	00:00:00 RUL: DIMMER#BOOT performs "ruletimer1 30"
	00:00:00 RSL: tasmota/stat/tempterrace/RESULT = {"T1":30,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
	00:00:00 WIF: Connecting to AP2 badwifi in mode 11N as tempterrace-2048...
	00:00:09 WIF: Connect failed as AP cannot be reached
	00:00:09 WIF: Connecting to AP1 otherwifi in mode 11N as tempterrace-2048...
[... 30 lines cut ...]

	load 0x4010f000, len 1384, room 16 
	tail 8
	chksum 0x2d
	csum 0x2d
	v482516e3
	~ld

	00:00:00 CFG: Loaded from flash at FA, Count 394
	00:00:00 Project tasmota tempterrace Version 7.1.2.5(21fa1fa0)-2_6_1
	00:00:00 RUL: DIMMER#BOOT performs "ruletimer1 30"
	00:00:00 RSL: tasmota/stat/tempterrace/RESULT = {"T1":30,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
	00:00:00 WIF: Connecting to AP1 otherwifi in mode 11N as tempterrace-2048...

```

### TO REPRODUCE
Send Command to create the rule, specify wrong AP's, and set deepsleeptime:
```Backlog setoption60 0 ; ssid1 otherwifi; password2 password ; ssid2 badwifi; password2 password ; rule1 1 ; deepsleeptime 30 ;rule1 on Dimmer#Boot do ruletimer1 30 endon on rules#timer=1 do deepsleeptime 30 endon```

### EXPECTED BEHAVIOUR
It should enter deepsleep mode for 30'', in case there is no wifi (if there were wifi, it would also enter deepsleep mode, obviously)

Perhaps there is a way to debug, I have not found information about it. I could then give more information, even detect the problem (if any). I do not know tasmota code, so it is not affordable for me to review it
Thanks a lot!

Jason2866, 2019-12-14:
Wifi is needed for correct working of Deep sleep function. See https://tasmota.github.io/docs/#/DeepSleep?id=deep-sleep-algorithm-general-timing

peret2000, 2019-12-14:
Thanks a lot @Jason2866 for the quick and clear answer. The idea is to be able tu power with a battery, but there is the risk of wasting a lot of energy in case there is no wifi. 
If it does not bother anybody, I will leave this issue open, maybe somebody has opinions or ideas that can help. In a few days I will close it.
Regards

andrethomas, 2019-12-14:
I believe this is an issue which needs to be addressed.

The question is how...

meingraham, 2019-12-14:
Why does Deep Sleep care about the absolute time? I see the need for it if specifying a time of day. But then this sets itself up to wake periodically to reach the user configured deep sleep period. Can the "time of day" portion be an option when setting up Deep Sleep? If one specifies a specific time of day for waking, then the device obviously has to have its clock set somehow (NTP, or via an RTC module, i.e., no Wi-Fi)? But, even then, once the time is set, won't Deep Sleep operate on the user's "delta" sleep period and not the time of day? Perhaps in this mode (time unavailable), the time drift adjustment is skipped... but the deep sleep continues to function? This might even include a SetOption (e.g., ignore drift adjustment/enforce drift adjustment if TOD unavailable)

Mike

stefanbode, 2019-12-18:
Let me check what went wrong. I assume the newly initiated deepsleep_flag (uint8_t) goes to -1 in this case. I agree that in this situation it should not crash and just get into deepsleep for another cycle. The rule is not bad, because this makes it flexible. I will add this to the wiki as the issue is solved.

stefanbode, 2019-12-20:
The issue is caused by a DIV 0 operation initiated in the rule.
`rule1 on Dimmer#Boot do ruletimer1 30 endon on rules#timer=1 do deepsleeptime 30 endon`
The current implementation does not work if the rule is initiated exactly at the time that matched the deepsleep time. In this example 30=30. This causes a DIV 0. It should work with a 60sec deepsleep which will result in 30sek sleep because already 30sek awake. I will push a small fix preventing the DIV0 exception.

stefanbode, 2019-12-20:
Wiki updated. Recommend:
`rule1 on Dimmer#Boot do ruletimer1 30 endon on rules#timer=1 do deepsleeptime 3600 endon`

peret2000, 2019-12-20:
Thanks a lot, @stefanbode, it was an unfortunate coincidence with the periods (deep sleep and timer). Although it has helped discover a bug. Should I choose the issue? Or maybe you can.
For further issues, I will try to learn how to debug tasmota, if possible, so that I can contribute to be more precise with the problems, if any

closed 2019-12-20 by arendst

stefanbode, 2019-12-20:
All Fine. You are always welcome. Every finding helps improving the software

commit 46696f8b (2019-12-20): Merge pull request #7265 from stefanbode/patch-6

fix #7213
commit f49b498c (2019-12-20): fix #7213

DIV0 fix