arendst/Tasmota #7215: Shelly 1PM Power monitoring

djashjones, 2019-12-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

The power on the UI is reporting correctly but MQTT but the tele/sensor is not firing.

Using a 60w lightbulb.

POWER ON
0W-> 60w, within a second, no mqtt message fired.

POWER OFF
60W -> 27.5W, within a second, mqtt message fired.
275W -> 0W, 3 seconds later, mqtt message not fired.

teleperiod 300 (works as expected)
PowerDelta 1 (also did 15 with no effect)
setoption21 1
seriallog 0

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____Shelly 1PM
- [ ] Tasmota binary firmware version number used: _____7.1.2
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: _____tasmotizer-1.1
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
14:44:21 MQT: tasmota/stat/RESULT = {"NAME":"Shelly 1PM","GPIO":[56,0,0,0,82,134,0,0,0,0,0,21,0],"FLAG":2,"BASE":18}
14:44:22 MQT: tasmota/stat/RESULT = {"Module":{"0":"Shelly 1PM"}}
14:44:22 MQT: tasmota/stat/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"82":"Switch1n"},"GPIO5":{"134":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"21":"Relay1"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

14:44:39 MQT: tasmota/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

power on and wait for the power in W to be updated and then power off and see the results.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

tele/sensor to be fired when output power changes per setting in powerdelta

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-12-14:
Have you tried absolute values for `PowerDelta` too. Percent change is not always working great. See https://tasmota.github.io/docs/#/Commands?id=power-monitoring

djashjones, 2019-12-14:
Yes, even tried powerdelta 101

machadof1, 2019-12-15:
This is the same issue I am having and reported [#7050](https://github.com/arendst/Tasmota/issues/7050), in my case it is not a shelly

djashjones, 2019-12-15:
Looks very similar.

I can report the same with my SP23's and my Sonoff Pow R2 (v2)

I'm in the process of downgrading my SP23's to 6.6.0 (powerdelta 5) as they seem to work better but still not right and they are in use.

I have not installed my shellys or sonoff yet until a suitable solution for me can be found

ascillato2, 2019-12-17:
See https://github.com/arendst/Tasmota/issues/7050#issuecomment-566739684

djashjones, 2019-12-18:
That makes sense but its disappointing to hear

djashjones, 2019-12-18:
If thats the case then the values should read OC (open circuit) and not zero volts?

meingraham, 2019-12-18:
@djashjones 

See SetOption21

djashjones, 2019-12-18:
See above mate, It's already set to 1.

Anything after 6.6.0 does not work as expected.

meingraham, 2019-12-18:
So many users. So many posts. So little time :wink:

TelePeriod will report telemetry automatically regardless. Not in the picture for this dialog.

PowerDelta was changed in 6.6.0.20 to support both percent change and absolute value change. This probably explains the change from 6.6.0 to how telemetry behaves now.

```
POWER OFF
60W -> 27.5W, within a second, mqtt message fired.
27.5W -> 0W, 3 seconds later, mqtt message not fired.
```

@ascillato 
PowerDelta not firing when the relay is `OFF` would explain the second message not being published but it does not explain why the first message does get published.

djashjones, 2019-12-18:
Thanks for your time mate. I have a good understand of powerdelta thanks to you. Also I have various options to get the results I need. 

The way that Shelly do it I reckon through their MQTT is by sending 0W telemetry as soon as its switched of because it fires far too quickly.

meingraham, 2019-12-18:
The question remaining for me is why Tasmota reports the first change at all (if indeed `Power OFF` should not)... or whether the state of the relay should not be a factor in reporting changes. `SetOption21` and `SetOption39` should determine how `PowerDelta` behaves.

> The way that Shelly do it I reckon through their MQTT is by sending 0W telemetry as soon as its switched

You are referring to the original Shelly; yes? How they report the power monitoring metrics is probably quite different than how Tasmota does... and likely is not as configurable in terms of recurring reporting of values.

djashjones, 2019-12-18:
Wild crazy guess, there's still current across the shunt, if it uses a shunt to measure current. Not sure how the measuring IC works to be honest.

djashjones, 2019-12-20:
Just playing around with Betty and set PowerDelta to 10.

13:16:28 MQT: sound_system/tele/SENSOR = {"Time":"2019-12-20T13:16:28","ENERGY":{"TotalStartTime":"2019-12-15T12:25:12","Total":0.005,"Yesterday":0.000,"Today":0.002,"Power":9.00,"ApparentPower":17.53,"ReactivePower":15.00,"Factor":0.51,"Voltage":234,"Current":0.075}}
13:16:31 MQT: sound_system/tele/SENSOR = {"Time":"2019-12-20T13:16:31","ENERGY":{"TotalStartTime":"2019-12-15T12:25:12","Total":0.005,"Yesterday":0.000,"Today":0.002,"Power":8.90,"ApparentPower":17.79,"ReactivePower":15.40,"Factor":0.50,"Voltage":234,"Current":0.076}}
13:16:34 MQT: sound_system/tele/SENSOR = {"Time":"2019-12-20T13:16:34","ENERGY":{"TotalStartTime":"2019-12-15T12:25:12","Total":0.005,"Yesterday":0.000,"Today":0.002,"Power":9.00,"ApparentPower":17.56,"ReactivePower":15.10,"Factor":0.51,"Voltage":234,"Current":0.075}}
13:16:37 MQT: sound_system/tele/SENSOR = {"Time":"2019-12-20T13:16:37","ENERGY":{"TotalStartTime":"2019-12-15T12:25:12","Total":0.005,"Yesterday":0.000,"Today":0.002,"Power":8.80,"ApparentPower":17.56,"ReactivePower":15.20,"Factor":0.50,"Voltage":234,"Current":0.075}}
13:16:39 MQT: sound_system/tele/SENSOR = {"Time":"2019-12-20T13:16:39","ENERGY":{"TotalStartTime":"2019-12-15T12:25:12","Total":0.005,"Yesterday":0.000,"Today":0.002,"Power":9.00,"ApparentPower":17.56,"ReactivePower":15.10,"Factor":0.51,"Voltage":234,"Current":0.075}}
13:16:40 MQT: sound_system/tele/SENSOR = {"Time":"2019-12-20T13:16:40","ENERGY":{"TotalStartTime":"2019-12-15T12:25:12","Total":0.005,"Yesterday":0.000,"Today":0.002,"Power":8.90,"ApparentPower":17.56,"ReactivePower":15.10,"Factor":0.51,"Voltage":234,"Current":0.075}}
13:16:41 MQT: sound_system/tele/SENSOR = {"Time":"2019-12-20T13:16:41","ENERGY":{"TotalStartTime":"2019-12-15T12:25:12","Total":0.005,"Yesterday":0.000,"Today":0.002,"Power":9.10,"ApparentPower":17.56,"ReactivePower":15.00,"Factor":0.52,"Voltage":234,"Current":0.075}}

It should only trigger when the difference is greater than 0.9W??

meingraham, 2019-12-20:
`PowerDelta` `0..100` specifies a percentage change. So `PowerDelta 10` should trigger when there is a 10% change between measured power values. However, this appears to be triggering on 2%-3% rather than 10%.

hberntsen, 2019-12-22:
> PowerDelta not firing when the relay is `OFF` would explain the second message not being published ...

I have no problems with my Shelly Plug S on v7.1.2 except this. When I turn the power off, the web interface reports 0W but no MQTT update is triggered. A mitigation is `SetOption21 1` to always enable power monitoring. When it is on, `PowerDelta` works fine for me. I would expect that PowerDelta trigers an MQTT update when the power is turned off, shall I create a new issue for that?

ascillato, 2019-12-22:
Powerdelta is to publish a mqtt message when the power changes while the relay is ON. To publish or not a mqtt message when the relay is being turned off is not a bug. It is a design decision. Some people wants to have that message. Some others won't.

djashjones, 2019-12-23:
Yes option21 is confusing. "Energy monitoring when power is off"

hberntsen, 2019-12-23:
> Powerdelta is to publish a mqtt message when the power changes while the relay is ON. To publish or not a mqtt message when the relay is being turned off is not a bug. It is a design decision. Some people wants to have that message. Some others won't and find that extra message as inadecuate for their automations.

In that case I'd like to see better documentation for `PowerDelta`. My confusion originates from the fact that I see turning the relay to OFF, a change of the `Power` value to 0W but not a report of power load change w.r.t. the `PowerDelta` option.

ascillato, 2019-12-23:
> In that case I'd like to see better documentation for PowerDelta.

Please help us with that and do a PR to the Tasmota docs. Thanks

ascillato2, 2019-12-25:
Is there anything else that we can do to help on this issue?

meingraham, 2019-12-27:
```
POWER OFF
60W -> 27.5W, within a second, mqtt message fired.
27.5W -> 0W, 3 seconds later, mqtt message not fired.
```

This is likely explained by the timing of the relay state and the lag in the posting of the first message in relation to the instant the relay was turned off where the consumed power was reduced while the second message was "long" after the relay was off and thus PowerDelta would not report that since the relay was off. The update to the docs noting the relay/PowerDelta dependence has been made.

This issue is now answered. Thus closing.

closed 2019-12-27 by meingraham

djashjones, 2019-12-28:
> Just playing around with Betty and set PowerDelta to 10.
>
> ```
> 13:16:28` MQT: sound_system/tele/SENSOR = [...]

Does not answer the above thou?

djashjones, 2019-12-28:
> 
> 
> Yes option21 is confusing. "Energy monitoring when power is off"

Please also explain how option 21 works. Becuase that seems like it does nothing