arendst/Tasmota #7219: Always get "IRSend":"Invalid JSON" when sending raw data

ChrisSutton123, 2019-12-15:
### PROBLEM DESCRIPTION
I am trying to send raw IR data via the console
IRsend 0,896,876,900,888,894,876,1790,874,872,1810,1736,948,872,880,872,936,872,1792,900

Always get "IRSend":"Invalid JSON" error

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 7.1.2
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tasmotizer-1.0.exe
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2019-12-15:
Hi,
can confirm this behaviour. 7.12 precompiled.

s-hadinger, 2019-12-15:
Sending raw IR has been removed from Tasmota default binary. Please use Tasmota-IR to get all the IR protocols and raw sending.

schrej-zz, 2019-12-15:
My bad. Thanks. And what's behind "ir-custom"?
Thanks Joerg

ascillato2, 2019-12-15:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-15 by ascillato2

s-hadinger, 2019-12-15:
Ir-custom is the same as IR but allows you to custom select your compile options

jvera999, 2020-03-18:
Hello, I have this same problem with tasmota-ir.bin 8.1.0 on a Tuya IR Bridge.

I'm using the sample from https://github.com/arendst/Tasmota/wiki/IRSend-RAW-Encoding for testing:

23:07:03 CMD: IRsend 0,926,844,958,832,1798,868,902,848,900,870,900,852,908,918,958,794,934,874,928,1738,934,856,1764
23:07:03 SRC: WebConsole from 192.168.10.5
23:07:03 CMD: Group 0, Index 1, Command "IRSEND", Data "0,926,844,958,832,1798,868,902,848,900,870,900,852,908,918,958,794,934,874,928,1738,934,856,1764"
23:07:03 MQT: stat/IR-01/RESULT = {"IRSend":"Invalid JSON"}
23:07:22 WIF: Checking connection...

The same problem appears if I try to send data I received. ie:

21:42:44 MQT: tele/IR-01/RESULT = {"Time":"2020-03-17T21:42:44","IrReceived":{"Protocol":"UNKNOWN","Bits":37,"Hash":"0xDCE100F1","RawData":[8968,4440,644,1646,640,556,640,556,642,1644,642,1644,666,1620,642,1646,640,556,640,556,640,556,640,556,666,1620,642,558,638,556,644,552,640,556,640,556,640,556,640,556,640,556,648,548,640,1646,642,1644,642,1646,640,556,642,556,640,556,664,532,664,1622,640,556,640,1646,642,556,654,542,664,1622,646,550,640],"RawDataInfo":[73,73,0]}}

Will produce:

23:19:07 CMD: irsend 0,8968,4440,644,1646,640,556,640,556,642,1644,642,1644,666,1620,642,1646,640,556,640,556,640,556,640,556,666,1620,642,558,638,556,644,552,640,556,640,556,640,556,640,556,640,556,648,548,640,1646,642,1644,642,1646,640,556,642,556,640,556,664,532,664,1622,640,556,640,1646,642,556,654,542,664,1622,646,550,640
23:19:07 SRC: WebConsole from 192.168.10.5
23:19:07 CMD: Group 0, Index 1, Command "IRSEND", Data "0,8968,4440,644,1646,640,556,640,556,642,1644,642,1644,666,1620,642,1646,640,556,640,556,640,556,640,556,666,1620,642,558,638,556,644,552,640,556,640,556,640,556,640,556,640,556,648,548,640,1646,642,1644,642,1646,640,556,642,556,640,556,664,532,664,1622,640,556,640,1646,642,556,654,542,664,1622,646,550,640"
23:19:07 MQT: stat/IR-01/RESULT = {"IRSend":"Invalid JSON"}

Here's status 0 output:

23:21:18 CMD: Group 0, Index 1, Command "STATUS", Data "0"
23:21:18 MQT: stat/IR-01/STATUS = {"Status":{"Module":62,"FriendlyName":["Tasmota"],"Topic":"IR-01","ButtonTopic":"0","Power":0,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
23:21:18 MQT: stat/IR-01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-ir.bin","RestartReason":"Power on","Uptime":"0T00:00:59","StartupUTC":"2020-03-18T02:20:19","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":14,"SaveAddress":"F4000"}}
23:21:18 MQT: stat/IR-01/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"326/699"}}
23:21:18 MQT: stat/IR-01/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","2805C8000100060000005A00000000000000","000001C8","00000000"]}}
23:21:18 MQT: stat/IR-01/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":0,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
23:21:18 MQT: stat/IR-01/STATUS5 = {"StatusNET":{"Hostname":"IR-01-5930","IPAddress":"192.168.10.67","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.1","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":4}}
23:21:18 MQT: stat/IR-01/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.40","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_37172A","MqttUser":"ir-01","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
23:21:18 MQT: stat/IR-01/STATUS7 = {"StatusTIM":{"UTC":"Wed Mar 18 02:21:18 2020","Local":"Tue Mar 17 23:21:18 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"-03:00","Sunrise":"02:58","Sunset":"14:58"}}
23:21:18 MQT: stat/IR-01/STATUS10 = {"StatusSNS":{"Time":"2020-03-17T23:21:18"}}
23:21:18 MQT: stat/IR-01/STATUS11 = {"StatusSTS":{"Time":"2020-03-17T23:21:18","Uptime":"0T00:00:59","UptimeSec":59,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WifiCasa","BSSId":"xx:xx:xx:xx:xx:xx","Channel":2,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:07"}}}

I'm sorry for the formatting, using the code blocks seems to look worst.

Thank you!

s-hadinger, 2020-03-18:
You need to use the `Tasmota-IR` version to send raw data.

jvera999, 2020-03-18:
> You need to use the `Tasmota-IR` version to send raw data.

I'm sorry I was sure I was using -IR.bin, but I wasn't. It works now. Thank you!

teewhey, 2021-06-24:
I'm hitting the same issue as well.

```
Ouput of STATUS 0:

15:49:12.195 MQT: stat/smartIR01/STATUS = {"Status":{"Module":62,"DeviceName":"smartIR01","FriendlyName":["smartIR01"],"Topic":"smartIR01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:49:12.202 MQT: stat/smartIR01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://ota.tasmota.com/tasmota/tasmota-ir.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:57","StartupUTC":"2021-06-24T14:44:15","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2021-06-20T14:27:13","SaveCount":49,"SaveAddress":"FA000"}}
15:49:12.209 MQT: stat/smartIR01/STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"402/699"}}
15:49:12.216 MQT: stat/smartIR01/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["LaLiLuLeLo",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000180","00006000","00000000"]}}
15:49:12.230 MQT: stat/smartIR01/STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
15:49:12.239 MQT: stat/smartIR01/STATUS5 = {"StatusNET":{"Hostname":"smartIR01-6472","IPAddress":"192.168.1.58","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"BC:DD:C2:51:99:48","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:49:12.244 MQT: stat/smartIR01/STATUS6 = {"StatusMQT":{"MqttHost":"12.12.12.12","MqttPort":12,"MqttClientMask":"smartIR01","MqttClient":"smartIR01","MqttUser":"hehehe","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:49:12.252 MQT: stat/smartIR01/STATUS7 = {"StatusTIM":{"UTC":"2021-06-24T14:49:12","Local":"2021-06-24T15:49:12","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:47","Sunset":"20:57"}}
15:49:12.259 MQT: stat/smartIR01/STATUS10 = {"StatusSNS":{"Time":"2021-06-24T15:49:12"}}
15:49:12.267 MQT: stat/smartIR01/STATUS11 = {"StatusSTS":{"Time":"2021-06-24T15:49:12","Uptime":"0T00:04:57","UptimeSec":297,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"LaLiLuLeLo","BSSId":"DC:A6:32:EC:BB:E4","Channel":7,"Mode":"11n","RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:03"}}}```

Console Output : 

15:45:04.307 CMD: IRSend 0,350,300,350,600,300,300,650,600,350,300,600,600,650,600,650,600,300,300,350,300,350,250,650,600,300
15:45:04.314 MQT: stat/smartIR01/RESULT = {"IRSend":"Invalid JSON"}
15:53:41.108 CMD: IRsend 0 +3565-1640+480-360+420-1290+430dE-380D-495+475-410J-335+485dJd+425-445CdN-385MdC-1205N-390+375-460Ct+405dCdNrC-1250Cs+455qG-1260+530rMdC-1255E-345Y-365WdC-365N-1235WcU-330YlMdC-1210E-1230MdCd+370-440CvSt+510dJdN-1240C-355J-365N
15:53:41.115 MQT: stat/smartIR01/RESULT = {"IRSend":"Invalid JSON"}
15:53:59.740 CMD: irsend 0,+3565-1640+480-360+420-1290+430dE-380D-495+475-410J-335+485dJd+425-445CdN-385MdC-1205N-390+375-460Ct+405dCdNrC-1250Cs+455qG-1260+530rMdC-1255E-345Y-365WdC-365N-1235WcU-330YlMdC-1210E-1230MdCd+370-440CvSt+510dJdN-1240C-355J-365N
15:53:59.748 MQT: stat/smartIR01/RESULT = {"IRSend":"Invalid JSON"}
```

Sorry for bring up this old topic. Is there something that I missed out? I'm using the version 9.5.0 tasmota-ir.bin file alright.