arendst/Tasmota #7220: Light Color Schemes Not working since 7.1.2

GauthamVarmaK, 2019-12-15:
### PROBLEM DESCRIPTION
The light color schemes no longer work since 7.1.2 . The colors no longer cycle. This could possibly due to the changes made after 7.1.1 to the light color schemes. (This used to work till 7.1.1 so the issue is not with tuya convert.)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): WiproSmartBulb
- [x] Tasmota binary firmware version number used: 7.1.2
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tuya Convert to flash 7.0.x, then ota upgrade
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
06:56:42 MQT: bulb1/stat/RESULT = {"NAME":"WiproSmartBulb","GPIO":[255,255,255,255,37,41,255,255,38,40,39,255,255],"FLAG":0,"BASE":18}
06:56:42 MQT: bulb1/stat/RESULT = {"Module":{"0":"WiproSmartBulb"}}
06:56:42 MQT: bulb1/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"37":"PWM1"},"GPIO5":{"41":"PWM5"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"38":"PWM2"},"GPIO13":{"40":"PWM4"},"GPIO14":{"39":"PWM3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
06:58:02 CMD: Status 0
06:58:02 MQT: bulb1/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Bulb1"],"Topic":"bulb1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
06:58:02 MQT: bulb1/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"5T18:40:15","StartupUTC":"2019-12-09T11:17:47","Sleep":50,"CfgHolder":4617,"BootCount":40,"SaveCount":907,"SaveAddress":"F4000"}}
06:58:02 MQT: bulb1/stat/STATUS2 = {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019-12-06T11:22:22","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285"}}
06:58:02 MQT: bulb1/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MadhuNet",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C800012F060000005A00000000000000","00000200","00000000"]}}
06:58:02 MQT: bulb1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
06:58:02 MQT: bulb1/stat/STATUS5 = {"StatusNET":{"Hostname":"bulb1-7449","IPAddress":"192.168.0.101","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.110","Mac":"24:62:AB:26:1D:19","Webserver":2,"WifiConfig":2}}
06:58:02 MQT: bulb1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.110","MqttPort":1883,"MqttClientMask":"Bulb1","MqttClient":"Bulb1","MqttUser":"SmartBulb","MqttCount":25,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
06:58:02 MQT: bulb1/stat/STATUS7 = {"StatusTIM":{"UTC":"Sun Dec 15 05:58:02 2019","Local":"Sun Dec 15 06:58:02 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:36","Sunset":"16:53"}}
06:58:02 MQT: bulb1/stat/STATUS10 = {"StatusSNS":{"Time":"2019-12-15T06:58:02"}}
06:58:02 MQT: bulb1/stat/STATUS11 = {"StatusSTS":{"Time":"2019-12-15T06:58:02","Uptime":"5T18:40:15","UptimeSec":499215,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":59,"MqttCount":25,"POWER":"OFF","Dimmer":20,"Color":"0,0,0,0,51","HSBColor":"345,100,0","Channel":[0,0,0,0,20],"CT":500,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxx","BSSId":"xx:xx:xx:xx:xx:xx","Channel":11,"RSSI":76,"LinkCount":11,"Downtime":"0T00:04:00"}}}

```

### EXPECTED BEHAVIOUR
The light schemes should set the colors to cycle as with previous verions.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-15:
Fixed on development branch. Please update to latest Tasmota from thehackbox.org/tasmota/tasmota.bin

Th

closed 2019-12-15 by ascillato2

GauthamVarmaK, 2019-12-15:
Sorry for re-opening the issue, but color schemes are still partially broken causing the following issues:

1. Color schemes only work if one the RGB color was initially selected before selecting the color scheme. 
2. The time to cycle through the colors is almost same between 2,3 and 4.
3. There is noticeable flicker while using the scheme 4 while there is little flicker with schemes 2,3.

GauthamVarmaK, 2019-12-15:
Kindly reopen this issue so we could close it once the related issues are fixed.

ascillato, 2019-12-15:
Please update to latest Tasmota from http://thehackbox.org/tasmota/tasmota.bin and test again.