arendst/Tasmota #7222: reset a device that has no config

Chavell3, 2019-12-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
my flashed Blitzwolf BW-SHP2(first flashed to 6.6.0) is not reachable after an upgrade to the latest firmware via the plugs web interface.
I had installed the 6.6.0 release what was working some time and now wanted to upgrade to the 6.7.1 release but after the upgrade it seems the device has no config, as it is not reacting to the button press nor it is connecting to my wifi(what has worked before).
I've searched, how to reset the device so it would publish it's default wifi, but all I've read about are not working.
I tried fast poweron and poweroff 10 times, 4 times... (what I have read somewhere) but nothing changes...
Maybe some of you have an idea how to reset the device ?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Blitzwolf BW-SHP2
- [x] Tasmota binary firmware version number used: 6.6.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: tuya-convert
- [ ] Flashing tools used: tuya-convert
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
not applicable
  Configuration output here:
not applicable
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
not applicable
  Rules output here:
not applicable
- [ ] Provide the output of this command: `Status 0`:
not applicable
  STATUS 0 output here:
not possible
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
not applicable
  Console output here:
not applicable

### TO REPRODUCE
_Steps to reproduce the behavior:_
not recommended

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
after the update I thought the set settings will remain...

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
not applicable

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Maybe some of you have an idea how to reset the device ?

Jason2866, 2019-12-15:
Have you looked, if the device opens a AP?

Chavell3, 2019-12-15:
yes that was one of the first thing I checked too, should have mentioned that.

closed 2019-12-15 by ascillato2