arendst/Tasmota #7224: De-Brick a SmartPlug GSP-1311D after dying due to testing templates to find working one

Darkyputz, 2019-12-15:
PROBLEM DESCRIPTION

I had this Smartplug working on 7.1.2 and was trying to find a working configuration to have it correct working according to the offered buttons in different configurations.
Sonoff 4port was onlt one working working with it but since my plug has only two outlets, only button 1 and four were working.
So after i went through all the configurations, i was trying my luck with the templates in the wiki...
Checked a couple dual plugs and after some working and some dont i ended up with "Gosund WP2 Plug " template which, after the save button, made the plug unreachable.
Now i am looking for a way to get this baby back to life.
It does blink one very short when plugged in but that is about it.
Tried 1 min+ pressing the button and tried tuya-convert but no life sign, no pinging.

Maybe some of you have an idea how to reset the device ?
Hopefully without breaking it open :-)

REQUESTED INFORMATION

Read the Contributing Guide and Policy and the Code of Conduct
Checked: Searched the problem in issues
Checked: Searched the problem in the wiki
Checked: Searched the problem in the forum
Device used (e.g., Sonoff Basic): Smart Plug GSP-1311D
Tasmota binary firmware version number used: 7.1.2 Pre-compiled
 
Flashing tools used: tuya-convert
Provide the output of command: Backlog Template; Module; GPIO 255: not applicable

TO REPRODUCE
not recommended

EXPECTED BEHAVIOUR

booting plug with two working buttons

SCREENSHOTS
not applicable

ADDITIONAL CONTEXT

Maybe some of you have an idea how to reset the device ?

ascillato, 2019-12-15:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Darkyputz, 2019-12-15:
Will do...thx for the advice

closed 2019-12-15 by ascillato2