arendst/Tasmota #7225: TUYA Dimmer. The power state setting does not work correctly when power is on

Remeslenik, 2019-12-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
TUYA Dimmer. The power state setting does not work correctly when power is on.
When I press the off button in the console and reboot the device from the console, the power state is saved correctly.
But if you press the off button and turn the power off / on, the device boots up with the load on 100%.
The log shows that it starts in the off state.

00:00:05 RSL: INFO1 = {"Module":"RJWF-02A","Version":"7.1.2(tasmota)","FallbackTopic":"cmnd/DVES_3D0090_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:05 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-0144","IPAddress":"192.168.0.104"}
00:00:05 RSL: INFO3 = {"RestartReason":"Power on"}
00:00:05 RSL: RESULT = {"POWER":"OFF"}
00:00:05 RSL: POWER = OFF

And after loading, in the console the state is turned off.
But physically the load is turned on 100%, no matter how many percent the dimmer was configured before turning off the power.
After that, if I press the change button. The status will change to ON and the dimmer settings will be set as before the power was turned off.
The next press, sets the off status and the load is disconnected.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): RJWF-02A https://blakadder.github.io/templates/rjwf-02a.html
- [x] Tasmota binary firmware version number used: 7.1.2
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmota-PyFlasher-1.0
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
00:01:20 CMD: Backlog Template; Module; GPIO 255
00:01:20 SRC: WebConsole from 192.168.0.105
00:01:20 CMD: Группа 0, Индекс 1, Команда "BACKLOG", Данные "Template; Module; GPIO 255"
00:01:20 SRC: Backlog
00:01:20 CMD: Группа 0, Индекс 1, Команда "TEMPLATE", Данные ""
00:01:20 RSL: RESULT = {"NAME":"RJWF-02A","GPIO":[17,107,0,108,0,0,0,0,0,108,52,0,0],"FLAG":0,"BASE":54}
00:01:20 SRC: Backlog
00:01:20 CMD: Группа 0, Индекс 1, Команда "MODULE", Данные ""
00:01:20 RSL: RESULT = {"Module":{"0":"RJWF-02A"}}
00:01:21 SRC: Backlog
00:01:21 CMD: Группа 0, Индекс 1, Команда "GPIO", Данные "255"
00:01:21 RSL: RESULT = {"GPIO0":{"17":"Кнопка1"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"-нет-"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"-нет-"},"GPIO5":{"0":"-нет-"},"GPIO9":{"0":"-нет-"},"GPIO10":{"0":"-нет-"},"GPIO12":{"0":"-нет-"},"GPIO13":{"108":"Tuya Rx"},"GPIO14":{"52":"Led1"},"GPIO15":{"0":"-нет-"},"GPIO16":{"0":"-нет-"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
15:36:21 CMD: Status 0
15:36:21 SRC: WebConsole from 192.168.0.105
15:36:21 CMD: Группа 0, Индекс 1, Команда "STATUS", Данные "0"
15:36:21 RSL: STATUS = {"Status":{"Module":0,"FriendlyName":["WC_Vents"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:36:21 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:52","StartupUTC":"2019-12-15T14:31:29","Sleep":50,"CfgHolder":4617,"BootCount":11,"SaveCount":47,"SaveAddress":"F5000"}}
15:36:21 RSL: STATUS2 = {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019-12-06T11:47:51","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
15:36:21 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Chulan","Recovery"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000000","00000000"]}}
15:36:21 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000419","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
15:36:21 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-0144","IPAddress":"192.168.0.104","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"EC:FA:BC:3D:00:90","Webserver":2,"WifiConfig":4}}
15:36:21 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sun Dec 15 14:36:21 2019","Local":"Sun Dec 15 15:36:21 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:36","Sunset":"16:53"}}
15:36:21 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-12-15T15:36:21"}}
15:36:21 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-12-15T15:36:21","Uptime":"0T00:04:52","UptimeSec":292,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":9,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Chulan","BSSId":"00:90:4C:C1:00:00","Channel":1,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
00:00:00 CFG: Загружено из флэш-памяти FA, Подсчет 34
00:00:00 QPC: Flag 0E
00:00:00 SRC: Restart
00:00:00 TYA: Send dim skipped value=0
00:00:00 Проект tasmota WC_Vents Версия 7.1.2(tasmota)-2_6_1
00:00:00 TYA: Request MCU configuration
00:00:00 TYA: Send "55aa0001000000"
00:00:00 TYA: Set WiFi LED 2 (-1)
00:00:00 TYA: Send "55aa000300010205"
00:00:00 WIF: Проверка соединения...
00:00:00 WIF: Попытка подключения...
00:00:00 WIF: Подключение к AP1 Chulan в режиме 11N как tasmota-0144...
00:00:01 WIF: Проверка соединения...
00:00:01 WIF: Попытка подключения...
00:00:02 WIF: Проверка соединения...
00:00:02 WIF: Попытка подключения...
00:00:04 WIF: Проверка соединения...
00:00:04 WIF: Попытка подключения...
00:00:05 WIF: Проверка соединения...
00:00:05 WIF: Соединен
00:00:05 HTP: Веб-сервер активен tasmota-0144 с IP-адресом 192.168.0.104
00:00:05 RSL: INFO1 = {"Module":"RJWF-02A","Version":"7.1.2(tasmota)","FallbackTopic":"cmnd/DVES_3D0090_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:05 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-0144","IPAddress":"192.168.0.104"}
00:00:05 RSL: INFO3 = {"RestartReason":"Power on"}
00:00:05 RSL: RESULT = {"POWER":"OFF"}
00:00:05 RSL: POWER = OFF
00:00:05 TYA: Set WiFi LED 3 (0)
00:00:05 TYA: Send "55aa000300010306"
00:00:06 QPC: Reset
00:00:08 HTP: Главное Меню
00:00:08 APP: Количество загрузок 9
00:00:08 CFG: Сохранено во флэш-память F9, Подсчет 35, Байт 4096
00:00:09 RSL: STATE = {"Time":"1970-01-01T00:00:09","Uptime":"0T00:00:11","UptimeSec":11,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":9,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Chulan","BSSId":"00:90:4C:C1:00:00","Channel":1,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:06"}}
00:00:09 TYA: Send "55aa00000000ff"
00:00:13 HTP: Консоль
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Set the dimmer to any position.
2. Turn off the load in the console.
3. Turn off the power.
4. Turn on the power.
5. Observe the connected load at 100% power and Status off in the console.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The load must remain state after a power failure

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Logs between normal and abnormal behavior differ only in the reason for rebooting the device.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-15:
Please, try latest Tasmota version from the development branch (thehackbox.org/tasmota/tasmota.bin) thanks

Remeslenik, 2019-12-15:
Tried. All the same. Starts with load on. Then the first switch, the dimming changes. The second switch is turned off.
```
00:00:00 CFG: Loaded from flash at F9, Count 61
00:00:00 QPC: Flag 0E
00:00:00 SRC: Restart
00:00:00 TYA: Send dim skipped value=0
00:00:00 Project tasmota WC_Vents Version 7.1.2.6(7aab540-tasmota)-2_6_1
00:00:00 TYA: Request MCU configuration
00:00:00 TYA: Send "55aa0001000000"
00:00:00 TYA: Set WiFi LED 2 (-1)
00:00:00 TYA: Send "55aa000300010205"
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Chulan in mode 11N as tasmota-0144...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota-0144 with IP address 192.168.0.104
00:00:05 RSL: INFO1 = {"Module":"RJWF-02A","Version":"7.1.2.6(7aab540-tasmota)","FallbackTopic":"cmnd/DVES_3D0090_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:05 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-0144","IPAddress":"192.168.0.104"}
00:00:05 RSL: INFO3 = {"RestartReason":"External System"}
00:00:05 RSL: RESULT = {"POWER":"OFF"}
00:00:05 RSL: POWER = OFF
00:00:05 TYA: Set WiFi LED 3 (0)
00:00:05 TYA: Send "55aa000300010306"
16:56:31 NTP: Drift 0, (UTC) Sun Dec 15 15:56:31 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
16:56:31 QPC: Reset
16:56:33 APP: Boot Count 13
16:56:33 CFG: Saved to flash at F8, Count 62, Bytes 4096
16:56:34 RSL: STATE = {"Time":"2019-12-15T16:56:34","Uptime":"0T00:00:11","UptimeSec":11,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":43,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Chulan","BSSId":"00:90:4C:C1:00:00","Channel":1,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:06"}}
16:56:34 TYA: Send "55aa00000000ff"
16:56:41 SRC: WebGui from 192.168.0.105
16:56:41 TYA: Send dim value=43 (id=2)
16:56:41 TYA: Send "55aa00060008020200040000002b40"
16:56:41 TYA: Send "55aa0006000501010001010e"
16:56:41 RSL: RESULT = {"POWER":"ON"}
16:56:41 RSL: POWER = ON
16:56:42 CFG: Saved to flash at F7, Count 63, Bytes 4096
16:56:43 SRC: WebGui from 192.168.0.105
16:56:43 TYA: Send dim skipped value=0
16:56:43 TYA: Send "55aa0006000501010001000d"
16:56:43 RSL: RESULT = {"POWER":"OFF"}
16:56:43 RSL: POWER = OFF
16:56:43 CFG: Saved to flash at F6, Count 64, Bytes 4096
16:56:45 TYA: Send "55aa00000000ff"
16:56:50 WIF: Checking connection...
16:56:50 WIF: Connected
16:56:56 TYA: Send "55aa00000000ff"
16:57:07 TYA: Send "55aa00000000ff"
16:57:10 WIF: Checking connection...
16:57:10 WIF: Connected
```

Remeslenik, 2019-12-15:
And log after reboot from console. Work fine.

```
00:00:00 CFG: Loaded from flash at FB, Count 67
00:00:00 QPC: Flag 0E
00:00:00 SRC: Restart
00:00:00 TYA: Send dim skipped value=0
00:00:00 Project tasmota WC_Vents Version 7.1.2.6(7aab540-tasmota)-2_6_1
00:00:00 TYA: Request MCU configuration
00:00:00 TYA: Send "55aa0001000000"
00:00:00 TYA: Set WiFi LED 2 (-1)
00:00:00 TYA: Send "55aa000300010205"
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Chulan in mode 11N as tasmota-0144...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota-0144 with IP address 192.168.0.104
00:00:05 RSL: INFO1 = {"Module":"RJWF-02A","Version":"7.1.2.6(7aab540-tasmota)","FallbackTopic":"cmnd/DVES_3D0090_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:05 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-0144","IPAddress":"192.168.0.104"}
00:00:05 RSL: INFO3 = {"RestartReason":"Software/System restart"}
00:00:05 RSL: RESULT = {"POWER":"OFF"}
00:00:05 RSL: POWER = OFF
00:00:05 TYA: Set WiFi LED 3 (0)
00:00:05 TYA: Send "55aa000300010306"
17:23:48 NTP: Drift 0, (UTC) Sun Dec 15 16:23:48 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
17:23:48 QPC: Reset
17:23:49 HTP: Main Menu
17:23:50 APP: Boot Count 15
17:23:50 CFG: Saved to flash at FA, Count 68, Bytes 4096
17:23:51 RSL: STATE = {"Time":"2019-12-15T17:23:51","Uptime":"0T00:00:11","UptimeSec":11,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":43,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Chulan","BSSId":"00:90:4C:C1:00:00","Channel":1,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:06"}}
17:23:51 TYA: Send "55aa00000000ff"
17:23:59 HTP: Console
17:24:02 TYA: Send "55aa00000000ff"
17:24:07 WIF: Checking connection...
17:24:07 WIF: Connected
```

Remeslenik, 2019-12-17:
I understood. The ESP module does not send stored values to the controller when power is applied. When rebooting from the console, only the ESP module is reloaded. In the dimmer controller, the values are saved. If you turn off the power, the controller will reset and will not get the saved settings.
Solved the problem by the rules:
```
Rule1 ON System#Boot DO event toggling=%mem2% ENDON
on event#toggling=0 do  TuyaSend1 1,0 endon
on event#toggling=1 do  TuyaSend2 2,%mem1% endon

Rule2 ON Dimmer#State DO mem1 %value% ENDON
ON Power1#State DO mem2 %value% ENDON
```
But the dimmer starts very late. And the load runs at full power about 6 seconds. I think that the TUYA should start first and the settings should be passed by default.

blakadder, 2019-12-18:
System#Boot occurs once after Tasmota is intialised (after the INFO1, INFO2 and INFO3 console messages) which is why you have a delay. 

Power state is stored by using `PowerRetain 1`.  Dimmer value doesn't have that option and you have to use mem

Remeslenik, 2019-12-19:
Run rules earlier than System#Boot, does not give a positive effect.

Remeslenik, 2020-01-11:
I flashed the corrected firmware from the topic #7412 It works incorrectly, after turning OFF the load:
```
18:00:57 SRC: WebGui from 192.168.0.105
18:00:57 TYA: Send dim skipped value=0
18:00:57 TYA: Send "55aa0006000501010001000d"
18:00:57 RSL: STATE = {"Time":"2020-01-11T18:00:57","Uptime":"0T00:04:33","UptimeSec":273,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":99,"MqttCount":0,"POWER":"OFF","Dimmer":37,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Chulan","BSSId":"00:90:4C:C1:00:00","Channel":1,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:06"}}
18:00:57 RSL: RESULT = {"POWER":"OFF"}
18:00:57 RSL: POWER = OFF
18:00:57 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpId":1,"DpIdType":1,"DpIdData":"00"}}
18:00:57 TYA: fnId=11 is set for dpId=1
18:00:57 TYA: RX Relay-1 --> MCU State: Off Current State:Off
18:00:57 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpId":1,"DpIdType":1,"DpIdData":"00"}}
18:00:57 TYA: fnId=11 is set for dpId=1
18:00:57 TYA: RX Relay-1 --> MCU State: Off Current State:Off
18:00:57 {"TuyaReceived":{"Data":"55AA00070008020200040000005C72","Cmnd":7,"CmndData":"020200040000005C","DpId":2,"DpIdType":2,"DpIdData":"0000005C"}}
18:00:57 TYA: fnId=21 is set for dpId=2
18:00:57 TYA: RX Dim State=92
18:00:57 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpId":1,"DpIdType":1,"DpIdData":"00"}}
18:00:57 TYA: fnId=11 is set for dpId=1
18:00:57 TYA: RX Relay-1 --> MCU State: Off Current State:Off
18:00:57 {"TuyaReceived":{"Data":"55AA00070008020200040000001026","Cmnd":7,"CmndData":"0202000400000010","DpId":2,"DpIdType":2,"DpIdData":"00000010"}}
18:00:57 TYA: fnId=21 is set for dpId=2
18:00:57 TYA: RX Dim State=16
18:00:57 CFG: Saved to flash at F5, Count 855, Bytes 4096
18:00:58 HTP: Console
```
And power restart, load is turning ON:
```
00:00:00 TYA: fnId=11 is set for dpId=1
00:00:00 TYA: RX Relay-1 --> MCU State: On Current State:Off
00:00:00 SRC: Switch
00:00:00 TYA: Send dim value=85 (id=2)
00:00:00 TYA: Send "55aa0006000802020004000000556a"
00:00:00 RSL: STATE = {"Time":"1970-01-01T00:00:00","Uptime":"0T00:00:01","UptimeSec":1,"Heap":31,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":30,"MqttCount":0,"POWER":"ON","Dimmer":34,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Chulan","BSSId":"FF:FF:FF:FF:FF:FF","Channel":3,"RSSI":100,"Signal":31,"LinkCount":0,"Downtime":"0T00:00:00"}}
00:00:00 RSL: RESULT = {"POWER":"ON"}
00:00:00 RSL: POWER = ON
00:00:00 {"TuyaReceived":{"Data":"55AA0007000802020004000000FF15","Cmnd":7,"CmndData":"02020004000000FF","DpId":2,"DpIdType":2,"DpIdData":"000000FF"}}
00:00:00 TYA: fnId=21 is set for dpId=2
00:00:00 TYA: RX Dim State=255
00:00:00 SRC: Switch
00:00:00 CMD: Group 0, Index 1, Command "DIMMER", Data "100"
00:00:00 RSL: STATE = {"Time":"1970-01-01T00:00:00","Uptime":"0T00:00:01","UptimeSec":1,"Heap":31,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":30,"MqttCount":0,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Chulan","BSSId":"FF:FF:FF:FF:FF:FF","Channel":3,"RSSI":100,"Signal":31,"LinkCount":0,"Downtime":"0T00:00:00"}}
00:00:00 RSL: RESULT = {"POWER":"ON","Dimmer":100}
00:00:00 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpId":1,"DpIdType":1,"DpIdData":"01"}}
00:00:00 TYA: fnId=11 is set for dpId=1
00:00:00 TYA: RX Relay-1 --> MCU State: On Current State:On
00:00:00 {"TuyaReceived":{"Data":"55AA0007000802020004000000647A","Cmnd":7,"CmndData":"0202000400000064","DpId":2,"DpIdType":2,"DpIdData":"00000064"}}
00:00:00 TYA: fnId=21 is set for dpId=2
00:00:00 TYA: RX Dim State=100
00:00:00 SRC: Switch
00:00:00 CMD: Group 0, Index 1, Command "DIMMER", Data "40"
00:00:00 RSL: STATE = {"Time":"1970-01-01T00:00:00","Uptime":"0T00:00:01","UptimeSec":1,"Heap":31,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":30,"MqttCount":0,"POWER":"ON","Dimmer":40,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Chulan","BSSId":"FF:FF:FF:FF:FF:FF","Channel":1,"RSSI":100,"Signal":31,"LinkCount":0,"Downtime":"0T00:00:00"}}
00:00:00 RSL: RESULT = {"POWER":"ON","Dimmer":40}
00:00:00 TYA: Send dim skipped value=100
00:00:01 CFG: Saved to flash at F8, Count 852, Bytes 4096
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota-0144 with IP address 192.168.0.103
00:00:05 RSL: INFO1 = {"Module":"RJWF-02A","Version":"8.1.0.3(tasmota)","FallbackTopic":"cmnd/DVES_3D0090_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:05 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-0144","IPAddress":"192.168.0.103"}
00:00:05 RSL: INFO3 = {"RestartReason":"External System"}
00:00:05 RSL: RESULT = {"POWER":"ON"}
00:00:05 RSL: POWER = ON
00:00:05 TYA: Set WiFi LED 3 (0)
00:00:05 TYA: Send "55aa000300010306"
00:00:05 {"TuyaReceived":{"Data":"55AA0003000002","Cmnd":3}}
00:00:05 TYA: RX WiFi LED set ACK
17:56:32 NTP: Drift 0, (UTC) Sat Jan 11 16:56:32 2020, (DST) Sun Mar 29 02:00:00 2020, (STD) Sun Oct 25 03:00:00 2020
17:56:32 QPC: Reset
17:56:34 APP: Boot Count 78
17:56:34 CFG: Saved to flash at F7, Count 853, Bytes 4096
17:56:35 HTP: Console
[... cut ...] e":"0T00:00:06"}}
17:56:35 TYA: Send "55aa00000000ff"
17:56:35 {"TuyaReceived":{"Data":"55AA000000010000","Cmnd":0,"CmndData":"00"}}
17:56:35 TYA: Heartbeat
17:56:35 TYA: Detected MCU restart
17:56:36 TYA: Set WiFi LED 3 (0)
17:56:36 TYA: Send "55aa000300010306"
17:56:36 {"TuyaReceived":{"Data":"55AA0003000002","Cmnd":3}}
17:56:36 TYA: RX WiFi LED set ACK
17:56:46 TYA: Send "55aa00000000ff"
17:56:46 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
17:56:46 TYA: Heartbeat
```

ascillato2, 2020-01-15:
As this issue is the same as #7412, let's continue this thread there. Thanks.

closed 2020-01-15 by ascillato2