arendst/Tasmota #7227: DS16xx temperature sensors

BASM, 2019-12-15:
**Have you looked for this feature in other issues and in the wiki?**  

Yes

**Is your feature request related to a problem? Please describe.**  
I can't use DS1624 and other DS16xx sensors.

**Describe the solution you'd like**  
I can add support for this, if no one does it and does not mind.

**Describe alternatives you've considered**  
Buy another sensors? booooo

**Additional context**  
I think to get the LM75AD as example, or give me a better example

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-15:
> I can add support for this, if no one does it and does not mind.

Great. Please do a PR. If you need help, please just ask. Thanks

BASM, 2019-12-20:
This is sensors support up to 8 devices on the i2c bus. This commit will be display only one device on the i2c bus. I did not understand how to make support for all devices, create 8 functions Xsns59 .. Xsns66 ??? Or how?

ascillato2, 2019-12-20:
If you want, you can check the ds18x20 driver on how to implement that https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_05_ds18x20.ino

PR #7587 (2020-01-24): ds1624/ds1621: Stabilization and bug fixes
* ds1621 datasheet is wrong for higher resolution, fixed
* ds1621 sometimes reset on bad (long) i2c bus, added autoreconfigure
* hotplug bug fixes

Issues are extremely related and apply only to ds1624. It makes no sense to make three commits.

## Description:

**Related issue (if applicable):** fixes #7227

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [x] The code change  [... cut]
closed 2019-12-20 by ascillato2