arendst/Tasmota #7228: Extend user_config_override.h

RedFoxy, 2019-12-15:
**Have you looked for this feature in other issues and in the wiki?** 
Yes I've 

**Is your feature request related to a problem? Please describe.**  
No it isn't

**Describe the solution you'd like**  
I use a lot of user_config_override.h because I prefer to put my configurations directly in the code so there is a trouble and the device delete its config, the default config is the right one to use it, but I can't put everything in user_config_override.h, like I can't put templates (I have some SWA1), or timer settings, shutter configurations and other, I can only change few variables, it will be usefull to have more in user_config_override.h

**Describe alternatives you've considered**  
I tryied to add SWA1 support in tasmota_template.h but I got error, and, as user_config_override.h, everytime I use git pull

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-12-16:
What do you want to have? If you dont tell exactly it cant be checked nor considered.

ascillato2, 2019-12-17:
Closing this issue as there is no feedback.

closed 2019-12-17 by ascillato2

RedFoxy, 2019-12-18:
sorry for late I was responding, give me a minute

RedFoxy, 2019-12-18:
btw, we can use user_config_override.h to change only few settings, like WiFi, MQTT, IP address, project and riendly name, but we can't add settings like pulsetime, shutter_position and all other commands https://github.com/arendst/Tasmota/wiki/commands and templates https://github.com/arendst/Tasmota/wiki/Templates

It will be usefull to use every commands directly in user_config_override.h so we can prepare a specific firmware for each device, and if the device isn't support yet, we can add support using template not by webinterface but directly by user_config_override.h

arendst, 2019-12-18:
Then we would be back at square ONE where every feature configured would need another compilation.

No-go. Learn to use commands like ``backlog`` to configure your device in bulk.

RedFoxy, 2019-12-18:
> Then we would be back at square ONE where every feature configured would need another compilation.
> 
> No-go. Learn to use commands like `backlog` to configure your device in bulk.

I know that for every feature I need to compile it, but I can use that to prepare firmware to send remotely and to have device with my specified basic configuration, so if I need to hardreset it or for any trouble it will be hard reset, it will restart with my basic configuration, aany extra can be done by web and command

RedFoxy, 2019-12-18:
Just to explain my trouble, I used 4 sonof 4ch pro for 4 shutters for a friend, the day after, he called me tolding me that his childs statrs to play with the shutters and now it doesn't goes anymore, I tryed to connect from remote but I don't find it than I move to my friend home and connect directly to one of them, it was to default settings so it doesn't goes, than I prepare a new firmware version with his wifi and all settings that I can change in user_config_override.h, after that, I don't know what his childs do with that shutters, it was to default settings another time, but now it was already connected to the wifi because I forced it directly in the firmware, after that I make a specific user_config_override.h for every my device, like electric switches and other.

I don't want replace the web and command line, just I'm looking for an additional way to configure it

Letzdeal3, 2021-02-08:
Hi RedFoxy, did you ever get this to work. I would like to do something similar

RedFoxy, 2021-02-11:
I fixed about the sonoff trouble changing it with Shelly 2.5.

About the user_config_override.h, just I download lastes Tasmota source, edit  user_config_override.h adding:
#include "_00_select_project.h"

Than edit my_user_config.h and uncomment (or add)
#define USE_CONFIG_OVERRIDE

in _00_select_project.h I've include of various project, I uncomment right file to compile it, like:

//#include "_10_wemos_temp_rest_room.h"
//#include "_19_wemos_gate.h"
//#include "_19_sonoff_ext_light.h"
#include "_19_sonoff_s20_desk_light.h"

so in _19_sonoff_s20_desk_light.h I've all the customizations about that harfware like:

#include "_10_generic_data.h"

#define PROJECT         "desk_light"

#define FRIENDLY_NAME   "Red desk light"
#define MODULE          SONOFF_S2X

#define WIFI_IP_ADDRESS "192.168.1.19"

in _10_generic_data.h I've all the common customizzations like:

#undef PROJECT
#undef FRIENDLY_NAME
#undef WIFI_IP_ADDRESS
#undef WIFI_GATEWAY
#undef WIFI_SUBNETMASK
#undef WIFI_DNS
#undef STA_SSID1
#undef STA_PASS1
#undef STA_SSID2
#undef STA_PASS2
#undef MQTT_HOST
#undef MQTT_PORT
#undef MQTT_USER
#undef MQTT_PASS
#undef MQTT_CLIENT_ID
#undef LATITUDE
#undef LONGITUDE
#undef CFG_HOLDER
#undef NTP_SERVER1
#undef NTP_SERVER2
#undef NTP_SERVER3
#undef MY_LANGUAGE

#define WIFI_GATEWAY    "192.168.1.1"
#define WIFI_SUBNETMASK "255.255.255.0"
#define WIFI_DNS        "8.8.8.8"

#define STA_SSID1       "My WiFi SSID"
#define STA_PASS1       "MyPassword`"

#define STA_SSID2       "My Alternative WiFi SSID"
#define STA_PASS2       "MyPassword2"

#define MQTT_HOST       "192.168.1.100"
#define MQTT_PORT       1883
#define MQTT_USER       "MQTTUSER"
#define MQTT_PASS       "MQTT_PASS"
#define MQTT_CLIENT_ID  "SWITCH_%06X"

#define LATITUDE        1.234567
#define LONGITUDE       1.234567
#define CFG_HOLDER      9521

#define NTP_SERVER1     "it.pool.ntp.org"
#define NTP_SERVER2     "0.it.pool.ntp.org"
#define NTP_SERVER3     "pool.ntp.org"

#define MY_LANGUAGE            it_IT

#define EMULATION       EMUL_WEMO
#define HOME_ASSISTANT_DISCOVERY_ENABLE   1      // [SetOption19] Home Assistant Discovery (0 = Disable, 1 = Enable)

I hope that it will be usefull for you