arendst/Tasmota #7229: I2C DHT12 Support (Contribution)

stefan-oskamp, 2019-12-15:
**Have you looked for this feature in other issues and in the wiki?**  
Yes
**Is your feature request related to a problem? Please describe.**  
Support for <https://wiki.wemos.cc/products:retired:dht_shield_v2.0.0>

**Describe the solution you'd like**  
Review and include driver code that I can contribute.

**Describe alternatives you've considered**  
Compile my own Tasmota version

**Additional context**  
I have compiled my own version of Tasmota 6.4.1 with code for the DHT12 shield in xsns_10_*.ino
It works fine, but updating Tasmota would be difficult, because the driver is not properly integrated.
So, therefore, I would like to contribute the attached source code for your review and inclusion in Tasmota.

**(Please, remember to close the issue when the problem has been addressed)**  
Sure.

Regards,
Stefan

[xsns_10_dht12.ino.txt](https://github.com/arendst/Tasmota/files/3965240/xsns_10_dht12.ino.txt)

ascillato, 2019-12-15:
For reference, this was discussed before:

https://github.com/arendst/Tasmota/issues/4768

More code also at:

https://github.com/xreef/DHT12_sensor_library

stefan-oskamp, 2019-12-15:
Hi Adrian,
Thanks for the references, but what is the message?
Just to clarify: The DHT12 shield is I2C and it is working fine with Tasmota using the code I've attached in my request. 
If you do not want to include the driver in Tasmota that's also fine with me.

[image]

ascillato2, 2019-12-15:
Hi,

There were no message, I have just added a reference to complete your request.

As DHT12 is Temp I2C sensor, you can base a new driver for it on the BME680 driver (https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_09_bmp.ino).

You can make a new file like _xsns_58_dht12.ino_ and do a PR to include it on Tasmota.

As you have the sensor and also you have a working file, you are almost done to do a new driver.

If you need assistance, please just ask.

stefan-oskamp, 2019-12-15:
Did you see the screenshot? I did what you just suggested some time ago. I just thought others might find it useful, too. And with the tasmota-sensors.bin alternative build, space for another tiny I2C driver should not be an issue. Again: The driver code was attached to my request.

ascillato, 2019-12-15:
Yes, I saw your screenshot and your file. :+1: 

As you said:

>It works fine, but updating Tasmota would be difficult, because the driver is not properly integrated. So, therefore, I would like to contribute the attached source code for your review and inclusion in Tasmota.

That is why I'm asking you to see and use as a base the other driver (that is properly integrated) and make a Pull Request to Tasmota to include it. You are almost done. Your driver is OK. You need to use a free driver number like for example _xsns_58_dht12.ino_.

Please, so a Pull Request. If you need assistance on that, please just ask.

stefan-oskamp, 2019-12-15:
Thanks for the clarification: I'll check out how to make a pull request.

ascillato, 2019-12-15:
The fastest way is to:

* go to https://github.com/arendst/Tasmota/tree/development/tasmota
* click on create new file.
* name it (xsns_58_dht12.ino).
* copy your code inside it.
* click on propose new file
* then click on new pull request.

BASM, 2019-12-16:
> The fastest way is to:
> 
> * name it (xsns_58_dht12.ino).

No, please use xsns_59_dht12.ino, I already i already took 58 for DS16xx sensors on the #7227

Jason2866, 2019-12-16:
@BASM  I see no PR where driver number 58 is used

localhost61, 2019-12-16:
@ascillato thanks for the tip, I didn't know that way to do.

ascillato2, 2019-12-17:
Closing this issue as the PR #7245 has been proposed for review.

Thanks

PR #7245 (2019-12-20): Support for I2C sensor DHT12
## Description:

**Related issue (if applicable):** fixes #7229

## Checklist:
  - [X] The pull request is done against the latest dev branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR.
  - [X] The code change is tested and works on core 2.6.1
  - [X] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [X] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).
closed 2019-12-17 by ascillato2