arendst/Tasmota #7232: Controlling ws2812 LED strip

wilqp, 2019-12-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When I turn tasmota on I can control whole LED strip (300xWS2812 LEDs) for few minutes. However after ~20min. I can only control just the part at the beginning (first 50 or so), rest of the LED strip is still on. 
in previous version(6.7.1) toggling LEDs off  and tasomta reset used to help to regain full control. however now it's not working anymore( on ver 7.1.2.6) , and I have to unplug it for a few seconds each time. Tried to use Domoticz and webGUI(with Domoticz turned off& turned on) 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic):  esp8266 nodemcu V2
- [ ] Tasmota binary firmware version number used: 7.1.2.6
  - [x ] Pre-compiled OTA upgrade
  - [ X] Self-compiled
    - [X ] IDE / Compiler used: ATOM
- [ X] Flashing tools used: NodeMCU-PyFlasher-4.0-x64
- [X ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
00:00:00 CFG: Wczytano z pamięci flash w F9, Licz 83
00:00:00 Projekt tasmota Sonoff-LED Wersja 7.1.2.6(tasmota)-2_6_1
00:00:00 WIF: Łączenie z AP1 Router w trybie 11N jak sonoff-LED-5837...
00:00:05 WIF: Połączony
00:00:05 HTP: Aktywny serwer Web sonoff-LED-5837 z adresem IP 192.XXX.XXX.XXX
22:05:07 MQT: Próba połączenia...
22:05:07 MQT: Połączony
22:05:07 MQT: tele/sonoff-LED/LWT = Aktywny (Zachowane)
22:05:07 MQT: cmnd/sonoff-LED/POWER = 
22:05:07 MQT: tele/sonoff-LED/INFO1 = {"Module":"Generic","Version":"7.1.2.6(tasmota)","FallbackTopic":"cmnd/DVES_1836CD_fb/","GroupTopic":"cmnd/tasmotas/"}
22:05:07 MQT: tele/sonoff-LED/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-LED-5837","IPAddress":"192.XXX.XXX.XXX"}
22:05:07 MQT: tele/sonoff-LED/INFO3 = {"RestartReason":"Software/System restart"}
22:05:07 MQT: stat/sonoff-LED/RESULT = {"POWER":"OFF"}
22:05:07 MQT: stat/sonoff-LED/POWER = OFF
22:05:08 MQT: domoticz/in = {"idx":81,"nvalue":0,"svalue":"100","Battery":100,"RSSI":10}
22:05:11 MQT: tele/sonoff-LED/STATE = {"Time":"2019-12-15T22:05:11","Uptime":"0T00:00:12","UptimeSec":12,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":100,"Color":"FFEFD4","HSBColor":"38,17,100","Channel":[100,94,83],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"XXXXXXXXX","BSSId":"XXXXXXXX","Channel":13,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:06"}}
22:05:11 MQT: domoticz/in = {"idx":11,"nvalue":0,"svalue":"24.1;33.5;2","Battery":100,"RSSI":10}
22:05:11 MQT: tele/sonoff-LED/SENSOR = {"Time":"2019-12-15T22:05:11","AM2301":{"Temperature":24.1,"Humidity":33.5},"TempUnit":"C"}
22:08:56 CMD: Backlog Template; Module; GPIO 255
22:08:56 MQT: stat/sonoff-LED/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
22:08:56 MQT: stat/sonoff-LED/RESULT = {"Module":{"18":"Generic"}}
22:08:56 MQT: stat/sonoff-LED/RESULT = {"GPIO0":{"0":"Brak"},"GPIO1":{"0":"Brak"},"GPIO2":{"0":"Brak"},"GPIO3":{"0":"Brak"},"GPIO4":{"7":"WS2812"},"GPIO5":{"0":"Brak"},"GPIO9":{"0":"Brak"},"GPIO10":{"0":"Brak"},"GPIO12":{"2":"AM2301"},"GPIO13":{"0":"Brak"},"GPIO14":{"0":"Brak"},"GPIO15":{"0":"Brak"},"GPIO16":{"0":"Brak"}}

```
- [X ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
No rules used

```
- [ X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
22:12:29 MQT: stat/sonoff-LED/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff-LED"],"Topic":"sonoff-LED","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:12:29 MQT: stat/sonoff-LED/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:30","StartupUTC":"2019-12-15T21:04:59","Sleep":50,"CfgHolder":4617,"BootCount":23,"SaveCount":84,"SaveAddress":"F8000"}}
22:12:29 MQT: stat/sonoff-LED/STATUS2 = {"StatusFWR":{"Version":"7.1.2.6(tasmota)","BuildDateTime":"2019-12-15T20:00:55","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"250/1151"}}
22:12:29 MQT: stat/sonoff-LED/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXXXX",""],"TelePeriod":300,"Resolution":"XXXXXXXX","SetOption":["00008009","XXXXXXXX","00000000","00000000"]}}
22:12:29 MQT: stat/sonoff-LED/STATUS4 = {"StatusMEM":{"ProgramSize":565,"Free":436,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":2,"Features":["00000415","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
22:12:29 MQT: stat/sonoff-LED/STATUS5 = {"StatusNET":{"Hostname":"sonoff-LED-5837","IPAddress":"192.XXX.XXX.XXX","Gateway":"192.XXX.XXX.XXX","Subnetmask":"255.255.255.0","DNSServer":"192.XXX.XXX.XXX","Mac":"XXXXXXXX","Webserver":2,"WifiConfig":4}}
22:12:29 MQT: stat/sonoff-LED/STATUS6 = {"StatusMQT":{"MqttHost":"192.XXX.XXX.XXX","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1836CD","MqttUser":"XXXXXXX","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:12:29 MQT: stat/sonoff-LED/STATUS7 = {"StatusTIM":{"UTC":"Sun Dec 15 21:12:29 2019","Local":"Sun Dec 15 22:12:29 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:36","Sunset":"16:53"}}
22:12:29 MQT: stat/sonoff-LED/STATUS10 = {"StatusSNS":{"Time":"2019-12-15T22:12:29","AM2301":{"Temperature":24.2,"Humidity":33.4},"TempUnit":"C"}}
22:12:29 MQT: stat/sonoff-LED/STATUS11 = {"StatusSTS":{"Time":"2019-12-15T22:12:29","Uptime":"0T00:07:30","UptimeSec":450,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":100,"Color":"FFEFD4","HSBColor":"38,17,100","Channel":[100,94,83],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"XXXXXX","BSSId":"XXXXXXXXX","Channel":13,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
turn on LEDs, wait ~20 min try to change color, or turn them off.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
All 300 LEDs, as defined in tasmota code, will be controlled all the time

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-15:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-16 by ascillato2

wilqp, 2019-12-19:
PSU is 5V/30A so it might not be an issue.

nevrrmind, 2020-01-13:
did you power the strip from both ends?

wilqp, 2020-01-13:
Yes, even from 3 points: both ends and in the middle.