arendst/Tasmota #7237: MQ-sensor support

DerNick, 2019-12-16:
**Have you looked for this feature in other issues and in the wiki?**  
Yep but I didn´t find anything that fits.
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No, no problems ;) Tasmota works perfect! I would just like to get support for that Sensor.
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I would like to have the possibility to use the MQ-sensors (such as MQ-7 for co messurement) in tasmota. 
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
/
**Additional context**  
_Add any other context or screenshots about the feature request here._
there are many diffrent and cheap types of MQ sensors! for example to messure gas,air quality... for example from aliexpress: https://de.aliexpress.com/item/32502213304.html?src=google&src=google&albch=shopping&acnt=494-037-6276&isdl=y&slnk=&plac=&mtctp=&albbt=Google_7_shopping&aff_platform=google&aff_short_key=UneMJZVf&&albagn=888888&albcp=1705854617&albag=67310370915&trgt=296904914040&crea=de32502213304&netw=u&device=c&gclid=Cj0KCQiA89zvBRDoARIsAOIePbCt0UXTyxOkg9mE1eYTS26c34O-uReLqNEyzKjjPTxzLYN1bLllAsMaAmuAEALw_wcB&gclsrc=aw.ds Thanks for reading my request! I realy love Tasmota and daily use it in my devices. THANKS!
**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-16:
It is already supported!!!!!!!

ascillato2, 2019-12-16:
In issues and in the wiki is explained how to use it. 

Search for MQ sensor and chirp sensor

closed 2019-12-16 by ascillato2

DerNick, 2019-12-17:
oh sorry i didn´t see that. Thanks for your quick answer! I am gona buy these sensors now ;)